MichaIng/DietPi #3272: Redis-Server fails to start

g7kse, 2019-12-20:
RPI Buster fresh install on RPi3

This is just a quick bug report on a test set up I am running. The Redis-Server is a bit hit and miss, sometimes it starts but more often than not it fails. the most useful output I can find is

`root@DietPi:~# redis-server /etc/redis/redis.conf

*** FATAL CONFIG FILE ERROR ***
Reading the configuration file, at line 171
>>> 'logfile /var/log/redis/redis-server.log'
Can't open the log file: No such file or directory
`

I guess this points to a missing file or one that wasn't successfully moved when I moved everything from the micro SD to an SSD using the drive tool. I'll keep looking for a reason but just in case others might find it. It is conceivable that the micro SD is shot though.

MichaIng, 2019-12-20:
@g7kse 
Many thanks for your report.

Please start Redis via it's systemd unit: `systemctl start redis-server`
The systemd unit, if I see right, creates the log dir and file automatically, and assures permissions to write to it, with the run user "redis".

Note that service start is done on boot automatically, hence when you start it manually, you'll have two mist likely conflicting instances. If no Redis is available after boot, check the service log:
```
journalctl -u redis-server
```

MichaIng, 2020-01-01:
Most likely a duplicate of: #3291

closed 2020-01-01 by MichaIng