MichaIng/DietPi #3273: .rtorrent.rc using deprecated commands

bbsixzz, 2019-12-20:
Creating a bug report/issue

Required Information

DietPi version |
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version | buster 10.2

Kernel version | 
Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

SBC device | RPi 4 Model B (armv7l)

Power supply used | 3A Official

SDcard used | SanDisk ultra

Extra details

There are lots of errors in rTorrent console [as per this post](https://gist.github.com/bryanjswift/1525912/9033c9b9f9ea41b5dc69e6061501a82c8ebb5b52#gistcomment-2969399)
 
[image]

I edited the file manually and fixed them.

MichaIng, 2019-12-20:
@bbsixzz 
Many thanks for your report. If I see right, those commands/config entries are deprecated already for a long time and have now finally been removed? I remember I updated the config once already, but the `session_save` schedule entry was missed obviously.

... okay there were lots of commands/variables which were not updated to post 0.9.

I hope I got all now: https://github.com/MichaIng/DietPi/commit/73b07d10d8eee2bdc282e1ca14a10516aea4a51b
Might you compare with your config?

Great find and report btw :+1:, quite important one!

bbsixzz, 2019-12-21:
I'm testing your file as it was more complete, everything seems in order.

MichaIng, 2019-12-21:
@bbsixzz 
Perfect, many thanks for testing. Since it was not created by me and also quite differs from the official [template/example](https://github.com/rakshasa/rtorrent/wiki/CONFIG-Template) AND I do not 100% understand everything (e.g. how setting a variable save.session can trigger an event), I am open for suggestions about changes. Probably we should even switch to the official template and adjust the base/download/session/log/config paths only?

Another thing I'll do is pulling the latest release automatically: https://api.github.com/repos/rakshasa/rtorrent/releases/latest
Currently it is a fixed one that needs to be updated in code manually.

bbsixzz, 2019-12-21:
@MichaIng 
I too had a look at that template and mainly noticed settings like these:

Original config:
```
throttle.max_uploads.set = 4
throttle.max_downloads.set = 30
```

New config:
```
throttle.max_uploads.set = 100
throttle.max_downloads.set (missing, load many simultaneous torrents = crash?)
```

The new template is modern, but I have no idea what settings would be good for SBC's.

It would be great if someone with rTorrent experience could advise.

MichaIng, 2019-12-21:
@bbsixzz 
Limiting downloads and seeds by default sounds reasonable to me, as long as clients/web UI show reasonable errors when the limit has been reached, so users can adjust, when required.

However yeah long term someone with more experience and regular rTorrent use would be great.

bbsixzz, 2019-12-21:
On a tangent, I noticed torrents appear as **Completed** after restarting rTorrent.

Out of 43 torrents (all from 3 private trackers) only 3 start seeding automatically!

Two of the **Seeding** torrents are from the same tracker; doesn't seem to makes any difference.

[image] [image]

If I **Start** any **Completed** torrent in ruTorrent it will change to **Paused**.

If I **Stop** the same torrent it will show **Completed** and ALL other torrents will show **Seeding**.

Finally, if I start the remaining **Completed** torrent it will show **Seeding** as well.

If I start multiple **Completed** torrents they will all show **Paused** until one of them is stopped.

All OTHER torrents will then show **Seeding** and I will have to manually start each **Paused** torrent.

I have no idea what's going on, but I know rTorrent should automatically start seeding.

I uncommented the following lines in .rtorrent.rc but it didn't change anything:

```
# Enable the default ratio group.
# ratio.enable=yes
# Change the limits, the defaults should be sufficient.
# Upload to a minimum ratio of 1.01
# ratio.min.set=101
# Upload to a maximum ratio of 1.25
# ratio.max.set=125
# Upload a minimum of x MB
# ratio.upload.set=1M

# When seeding ratio is reached close the torrent
# method.set = group.seeding.ratio.command,d.close=
```

Bonus: all of those commands are deprecated except maybe group.seeding.ratio.command

Is this making any sense?

It's like rTorrent doesn't know where the files are until a torrent is first started and then stopped!

MichaIng, 2019-12-22:
@bbsixzz 
Interesting, this years old issue here states the same: https://github.com/rakshasa/rtorrent/issues/29
This setting seems to never have worked as expected, lol. It should limit the number of seeding torrents, although documentation is missing and name as well not self-explaining.

There is another one: `throttle.max_uploads.set`
Does this limit in your case the max number of seeding `torrents` or the max number of uploading `connections`, e.g. when having multiple peers per seeing torrent.

closed 2020-01-01 by MichaIng