MichaIng/DietPi #3274: Transmission Client Permission Denied Error

AdamF100001, 2019-12-20:
### Creating a bug report/issue

#### Required Information

root@DietPi:~# #!/bin/bash
root@DietPi:~# G_DIETPI_VERSION_CORE=6
root@DietPi:~# G_DIETPI_VERSION_SUB=26
root@DietPi:~# G_DIETPI_VERSION_RC=3
root@DietPi:~# G_GITBRANCH='master'
root@DietPi:~# G_GITOWNER='MichaIng'

DietPi v6.26.3
Device model : RPi 4 Model B (armv7l)

#### Additional Information (if applicable)
- Software title | Transmission Client
software installed freshly
- Can this issue be replicated on a fresh installation of DietPi?

Bug report sent, reference code: acfeb6c1-f9bc-4ee9-a5aa-972f3aca413e
#### Steps to reproduce

1. Start a torrent download.
2. Then the Error Permission denied (/mnt/ will be shown in the client.

#### Expected behaviour

- ... The torrent should successfully download. 

#### Actual behaviour

- ... Torrent download not successfully completing 

#### Extra details

full error string

Permission denied (/mnt/USB256GB/downloads/Brassic.S01E02.1080p.HDTV.x264-LiNKLE[rarbg]/Brassic.S01E02.1080p.HDTV.x264-LiNKLE.m
- ...
[image]

MichaIng, 2019-12-20:
@AdamF100001 
The user must be member of the `dietpi` group to have full R/W access to the downloads dir. I guess you have a custom user account? Then do:
```
usermod -aG dietpi <username>
```

Fourdee, 2019-12-21:
@AdamF100001 

Thanks for the bug report.

Based on @MichaIng , @AdamF100001 copy and paste these commands:
```
systemctl stop transmission-daemon
sync
usermod -aG dietpi debian-transmission
chmod 775 -R /mnt/dietpi_userdata/downloads
chown -R debian-transmission:dietpi /mnt/dietpi_userdata/downloads
systemctl restart transmission-daemon
```

Then check Transmission web page for errors.

MichaIng, 2019-12-21:
Is it about the server or the client? The server service starts with `Group=dietpi` and `usermod -aG dietpi debian-transmission` is part of the install process. With 775 permissions on download dir this should be sufficient, it is with my tests.

But when using some CLI/GUI client (https://transmissionbt.com/) one must probably take care those permissions. Ahh but now I see it's about the web interface, which is part of the server install. Sorry for misunderstanding 😉.

AdamF100001, 2019-12-22:
I have just checked my transmission client. It is still displaying the same permission denied error as before.

AdamF100001, 2019-12-22:
> 
> 
> Is it about the server or the client? The server service starts with `Group=dietpi` and `usermod -aG dietpi debian-transmission` is part of the install process. With 775 permissions on download dir this should be sufficient, it is with my tests.
> 
> But when using some CLI/GUI client (https://transmissionbt.com/) one must probably take care those permissions. Ahh but now I see it's about the web interface, which is part of the server install. Sorry for misunderstanding wink.
> 
> Just tested again, works fine. When using a custom dir (not /mnt/dietpi_userdata or the one moved via dietpi-drive_manager), the following should be sufficient:
> 
> ```
> chmod 775 /mnt/USB256GB/downloads
> chown :dietpi /mnt/USB256GB/downloads
> ```
> 
>     * This does not change owner, but group only and allows dietpi group R/W access, which includes media players and Sonarr/Radarr etc access, when installed via dietpi-software.
> 
>     * `/mnt/dietpi_userdata/downloads` however should have correct permission, even when moved to external drive thinking.

I have ran the following commands

chmod 775 /mnt/USB256GB/downloads
chown :dietpi /mnt/USB256GB/downloads

I will keep you posted.

AdamF100001, 2019-12-23:
its still the same issue & error.

MichaIng, 2019-12-23:
@AdamF100001 
Which file system is it, probably it does not support UNIX permissions?
```
df -T /mnt/USB256GB
ls -l /mnt/USB256GB
```

Ah and btw, if its on the same files, my commands only changed the downloads directory permissions but not for any sub directory or contained file that existed already. It should have been better:
```
chmod -R 775 /mnt/USB256GB/downloads
chown -R :dietpi /mnt/USB256GB/downloads
```
`-R` means to include all contained files/dirs **r**ecursively.

AdamF100001, 2019-12-24:
> 
> 
> @AdamF100001
> Which file system is it, probably it does not support UNIX permissions?
> 
> ```
> df -T /mnt/USB256GB
> ls -l /mnt/USB256GB # Erase/rephrase any private date
> ```
> 
> Ah and btw, if its on the same files, my commands only changed the downloads directory permissions but not for any sub directory or contained file that existed already. It should have been better:
> 
> ```
> chmod -R 775 /mnt/USB256GB/downloads
> chown -R :dietpi /mnt/USB256GB/downloads
> ```
> 
> `-R` means to include all contained files/dirs **r**ecursively.

I have ran all the commands above & rebooted my pi.

please see terminal screenshot.

[image]

MichaIng, 2019-12-24:
@AdamF100001 
Okay, file system is `ext4`, which is fine. Owner of the one dir is `777`? Probably a removed user name? However should not really matter since dietpi group has R/W access on them, although probably transmission does not like invalid owners.
Please try `chown -R dietpi <dir_name>` on that dir with user `777`, if this is the downloads dir?

Btw I guess you erased the directory names from the screen, since they should be just on the right side of the modification time/year?

AdamF100001, 2019-12-26:
I think the error has been resolved. I am currently downloading a Linux distro. I will seed this torrent for 24 hours if there are no issues, this thread can be closed.

Thanks for the help & support.

AdamF100001, 2019-12-26:
> @AdamF100001
> Okay, file system is `ext4`, which is fine. Owner of the one dir is `777`? Probably a removed user name? However should not really matter since dietpi group has R/W access on them, [...]

I think the error has been resolved. I am currently downloading a Linux distro. I will seed this torrent for 24 hours if there are no issues, this thread can be closed.

Thanks for the help & support.

AdamF100001, 2019-12-27:
Please close this thread as the issue has been resolved. Thanks

[image]

Fourdee, 2019-12-27:
@AdamF100001 

Thank you Adam, and for the donation :heart: :+1: 

Great to hear its now working, i'll mark this as closed, however if the issue reoccurs, please reopen this ticket, then I will arrange a remote connection allowing us to do a speedy resolve.

closed 2019-12-27 by Fourdee