MichaIng/DietPi #3275: Nextcloud Fails to Install

DevinCharles, 2019-12-21:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.26.3 master MichaIng
- Distro version | Debian 10.2
- Kernel version | Linux DietPi 4.19.57-v7l+ #1244 SMP Thu Jul 4 18:48:07 BST 2019 armv7l GNU/Linux
- SBC device | RPi 4 Model B (armv7l)
- Power supply used | CanaKit 3.5A USB-C
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Nextcloud
- Fresh Install of Nextcloud on fresh install of DietPi, no other software installed
- Can be replicated (this was a fresh install)

#### Steps to reproduce
0. Login as root (also tried as dietpi with sudo)
1. Install Dietpi
2. Install Nextcloud

#### Expected behaviour
Nextcloud should install

#### Actual behaviour
Install fails

#### Extra details
DietPi-Software: ncc maintenance:install
 ¦  - Exit code: 1
 ¦  - DietPi version: v6.26.3 (MichaIng/master) | HW_MODEL:4 | HW_ARCH:2 | DISTRO:5
 ¦  - Image creator: DietPi Core Team
 ¦  - Pre-image: Raspbian Lite
 ¦
 ¦ Log file content:
 ¦ Error while trying to create admin user: Failed to connect to the database: An
 ¦ exception occurred in driver: SQLSTATE[HY000] [1698] Access denied for user
 ¦ 'oc_admin'@'localhost'
 ¦  ->
 ¦
 ¦ If problems persist, please report at https://github.com/MichaIng/DietPi/issues
 ¦ for investigation.
 ¦
 ¦ Unable to continue, DietPi-Software will now terminate.

MichaIng, 2019-12-21:
@DevinCharles 
Bug confirmed. Somehow the install command does not create the default oc_admin database user before using it. I rechecked the official docs and it seems to be a bug on Nextcloud side, as it is not intended to create this user prior to running the command, nor is there an option to e.g. define its name: https://docs.nextcloud.com/server/17/admin_manual/configuration_server/occ_command.html#command-line-installation-label

MichaIng, 2019-12-21:
Jep, no change to use the command with a pre-created database user:
> Error while trying to create admin user: Failed to connect to the database: An exception occurred in driver: SQLSTATE[HY000] [1045] Access denied for user 'oc_admin1'@'localhost' (using password: YES)

Nasty bug... we can of course do a complete manual install but that would be a VERY bad step back...

DevinCharles, 2019-12-21:
Thanks for addressing this so quickly! 

For anyone coming here with the same issue looking for a hacky workaround until this is fixed in v17.0.1:

Editing /DietPi/dietpi/dietpi-software to use:

```https://download.nextcloud.com/server/releases/nextcloud-17.0.1.tar.bz2```
instead of 
```https://download.nextcloud.com/server/releases/latest.tar.bz2```
seems to work.

Search for the line in nano using Ctrl+w

MichaIng, 2019-12-22:
@DevinCharles 
Or make it a one-liner:
```
sed -i '/download\.nextcloud\.com/s/latest\.tar\.bz2/nextcloud-17.0.1.tar.bz2/' /DietPi/dietpi/dietpi-software
```

Joulinar, 2019-12-22:
btw: I can confirm the one-liner workaround working 👍

closed 2020-01-01 by MichaIng