MichaIng/DietPi #3276: Raspberry pi froze on apt-get upgrade

Dr0bac, 2019-12-21:
Hi @MichaIng last night i ran apt-get update and apt-get upgrade. And i noticed it was stuck on mono compile.. before i also noticed that on this step raspberry pi 3b can be slow so i leave it overnight.. When i woke up, after 5 hours raspberry pi was still unresponsive so i unplugged it from power suply.

Is it mistake on freeze like this to leave rpi for few hours? I guess cpu temps was high?
And what should i do differently next time to avoid unplugging rpi from power ( i guess i can damage system and micro sd card that way?)

MichaIng, 2019-12-21:
@Dr0bac 
Hmm, mono should take long, but not that long. Did you try to access on a second SSH session, so did it really crash?

When CPU temps raise over 70°C, RPi should start to throttle clocks down, when raising over 75°C down to a minimum which should never lead to a crash.

So always try to connect on a second TTY/SSH session first and check e.g. via `htop` what the process does, in case kill it. If RPi is really completely unresponsive after hours, then yeah no other change the plugging from power. To check what made it crash, enable persistent system log:
- Disable/uninstall DietPi-RAMdisk, choose `None`, since rsyslog is usually unnecessary overhead, only doubling what journald does already.
- Enable persistent journald: `mkdir /var/log/journal`
- After boot/crash `journalctl` can be used to check all related system and service logs.

Dr0bac, 2019-12-22:
I found this threads:

https://github.com/mono/mono/issues/11776
https://stackoverflow.com/questions/53214618/failed-to-precompile-microsoft-codeanalysis-csharp-on-armbian-stretch

Few things come trough my mind, i remember when i update dietpi with dietpi-update dietpi then turn off all dietpi services (programs).. when i hit apt-get upgrade dietpi services was running i guess, so they were using some chunk of ram that mono compiler needed.. also i disabled swap file, so this also may be a problem..

Can you tell me what is the command to turn off and on dietpi services? I should turn them off befire next apt-get upgrade just in case

MichaIng, 2019-12-22:
@Dr0bac 
Mono requires up to 1 GiB RAM to compile if I remember right, so yep a swap file is required on RPi. However if RAM is insufficient, the compiling should terminate with an error message instead of system freezing, at least I regularly run into such on my VMs, when I forgot to raise RAM for compilation installs. Persistent syslogs/journal should reveal what was killed first by OOM killer 😉.

Dr0bac, 2019-12-22:
@MichaIng today i tried to install some dietpi optimised software and after
`Mode: Update & upgrade APT`
i got error 
`[  OK  ] DietPi-Software | G_AGUP`
`[ INFO ] DietPi-Software | Running apt simulation to check for errors, please wait...`
`[ INFO ] DietPi-Software | APT installation for: bash-doc -s, please wait...`
`E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).`
`[FAILED] DietPi-Software | G_AGI bash-doc -s`
`[ INFO ] DietPi-Software | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.`

then i tried `apt --fix-broken install` and got another error, don't remember which one but it asked me to try this `dpkg --configure -a` and i think i got error again.
later i tried again to install software and i got again `[FAILED] DietPi-Software | G_AGI bash-doc -s`
tried again `apt --fix-broken install` and now it went to mono compile, that was stuck on before.. and it succeed this time without problems..
Are those errors conected with apt upgrade i ran few days ago and never completed because of power unplug?

MichaIng, 2019-12-23:
> Are those errors conected with apt upgrade i ran few days ago and never completed because of power unplug?

Exactly. The mono core package failed to "configure" (which includes compiling), hence some front-end packages could not be configured as well, leaving APT in an inconsistent state, which breaks any further installs until resolved.

Dr0bac, 2019-12-23:
Thanks @MichaIng 
Another quick question, don't want to open new issue?
It's about emby server. I see that server have update available, i tried with apt-get update, but they don't have repository. So i invastigated a little and saw this on there forum..
So how i do this on dietpi if i installed emby trough dietpi-software?

https://emby.media/community/index.php?/topic/76711-how-to-update-emby-version-in-linux/

MichaIng, 2019-12-23:
@Dr0bac 
Simply reinstall it: `dietpi-software reinstall 41`

I just added this useful info to our online docs as well: https://dietpi.com/phpbb/viewtopic.php?p=1789#p1789

closed 2019-12-23 by Dr0bac