MichaIng/DietPi #3279: NextCloud Installation Error

Jerremy007, 2019-12-21:
#### Details:
- Date           | Sat 21 Dec 21:55:36 GMT 2019
- Bug report     | N/A
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (index=5)
- Command        | systemctl start redis-server
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. Go into diet optimized software
2. install NextCloud

#### Expected behaviour:

- install nextcloud without any errors

#### Actual behaviour:

 not long into the nextcloud installtion the error pops up
Job for redis-server.service failed because the control process exited with error code.

#### Extra details:

- ...
[image]

#### Additional logs:
```
Job for redis-server.service failed because the control process exited with error code.
See "systemctl status redis-server.service" and "journalctl -xe" for details.
```

MichaIng, 2019-12-22:
@Jerremy007 
Many thanks for your report.

First of all please see this workaround for a current bug in latest Nextcloud release: https://github.com/MichaIng/DietPi/issues/3275
But your error is of another kind, Redis server failed for some reason. Please paste: `journalctl -u redis-server`

Probably related: https://github.com/MichaIng/DietPi/issues/3272
Although using systemctl to start the service should allow it to pre-create the config file.

Jerremy007, 2019-12-22:
Hello, thank you very much for the reply.
I tried to install the NextCloud again following the same steps , although i already have pi-hole and open ssh installed, NextCloud installed successfully although with a few errors but the web interface works great. It did not want to work through Root login but it worked through DietPi login.
Here are the errors i received when installing it today , sorry if these can be categorized as another bug.

[image]
[image]
[image]

MichaIng, 2019-12-22:
@Jerremy007 
Yes this is expected since the database has already been created on previous install attempt and then the "maintenance:install" step is skipped. However you agreed the option to run it instead of next occ call. To apply the other intended changes (if you did not do this already):
```
ncc maintenance:update:htaccess # Mainly enables pretty URLs without the "/index.php/" part
ncc background:cron # Use system cron for background jobs instead of AJAX (very recommended!)
ncc db:convert-filecache-bigint -n # Convert the ID columns of the filecache to BigInt, the admin panel should else show a related warning
```
- You can do this all on-the-fly, no service stop required or such, afterwards your Nextcloud instance is identical to what is intended by succeeding DietPi-Software install.

MichaIng, 2019-12-23:
I'll mark this issue as closed now. Feel free to reopen if required.

closed 2019-12-23 by MichaIng

Nalsai, 2020-01-11:
I'm experiencing a similar issue.

This is my output of `journalctl -u redis-server`:
```
Jan 11 13:11:15 DietPi systemd[1]: Starting Advanced key-value store...
Jan 11 13:11:15 DietPi run-parts[834]: run-parts: executing /etc/redis/redis-server.pre-up.d/00_example
Jan 11 13:11:15 DietPi systemd[1]: redis-server.service: Control process exited, code=exited status=1
Jan 11 13:11:15 DietPi systemd[1]: Failed to start Advanced key-value store.
```

I'm on v6.28.0 (Fourdee/master) stretch for the Odroid HC1
Kernel version | Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l GNU/Linux

MichaIng, 2020-01-11:
@Nalsai 
Could you please paste the output of: `cat /var/log/redis/redis-server.log`
If it's empty (hourly RAMlog clearing), please run:
```
systemctl restart redis-server
sleep 2
cat /var/log/redis/redis-server.log
```

Nalsai, 2020-01-11:
The file does not exist and doesn't get created.

However I found out by searching through `journalctl -xe` that 
`always-show-logo no` in the last line of `/etc/redis/redis.conf` was the issue so I deleted it and now it works.

MichaIng, 2020-01-11:
@Nalsai 
Ah okay, yes on newest version we moved logs to journalctl, I assumed that yours was still configured to log to file because `journalctl -u redis-server` didn't contain much.

The with Debian Stretch incompatible setting has been solved for v6.29 with: https://github.com/MichaIng/DietPi/issues/3314
With MOTD enabled the fix is applied to v6.27/28 systems as well, however a bid hacky and of course not everyone has MOTD enabled.