MichaIng/DietPi #3283: V.6.27.2 - Openhab2 update error

ziocharly, 2019-12-23:
#### Details:
- Date           | Mon 23 Dec 16:02:19 CET 2019
- Bug report     | N/A
- DietPi version | v6.26.0 (MichaIng/dev)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux sdomotica 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (ID=4)
- Command        | G_AGUG
- Exit code      | 100
- Software title | DietPi-Update

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
WARNING: The following packages cannot be authenticated!
  openhab2 openhab2-addons
E: There were unauthenticated packages and -y was used without --allow-unauthenticated
```

MichaIng, 2019-12-23:
@ziocharly
It seems that there is an APT key missing for the related repo. Could you paste:
```
cat /etc/apt/sources.list{,.d/*.list}
```

Looks like openhab is pulled from bintray repo.

ziocharly, 2019-12-23:
> 
> @ziocharly <https://github.com/ziocharly>
> It seems that there is an APT key missing for the related repo. Could you paste:
> 
> cat /etc/apt/sources.list{,.d/*.list}
> Looks like openhab is pulled from bintray repo.

Thank you for promptly feedback !

I’ve paste the suggested source list, and run again the dietpi-update, but the error still remain :

Don’t worry, and thanks again !

Best wishes for a happy holidays !

Carlo.

Joulinar, 2019-12-23:
@ziocharly 
The request of @MichaIng was, that you ran the command `cat /etc/apt/sources.list{,.d/*.list}` and copy the output to the GitHub issue you opened 😃 

Like this:

```
root@DietPiDEMO:~# cat /etc/apt/sources.list{,.d/*.list}
deb http://raspbian.raspberrypi.org/raspbian/ buster main contrib non-free rpi
deb http://raspbian.raspberrypi.org/raspbian/ bullseye main
deb https://archive.raspberrypi.org/debian/ buster main ui
root@DietPiDEMO:~#
```

ziocharly, 2019-12-23:
> Il giorno 23 dic 2019, alle ore 21:52, Joulinar <notifications@github.com> ha scritto:
> 
> cat /etc/apt/sources.list{,.d/*.list}

Sorry, this is the output :

Joulinar, 2019-12-23:
does that mean there is no output at all?

ziocharly, 2019-12-23:
No, this is the output 

deb http://raspbian.raspberrypi.org/raspbian stretch main contrib non-free rpi
deb https://dl.bintray.com/openhab/apt-repo2 stable main
deb https://archive.raspberrypi.org/debian/ stretch main ui
deb https://dtcooper.github.io/raspotify jessie main
 
I don't know, unfortunately in the previous post it was not visible ...

ziocharly, 2019-12-23:
After suggested command:

wget -qO - 'https://bintray.com/user/downloadSubjectPublicKey?username=openhab'
sudo apt-key add -

the 'diet-update' proceed without any other issue.

Sorry for the initial misunderstanding.

MichaIng, 2019-12-23:
Ah jep sorry, I need to be a bid more clear, "please paste ..." should be "please run ... in terminal and paste the output here" 😅.

Great, indeed the bintray repo and I'm glad the key works as expected. Probably you added the repo a longer time ago (?) and the key has changed meanwhile or such. I know from another case where this happened, respectively a weak key was abandoned in favour of a stronger one or such, probably as well due to needs on Debian/cryptography library side.

ziocharly, 2019-12-24:
Yes, the original image has been configured by SDomotica
for HA and domotic use.
As you know, due to the Python 3.6 deprecation on HA,
I’ve updated successfully to the 6.27.X beta...
Thanks again !

MichaIng, 2019-12-26:
@ziocharly 
Great, I mark this issue as closed then. Feel free to reopen if required.

closed 2019-12-26 by MichaIng