MichaIng/DietPi #3284: Nextcloud Installation Error

symlagom, 2019-12-23:
### Creating a bug report/issue

#### Required Information
- DietPi version | v6.26.3
- Kernel version | Linux DietPi 4.14.66+
- SBC device | ODroid HC1
- Power supply used | 5V, 2A
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Nextcloud
- fresh installation  (Webserver Preference is NGINX)
- Can this issue be replicated on a fresh installation of DietPi? ->  Yes
- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`
b7dd696b-9c35-4baf-9830-f0f8859ea978

#### Steps to reproduce

1. Go to DietPi-Software
2. Install Nextcloud 

#### Expected behaviour
Complete Installation of Nextcloud

#### Actual behaviour
Installation stops with error

[image]

MichaIng, 2019-12-23:
@symlagom 
Many thanks for your report.

The error message is luckily pretty clear. How did you install PHP? DietPi-Software should pull PHP7.3 explicitly.

symlagom, 2019-12-23:
Thank you for the fast response.
I have not installed php by my own. It was already pre-installed, or am I wrong?
The php versions are a little bit confusing for me. The dietpi-service says that php version 7.3 is running. The php version output says, that version 7.4 is installed.

MichaIng, 2019-12-23:
@symlagom 
Strange, there should be no PHP pre-installed with DietPi. Where did you get the image from?

Indeed probably now two PHP versions are installed side-by-side. To get an overview:
```
dpkg --get-selections *php*
```

symlagom, 2019-12-24:
I have downloaded the dietpi image for the ODroid HC1 from the official dietpi homepage 2 days ago.

Yes, there are 2 php versions installed:
#dpkg --get-selections *php*
libapache2-mod-php7.4                           install
php                                             install
php-apcu                                        install
php-common                                      install
php-igbinary                                    install
php-mysql                                       install
php-pear                                        install
php-php-gettext                                 install
php-phpseclib                                   install
php-redis                                       install
php7.3-cli                                      install
php7.3-common                                   install
php7.3-curl                                     install
php7.3-fpm                                      install
php7.3-gd                                       install
php7.3-intl                                     install
php7.3-json                                     install
php7.3-mbstring                                 install
php7.3-mysql                                    install
php7.3-opcache                                  install
php7.3-readline                                 install
php7.3-xml                                      install
php7.3-zip                                      install
php7.4                                          install
php7.4-cli                                      install
php7.4-common                                   install
php7.4-json                                     install
php7.4-mysql                                    install
php7.4-opcache                                  install
php7.4-readline                                 install
phpmyadmin                                      install

I have installed the phpmyadmin tools from the dietpi-software environment. Maybe this is the reason for the 2 php versions. 
Is it possible to delete one version? Which version is needed for the nextcloud installation?

Joulinar, 2019-12-24:
usually you can install PHP using dietpi-software function. This should pull the required PHP version automatically. No need to install it by your own. I guess PHP7.4 would need to be removed. 

[image]

MichaIng, 2019-12-24:
@symlagom 
Okay its phpMyAdmin. The package pulls `php` meta packages without version string: https://packages.debian.org/stretch/phpmyadmin
This is kinda dangerous when using any 3rd party repos, like sury.org (which we use to install PHP7.3 on Stretch, since native PHP7.0 has been deprecated by many software titles), which has its own `php` package set, linking to a different version (7.4 obviously).

On Buster, the phpMyAdmin package has not been released for some reason, I was hoping for new maintainers which add it in a more intelligent and/or flexible way, either depend on an explicit version, then pull packages which explicit version string, or better depend on a set of PHP API strings, to allow any way of installing any supported PHP version (which should be all EXCLUDING PHP7.4, with that phpMyAdmin version, I guess).

Okay, mid term solution is to pull the phpMyAdmin scripts manually from their website and configure MariaDB and webserver manually. Should not be that hard. This as well enables install on Debian Buster then.

Short term solution for you is to make PHP7.3 the default PHP version again for the system:
```
update-alternatives --set php /usr/bin/php7.3
```

Now install `php7.4-cgi` which fulfils `php7.4` (=`php`) package dependencies, but is much more lightweight than the `libapache2-mod-php7.4` which is pulled by default and pulls even a whole Apache webserver.
```
apt install php7.4-cgi
apt purge libapache2-mod-php7.4 apache2-bin
apt autoremove --purge
```

Restart Nginx, I guess its startup conflicted with unintended Apache install, lets restart php7.3-fpm as well:
```
systemctl restart php7.3-fpm
systemctl restart nginx
```

Now see if you can access Nextcloud.

symlagom, 2019-12-25:
Thank you for your support.
I have deinstalled the phpMyAdmin before you wrote your answer and it worked. I could install the nextcloud. Later I had some additional problems because MariaDB version was required and the dietpi-software installer only installed version 10.1, but this is another topic.

I am not sure if I have to follow your steps. Do I need phpMyAdmin for the nextcloud installation?

Joulinar, 2019-12-25:
@symlagom 
Usually phpMyAdmin is not needed for NextCloud. If you select NextCloud from DietPi Software catalog, it should install all required components automatically. No need to setup MariaDB by its own. However there is a bug in current NextCloud version 17.0.2 leading installation to fail during setup. But @MichaIng already provided a workaround.

#3275

MichaIng, 2019-12-25:
@symlagom 
Yes phpMyAdmin is just a web UI frontend for MySQL/MariaDB. However it can be accessed from console as well via mysql/mariadb command and for our offered software title is not required to be accessed at all. All required databases and users are created automatically by our install scripts.

symlagom, 2019-12-31:
With your help I could install Nextcloud and MariaDB. You have written that all required configurations are already done. How can I understand this? What I have to do besides to install nextcloud and MariaDB via the wizard?
On other systems I have configured nextcloud and the database by my own. Is there any manual how to set up nextcloud on DietPi?

Joulinar, 2019-12-31:
@symlagom 
There is no need to do any manual steps because guys scripted it that way, that usually all steps are done automatically. One of the big advantages of DietPi. So if you are using `dietpi-software` catalog and it finished successfully, the only thing you need to do is to enjoy the software 😄 

Some small user guide can be found [here ](https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=3026#p3026)

MichaIng, 2019-12-31:
I made a start to implement the non-APT install method: https://github.com/MichaIng/DietPi/pull/3294
Needs to be done now since Stretch install is otherwise broken and as it enables install on Debian Buster as well, where no APT package is available.

PR #3294 (2020-01-01): DietPi-Software | phpMyAdmin: Switch to non APT install
**Status**: Ready

Fixes: #3284

**Commit list/description**:
+ DietPi-Software | phpMyAdmin: Switch to non APT install to be compatible with custom PHP versions (Stretch) and Debian Buster (no APT package available): https://github.com/MichaIng/DietPi/issues/3284
closed 2020-01-01 by MichaIng