MichaIng/DietPi #3285: dietpi-vpn - traffic splitting, preferably with rTorrent

bbsixzz, 2019-12-25:
I've read rumblings about an upcoming dietpi-vpn tool to simplify the setup.

Would you please consider including a config that only tunnels rTorrent traffic?

SSH/SFTP access is limited to LAN when wireguard is active right now.

Here's the current convoluted procedure with [openvpn](https://github.com/rakshasa/rtorrent/wiki/VPN-with-Traffic-Splitting).

Cheers

MichaIng, 2019-12-25:
@bbsixzz 
Many thanks for your request.

Jep including and excluding certain services from a VPN tunnel (when DietPi is the client) is a planned feature. However it might take a while until I find time to create this script.

bbsixzz, 2020-06-14:
Would a bitcoin donation speed this up? I could earmark it. It sucks not being able to do remote support.

MichaIng, 2020-06-14:
I'm currently working on a broader network setup rework which should be a good basement for this, but honestly not sure when I can pick it up. Probably we should setup some bounty system to motivate others picking up specific feature requests 😉.

For now I suggest you check out some guides on the web, e.g. this one which implements it based on a certain UNIX user name which are then forced to use the VPN: https://www.htpcguides.com/force-torrent-traffic-vpn-split-tunnel-debian-8-ubuntu-16-04/
This is for PIA but can be done for any other VPN as well which is based on OpenVPN.
It is quite complete with additional hardening (kill-switch, certain leak prevention) and some interesting thing I did not know about: Reverse Path Filtering

bbsixzz, 2020-06-18:
Ok, that's too rich for my blood. I would gladly contribute if this could be accelerated, WireGuard is allegedly perfect for this.

MichaIng, 2021-03-28:
Marking as closed in favour of the earlier request that has some idea collected already: #2758

closed 2021-03-28 by MichaIng