MichaIng/DietPi #3287: Img request for LOD linux on dex

krishnasen2401, 2019-12-28:
https://forum.xda-developers.com/galaxy-note-9/themes/linux-dex-void-linux-xfce4-t3976395/

Also please add asterix  server for Voip server reference image is freepbx for raspbian

MichaIng, 2019-12-28:
@krishnasen2401 
Many thanks for your request.

LoD seams to be an own OS, how can this be incorporated with DietPi OS?

krishnasen2401, 2019-12-28:
Linux on dex is android which run ubuntu images(xienial and ubuntu 16) it's a linux on arm-android similar to linux on windows with subsystem we need to .img file to create a vm like container here are some samples
Let me know if there are any generic-arm images to check for vm compatibility 
[image]
[image]

MichaIng, 2019-12-28:
@krishnasen2401 
Okay, first of all DietPi is based on Debian and Raspbian, not Ubuntu. Both distros are similar, but not identical, according to certain needs and configs that DietPi expect and does.

However LoD should be able to run different Linux distros anyway. We have no generic ARM image, since all ship kernel, bootloader and firmware required for the individual SBC model. And I am pretty sure that LoD as well is not able to run any image, but requires certain disc partitioning/Linux/bootloader/firmware to run or invokes own ones. So best is to find a Debian-based image that runs on LoD and then convert it into DietPi via: https://github.com/MichaIng/DietPi/blob/master/PREP_SYSTEM_FOR_DIETPI.sh

krishnasen2401, 2019-12-29:
would try this and also port to one of my old android phone(nexus 5) which can run Linux. Open a request for a astriex(voip server present for freepbx img but no seperate installation procedure for existing system) and any one of the chat servers hopefully rocketchat

closed 2019-12-29 by krishnasen2401

MichaIng, 2019-12-29:
@krishnasen2401 
Jep RocketChat request is open already: https://github.com/MichaIng/DietPi/issues/1902
When talking about Nextcloud, it's "Talk" app allows VoIP as well, but of course when not using file sharing features anyway it is overhead compared to bare VoIP server currently.