MichaIng/DietPi #3288: DietPi-RAMdisk | Using "vmtouch" to use native page cache

MichaIng, 2019-12-28:
I am not 100% happy with DietPi-RAMdisk. Changes to those files should not rely on a clean shutdown, which is currently required to write RAMdisk back to disk. As well it causes confusion when files are present on two locations.

Actually since Linux has a very good disk caching system already, why not using that one. It caches often used files/scripts automatically. But if we really want DietPi scripts to be most responsive even when being called the first time, I found some tool which is able to lock files in cache: https://github.com/hoytech/vmtouch

Available from Debian repo since Buster: https://packages.debian.org/buster/vmtouch
But I would compile own binaries and implement with systemd unit, since the sysvinit service is broken anyway: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=947603

Tested it and works as expected. Executing locked example scripts from /boot/dietpi results in exactly same speed compared to tmpfs scripts. The vmtouch binary itself is extremely lightweight, the overall used memory matches nearly exactly the locked files, hence not more then a few KiB overhead, most likely similar compared to an active tmpfs.

Requires some more testing but actually not much risk. Backwards compatibility can be simply achieved via symlink: `ln -sf /boot /DietPi`

Trunkzeh, 2019-12-28:
@MichaIng Are you going to push this as part of a 6.27.3? I'm going to be moving the majority of my internal services off my RPI0 to the RPI4 once 6.27 is stable; which will require a clean reinstall. The RPI0 will then be repurposed for beta testing.

MichaIng, 2019-12-28:
@Trunkzeh 
Nope, definitely not something for v6.27, as I aim to release before new year and this change is too deep to do required testing and all the code changes in this short time. I just wanted to write down the idea to not forget about this 😉.

Joulinar, 2019-12-29:
@MichaIng 
you know we are always happy to test new stuff. At least my 2 cents to support these great project

MichaIng, 2020-03-03:
I compiled vmtouch packages: https://dietpi.com/downloads/binaries/buster/
They are basically compatible with the default Debian packages, hence if they compile a newer version, it will continue to use the existing config file etc. I used a version string suffix "-dietpi1" that is treated as newer than all Debian patch version suffixes, but older if a new upstream version string is present.

By default the shipped config file locks `/boot/dietpi` and `/boot/dietpi.txt` in file system cache, however this can be changed via config file `/etc/default/vmtouch`, or the service can be disabled (`systemctl disable --now vmtouch`) or uninstalled (`apt purge vmtouch`) without breaking anything.

Joulinar, 2020-03-03:
ok did a quick test on my RPi4 and so far update to v6.29 was working without issues. As well next reboot was working fine. No error messages on `journalctl`.

MichaIng, 2020-03-03:
@Joulinar 
Many thanks ❤️, so far so good 👍.

closed 2020-05-03 by MichaIng