MichaIng/DietPi #3289: fonts-dejavu-core fatal error

psi5asp, 2019-12-28:
#### Details:
- Date           | Sat 28 Dec 23:49:24 GMT 2019
- Bug report     | 95cdd443-703c-4379-bd24-f4a38f1d8ca2
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | Linux Pi-4 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (index=5)
- Command        | G_AGI ca-certificates-java openjdk-11-jre-headless openjdk-11-jdk-headless
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...Installing JDK Software
2. ...

#### Expected behaviour:

- ...Software should install with relevant working dependencies 

#### Actual behaviour:

- ...producing a fatal error and aborting instead.

#### Extra details:

- ...unable to sudo apt-get purge fonts-dejavu-core
 dpkg: unrecoverable fatal error, aborting:
 files list file for package 'fonts-dejavu-core' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)

#### Additional logs:
```
Selecting previously unselected package java-common.
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'fonts-dejavu-core' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)
```

psi5asp, 2019-12-29:
may be related, also under stretch Distro:

#### Details:
- Date           | Sun 29 Dec 01:25:07 GMT 2019
- Bug report     | e4d9134a-be3a-4af6-924a-28245e25b984
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 3 Model B+ (armv7l) (index=3)
- Kernel version | Linux Pi-4 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
- Distro         | stretch (index=4)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...fresh install of DietPi OS
2. ...tried to install JDK

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...these are fresh OS installs with select software packages installed. Internet is enabled and accessible both via ETH0 and wifi.

#### Additional logs:
```
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:2 https://archive.raspberrypi.org/debian stretch InRelease
Reading package lists...
E: Encountered a section with no Package: header
E: Problem with MergeList /var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_stretch_main_binary-armhf_Packages
E: The package lists or status file could not be parsed or opened.
```

---------------------------------------------------------------------

psi5asp, 2019-12-29:
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                             Version                       Architecture                  Description
+++-================================================-=============================-=============================-=====================================================================================================
ii  adduser                                          3.115                         all                           add and remove users and groups
ii  adwaita-icon-theme                               3.22.0-1+deb9u1               all                           default icon theme of GNOME
ii  alsa-utils                                       1.1.3-1                       armhf                         Utilities for configuring and using ALSA
ii  apt                                              1.4.9                         armhf                         commandline package manager
ii  apt-transport-https                              1.4.9                         armhf                         https download transport for APT
ii  apt-utils                                        1.4.9                         armhf                         package management related utility programs
ii  aptitude                                         0.8.7-1                       armhf                         terminal-based package manager
ii  aptitude-common                                  0.8.7-1                       all                           architecture independent files for the aptitude package manager
ii  base-files                                       9.9+rpi1+deb9u11              armhf                         Debian base system miscellaneous files
ii  base-passwd                                      3.5.43                        armhf                         Debian base system master password and group files
ii  bash                                             4.4-5                         armhf                         GNU Bourne Again SHell
ii  bash-completion                                  1:2.1-4.3                     all                           programmable completion for the bash shell
ii  bc                                               1.06.95-9                     armhf                         GNU bc arbitrary precision calculator language
ii  bsdutils                                         1:2.29.2-1+deb9u1             armhf                         basic utilities from 4.4BSD-Lite
ii  bzip2                                            1.0.6-8.1                     armhf                         high-quality block-sorting file compressor - utilities
ii  ca-certificates                                  20161130+nmu1+deb9u1          all                           Common CA certificates
ii  connectd                                         2.3.17                        armhf                         remote.it is a cloud service that provides secure connections to
ii  console-setup                                    1.164                         all                           console font and keymap setup program
ii  console-setup-linux                              1.164                         all                           Linux specific part of console-setup
ii  coreutils                                        8.26-3                        armhf                         GNU core utilities
ii  cpp                                              4:6.3.0-4                     armhf                         GNU C preprocessor (cpp)
ii  cpp-6                                            6.3.0-18+rpi1+deb9u1          armhf                         GNU C preprocessor
ii  crda                                             3.18-1                        armhf                         wireless Central Regulatory Domain Agent
ii  cron                                             3.0pl1-128+deb9u1             armhf                         process scheduling daemon
ii  curl                                             7.52.1-5+deb9u9               armhf                         command line tool for transferring data with URL syntax
ii  dash                                             0.5.8-2.4                     armhf                         POSIX-compliant shell
ii  dbus                                             1.10.28-0+deb9u1              armhf                         simple interprocess messaging system (daemon and utilities)
ii  dbus-x11                                         1.10.28-0+deb9u1              armhf                         simple interprocess messaging system (X11 deps)
ii  dconf-gsettings-backend:armhf                    0.26.0-2                      armhf                         simple configuration storage system - GSettings back-end
ii  dconf-service                                    0.26.0-2                      armhf                         simple configuration storage system - D-Bus service
ii  debconf                                          1.5.61                        all                           Debian configuration management system
ii  debianutils                                      4.8.1.1                       armhf                         Miscellaneous utilities specific to Debian
[... cut ...]
ii  xserver-xorg-video-vesa                          1:2.3.4-1+b3                  armhf                         X.Org X server -- VESA display driver
ii  xterm                                            327-2                         armhf                         X terminal emulator
ii  xz-utils                                         5.2.2-1.2                     armhf                         XZ-format compression utilities
ii  zlib1g:armhf                                     1:1.2.8.dfsg-5                armhf                         compression library - runtime

psi5asp, 2019-12-29:
Packages were corrupt for some reason - strange. Fixed it with:

sudo rm /var/lib/apt/lists/* -vf
sudo apt-get update

closed 2019-12-29 by psi5asp

MichaIng, 2019-12-29:
@psi5asp 
Again something for error auto detection: "E: Problem with MergeList" == Purge lists and pull freshly. `/DietPi/dietpi/func/dietpi-software apt-cache clean` btw does this as well, `apt clean` sadly removes the two cache files only (if enabled), not the list files.