MichaIng/DietPi #3291: redis-server fails to start after nextcloud install

dankerthrone, 2019-12-30:
#### Details:
- Date           | Mon 30 Dec 23:17:12 GMT 2019
- Bug report     | c563d1ea-d845-4d66-a5f5-43e13ebe57cf
- DietPi version | v6.26.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi Zero W (armv6l) (index=1)
- Kernel version | Linux DietPi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux
- Distro         | buster (index=5)
- Command        | systemctl start redis-server
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. I hopped onto dietpi-software
2. I selected nextcloud from the available list of software
3. I hit install

#### Expected behaviour:

- the CLI should, at some point, tell me that all is done and well

#### Actual behaviour:

- the CLI/installer kills a bunch of services:

> root@DietPi:~# dietpi-software
> [  OK  ] DietPi-Software | Initialised database
> [  OK  ] DietPi-Software | Reading database
> [  OK  ] DietPi-Software | Connection test: http://raspbian.raspberrypi.org/rasp                         bian/
> [  OK  ] NTPD: Network time sync | Completed
> [  OK  ] DietPi-Software | Free space check: path=/ | available=10568 MiB | requ                         ired=500 MiB
> [  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
> [  OK  ] DietPi-Software | Connection test: http://raspbian.raspberrypi.org/rasp                         bian/
> [  OK  ] NTPD: Network time sync | Completed
> [ SUB1 ] DietPi-Services > unmask
> [  OK  ] DietPi-Services | unmask : redis-server
> [  OK  ] DietPi-Services | unmask : mariadb
> [  OK  ] DietPi-Services | unmask : php7.3-fpm
> [  OK  ] DietPi-Services | unmask : lighttpd
> [  OK  ] DietPi-Services | unmask : cron
> [ SUB1 ] DietPi-Services > stop
> [  OK  ] DietPi-Services | stop : cron
> [  OK  ] DietPi-Services | stop : lighttpd
> [  OK  ] DietPi-Services | stop : php7.3-fpm
> [  OK  ] DietPi-Services | stop : mariadb
> [  OK  ] DietPi-Services | stop : redis-server

and after installing nextcloud from the repositories tries to start them back up again:

> Enabling dietpi-nextcloud: ok
> Run "service lighttpd force-reload" to enable changes
> Enabling dietpi-dav_redirect: ok
> Run "service lighttpd force-reload" to enable changes
> [  OK  ] DietPi-Software | systemctl restart mariadb
> [FAILED] DietPi-Software | systemctl start redis-server
> [ INFO ] DietPi-Software | If problems persist, please report at https://github.                         com/MichaIng/DietPi/issues for investigation.
> [FAILED] DietPi-Software | systemctl start redis-server
> [ INFO ] DietPi-Software | If problems persist, please report at https://github.com/MichaIng/DietPi/issues for investigation.
> [ INFO ] DietPi-Bugreport | Packing upload archive, please wait...
> [  OK  ] DietPi-Bugreport | Connection test: ssh.dietpi.com
> [  OK  ] DietPi-Bugreport | Bug report sent, reference code: c563d1ea-d845-4d66-a5f5-43e13ebe57cf

#### Extra details:

- ...

#### Additional logs:
```
Job for redis-server.service failed because the control process exited with error code.
See "systemctl status redis-server.service" and "journalctl -xe" for details.
```

`systemctl status redis-server.service` then print out: 

> 

> ● redis-server.service - Advanced key-value store
>    Loaded: loaded (/lib/systemd/system/redis-server.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Mon 2019-12-30 23:22:21 GMT; 8min ago
>      Docs: http://redis.io/documentation,
>            man:redis-server(1)
>   Process: 6775 ExecStart=/usr/bin/redis-server /etc/redis/redis.conf (code=exited, status=1/FAIL
> URE)
> 
> Dec 30 23:22:21 DietPi systemd[1]: redis-server.service: Service RestartSec=100ms expired, scheduling res
> tart.
> Dec 30 23:22:21 DietPi systemd[1]: redis-server.service: Scheduled restart job, restart counter is at 5.
> Dec 30 23:22:21 DietPi systemd[1]: Stopped Advanced key-value store.
> Dec 30 23:22:21 DietPi systemd[1]: redis-server.service: Start request repeated t
> oo quickly.
> Dec 30 23:22:21 DietPi systemd[1]: redis-server.service: Failed with result 'exit
> -code'.
> Dec 30 23:22:21 DietPi systemd[1]: Failed to start Advanced key-value store.

dankerthrone, 2019-12-30:
journalctl -xe outputs

> - Logs begin at Mon 2019-12-30 23:46:34 GMT, end at Mon 2019-12-30 23:57:59 GMT
> . --
> Dec 30 23:46:38 DietPi systemd[1]: Mounted /DietPi.
> -- Subject: A start job for unit DietPi.mount has finished successfully
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- A start job for unit DietPi.mount has finished successfully.
> -- 
> -- The job identifier is 18.
> Dec 30 23:46:38 DietPi systemd[1]: Mounted /var/log.
> -- Subject: A start job for unit var-log.mount has finished successfully
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support
> -- 
> -- A start job for unit var-log.mount has finished successfully.
> -- 
> -- The job identifier is 11.
> Dec 30 23:46:38 DietPi systemd[1]: Mounted /tmp.
> -- Subject: A start job for unit tmp.mount has finished successfully
> -- Defined-By: systemd
> -- Support: https://www.debian.org/support

Joulinar, 2019-12-31:
Hi,
Maybe you can past output from `journalctl -u redis-server`

dankerthrone, 2019-12-31:
below is the output for
root@DietPi:~# journalctl -u redis-server

> -- Logs begin at Tue 2019-12-31 09:40:34 GMT, end at Tue 2019-12-31 09:58:42 GMT
> . --
> Dec 31 09:41:25 DietPi systemd[1]: Starting Advanced key-value store...
> Dec 31 09:41:27 DietPi redis-server[739]: *** FATAL CONFIG FILE ERROR ***
> Dec 31 09:41:27 DietPi redis-server[739]: Reading the configuration file, at line 171
> Dec 31 09:41:27 DietPi redis-server[739]: >>> 'logfile /var/log/redis/redis-server.log'
> Dec 31 09:41:27 DietPi redis-server[739]: Can't open the log file: No such file 
> or directory
> Dec 31 09:41:27 DietPi systemd[1]: redis-server.service:
>  Main process exited, code=exited, status=1/FAILURE
> Dec 31 09:41:27 DietPi systemd[1]: redis-server.service:
>  Failed with result 'exit-code'.
> Dec 31 09:41:27 DietPi systemd[1]: Failed to start Advan
> ced key-value store.
> Dec 31 09:41:28 DietPi systemd[1]: redis-server.service: Service RestartSec=100m
> s expired, scheduling restart.
> Dec 31 09:41:28 DietPi systemd[1]: redis-server.service: Scheduled restart job, 
> restart counter is at 1.
> Dec 31 09:41:28 DietPi systemd[1]: Stopped Advanced key-value store.
> Dec 31 09:41:28 DietPi systemd[1]: Starting Advanced key-value store...
> Dec 31 09:41:30 DietPi redis-server[765]: *** FATAL CONFIG FILE ERROR ***

Joulinar, 2019-12-31:
@dankerthrone 
looks like a similar issue like #3272 

pls can you check if you have a directory `/var/log/redis` on your system and what are the permissions.

Should looks like this
```
root@DietPiVM1:~# cd /var/log
root@DietPiVM1:/var/log# ls -l|grep redis
drwxr-s--- 2 redis adm    60 Dez 31 11:22 redis
root@DietPiVM1:/var/log#
```

dankerthrone, 2019-12-31:
So obviously for some reason I didn't have a log file for redis to write to, so
`nano /var/log/redis/redis-server.log`

Unfortunately, since the directory didn't exist, I couldn't use nano to create a file there. Did that via PCManFM, then tried above command again. 

`systemctl status redis-server` and `journalctl -u redis-server` still giving error messages, the latter about permission denied. So I chowned the permissions to 777 on the logfile using `chmod 777 /var/log/redis/redis-server.log` and now journalctl and systemctl outputs

> root@DietPi:~# journalctl -u redis-server

> -- Logs begin at Tue 2019-12-31 10:31:48 GMT, end at Tue 2019-12-31 10:34:51 GMT
> . --
> Dec 31 10:32:37 DietPi systemd[1]: Starting Advanced key-value store...
> Dec 31 10:32:38 DietPi systemd[1]: Started Advanced key-value store.
> root@DietPi:~# systemctl status redis-server
> ● redis-server.service - Advanced key-value store
>    Loaded: loaded (/lib/systemd/system/redis-server.service; disabled; vendor pr
> eset: enabled)
>   Drop-In: /etc/systemd/system/redis-server.service.d
>            └─override.conf
>    Active: active (running) since Tue 2019-12-31 10:32:38 GMT; 2min 4
> 9s ago
>      Docs: http://redis.io/documentation,
>            man:redis-server(1)
>  Main PID: 736 (redis-server)
>    Memory: 3.1M
>    CGroup: /system.slice/redis-server.service
>            └─736 /usr/bin/redis-server 127.0.0.1:6379
> 
> Dec 31 10:32:37 DietPi systemd[1]: Starting Advanced key-value store...
> Dec 31 10:32:38 DietPi systemd[1]: Started Advanced key-value store.

dankerthrone, 2019-12-31:
So I guess technically this issue can be closed now, although I wonder why I ran into this error in the first place.

Joulinar, 2019-12-31:
@dankerthrone 
pls can you do one more check under which user Redis is running

`ps -ef|grep redis`

dankerthrone, 2019-12-31:
The output for that is

> redis    10974     1  0 11:20 ?        00:00:09 /usr/bin/redis-server 127.0.0.1:6379
> root     11838  1636  0 11:36 pts/1    00:00:00 grep redis

Joulinar, 2019-12-31:
Ok the service is running with the correct user. Can you check who is owning the redis log dir?

MichaIng, 2020-01-01:
Resolved with: https://github.com/MichaIng/DietPi/commit/0a7bf0a250e40510533200f85537f4d73c414577
Changelog: https://github.com/MichaIng/DietPi/commit/fee9e80e032eefd2cae28c55a55f28e5d59610c5

This was a problem in the past on Debian Jessie already: The native APT package failed somehow to pre-create the logging dir and file that is used by default, hence it's doomed to fail. I could not replicate on Stretch/Buster/Bullseye but file logging is anyway nothing really useful here: Often the systemd unit fails for some reason already, then the log file does not give any related hint but `journalctl -u redis-server` must be checked. So it anyway make sense to log everything in one place that is active and in RAM by default, as systemd-journald/journalctl, which reduces overhead as well.

To resolve hence:
```
G_CONFIG_INJECT 'loglevel[[:blank:]]' 'loglevel warning' /etc/redis/redis.conf
G_CONFIG_INJECT 'logfile[[:blank:]]' 'logfile ""' /etc/redis/redis.conf
G_CONFIG_INJECT 'syslog-enabled[[:blank:]]' 'syslog-enabled yes' /etc/redis/redis.conf
G_CONFIG_INJECT 'always-show-logo[[:blank:]]' 'always-show-logo no' /etc/redis/redis.conf
systemctl restart redis-server
```

If one insists in file logging, skip/revert the above and instead do:
```
mkdir  -p /var/log/redis
> /var/log/redis/redis-server.log
chown -R redis:redis /var/log/redis
systemctl restart redis-server
```

closed 2020-01-01 by MichaIng