MichaIng/DietPi #3299: CPU Governor - Permission denied

bbsixzz, 2020-01-02:
Creating a bug report/issue

Required Information

DietPi version |
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=27
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version | buster 10.2

Kernel version | 
Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

SBC device | RPi 4 Model B (armv7l)

Power supply used | 3A Official

SDcard used | SanDisk ultra

Extra details

My CPU is locked at 600 MHz with Governor powersave.

[image]
[image]
[image]

I updated EEPROM after updating to DietPi 6.27.2; maybe related?

[image]

Joulinar, 2020-01-02:
hi,
as far as I can see some coding changed in `/DietPi/dietpi/func/dietpi-set_cpu` with new version v6.27.2. So I guess it has noting to do with the EEPROM update and might be a typo/bug in the scripting.

bbsixzz, 2020-01-02:
That did the trick, cheers!

[image]

On a tangent, is it supposed to be 1500 MHz when Governor is ondemand?

There is virtually no load.

[image]

Joulinar, 2020-01-02:
it should go down after a while

bbsixzz, 2020-01-02:
Well maybe this it the culprit?

`[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script, due to the processing required to run it.
`

It still hasn't gone down, did a reboot as well.

[image]

MichaIng, 2020-01-02:
@bbsixzz 
Indeed the script itself should raise the load for a tiny time which is enough to raise frequencies for 1 second (default before or goes down after load).

To check via less overhead:
```
cat /sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq
```

bbsixzz, 2020-01-02:
cpuinfo_cur_freq result: 1500000
scaling_cur_freq result: 600000

I don't think it's a big deal, but it seems locked at 1500 MHz.

Is the RPi4 really so slow that the CPU script throttles all 4 cores like that?

Joulinar, 2020-01-02:
For me it looks like using `ondemand` will look me at 1500 MHz and it's not stepped down. CPU temp is below 40'C. So it's doing nothing.
```
root@DietPi4:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temperature  |     39'C : 102'F (Cool runnings)
 Governor     |     ondemand
 Throttle up  |     50% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1500 MHz      600 MHz    1500 MHz
 CPU1         |      1500 MHz      600 MHz    1500 MHz
 CPU2         |      1500 MHz      600 MHz    1500 MHz
 CPU3         |      1500 MHz      600 MHz    1500 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script,
due to the processing required to run it.
```

While using `conservative` seems working correctly. It's stepping up if needed and going down once done.
```
root@DietPi4:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temperature  |     50'C : 122'F (Running warm, but safe)
 Governor     |     conservative
 Throttle up  |     50% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      1500 MHz      600 MHz    1500 MHz
 CPU1         |      1500 MHz      600 MHz    1500 MHz
 CPU2         |      1500 MHz      600 MHz    1500 MHz
 CPU3         |      1500 MHz      600 MHz    1500 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script,
due to the processing required to run it.

root@DietPi4:~#
root@DietPi4:~#
root@DietPi4:~#
root@DietPi4:~# cpu

 ─────────────────────────────────────────────────────
 DietPi CPU Info
 Use dietpi-config to change CPU / performance options
 ─────────────────────────────────────────────────────
 Architecture |     armv7l
 Temperature  |     38'C : 100'F (Cool runnings)
 Governor     |     conservative
 Throttle up  |     50% CPU usage

                 Current Freq    Min Freq   Max Freq
 CPU0         |      600 MHz      600 MHz    1500 MHz
 CPU1         |      600 MHz      600 MHz    1500 MHz
 CPU2         |      600 MHz      600 MHz    1500 MHz
 CPU3         |      600 MHz      600 MHz    1500 MHz

[ INFO ] DietPi-CPU_info | CPU current frequency, may be affected by this script,
due to the processing required to run it.
```

I used `dietpi-config: Stress Test` to simulate the load.

travisboss, 2020-01-03:
I am having a similar issue, the difference is I cannot change the governor to anything but powersave on the RPi 4 with the latest update.

bbsixzz, 2020-01-03:
@MichaIng just to be clear, you should investigate `ondemand ` as it doesn't seem to be working.

The original bug of "permission denied" is however fixed.

MichaIng, 2020-01-03:
@bbsixzz 
I am pretty sure that it works fine. Please check the following:
```
cat /sys/devices/system/cpu/cpu0/cpufreq/stats/time_in_state
```
It should show you how long (for current session) the CPU was in which state (only the two 600 and 1500 on RPi).

You can raise the CPU usage on which the freq is raised via up_treshold:
```
cat /sys/devices/system/cpu/cpufreq/ondemand/up_threshold
```
Its 50% by default on DietPi. When raising to e.g. 95% it might stay on lower freq during the commands.
```
echo 95 > /sys/devices/system/cpu/cpufreq/ondemand/up_threshold
```
Or via dietpi-config to be persistent.

bbsixzz, 2020-01-03:
Alright, after a reboot I got the following numbers which indicate everything works fine!

```
600000 3509
1500000 5709
```

Thank you for the education :]

MichaIng, 2020-01-03:
v6.28 is out, I mark this issue as closed.

closed 2020-01-03 by MichaIng