MichaIng/DietPi #3300: Can't create extraction directory when installing dietpi-nordvpn

dcallen7, 2020-01-02:
#### Details:
- Date           | Wed  1 Jan 21:12:07 PST 2020
- Bug report     | N/A
- DietPi version | v6.27.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | unzip -o 171.zip -d /etc/openvpn
- Exit code      | 2
- Software title | DietPi-Software

#### Steps to reproduce:
Install dietpi-nordvpn from dietpi-software

#### Expected behaviour:
It should complete installation successfully, including the unzipping of config files in /etc/openvpn.

#### Actual behaviour:
The installer can't seem to create an /etc/openvpn directory in which to extract the zip.

#### Extra details:
I have tried installing this with NO /etc/openvpn directory, and I tried creating one myself. I tried once with the dir owned by root, and another time with it owned by dietpi:dietpi with 775 permissions. It fails every time with what seems like such a simple step.

#### Additional logs:
```
Archive:  171.zip
checkdir:  cannot create extraction directory:  /etc/openvpn
           No such file or directory
```

MichaIng, 2020-01-02:
@dcallen7 
Many thanks for your report.

I was able to replicate. The directory exist and when I open a second SSH session and do:
```
cd /tmp/DietPi-Software
unzip -o 171.zip -d /etc/openvpn
```
it works perfectly fine...

MichaIng, 2020-01-02:
Ah I know it...
This one https://github.com/MichaIng/DietPi/commit/15a9b2c6a4548fdf94410e5f5d0cce2b33ce413a
causes here
```
[[ $target ]] && target="-d $target"
G_RUN_CMD unzip -o $file "$target"
```
the target variable to become a single argument instead of being split. This is great to treat paths with white spaces correctly, but now the `-d` is part of the path, although `-d` itself is removed by unzip correctly, but the white space is part of the path, AFAIK:
```
root@VM-Buster:/tmp/DietPi-Software# unzip -o 171.zip '-d /etc/openvpn'
Archive:  171.zip
checkdir:  cannot create extraction directory:  /etc/openvpn
           No such file or directory
root@VM-Buster:/tmp/DietPi-Software# unzip -o 171.zip -d' /etc/openvpn'
Archive:  171.zip
checkdir:  cannot create extraction directory:  /etc/openvpn
           No such file or directory
root@VM-Buster:/tmp/DietPi-Software# unzip -o 171.zip -d/etc/openvpn
Archive:  171.zip
  inflating: /etc/openvpn/ovpn_udp/al10.nordvpn.com.udp.ovpn
  inflating: /etc/openvpn/ovpn_udp/al11.nordvpn.com.udp.ovpn
...
```
- Luckily the white space is not required, hence easy to patch.

MichaIng, 2020-01-02:
I will patch this into master as well, it's small and important enough. To fix it now:
```
sed -i 's/target="-d \$target"/target="-d\$target"/' /DietPi/dietpi/dietpi-software
```

dcallen7, 2020-01-02:
> I will patch this into master as well, it's small and important enough. To fix it now:
> 
> ```
> sed -i 's/target="-d \$target"/target="-d\$target"/' /DietPi/dietpi/dietpi-software
> ```

So I just type this command into the terminal? Or am I supposed to do this while I'm installing the software?

MichaIng, 2020-01-03:
v6.28 is out, I mark this issue as closed.

closed 2020-01-03 by MichaIng