MichaIng/DietPi #3301: EXT4-fs error

svh1985, 2020-01-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
`
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
`
- Distro version | `stretch`
- Kernel version | `Linux RockPi 4.4.192-rockchip64 #1 SMP Tue Oct 8 18:39:24 CEST 2019 aarch64 GNU/Linux`
- SBC device | `Rock64 (aarch64)`
- Power supply used | Original Rock64 PSU
- SDcard used | 16GM EMMC

Hi there,

I keep getting EXT4-fs errors when I check the dmesg log. I think it's the ram disk not sure why im getting these errors. Is there a way to fix this?

```
[318719.785136] EXT4-fs error: 20 callbacks suppressed
[318719.785169] EXT4-fs error (device zram0): ext4_find_extent:916: inode #16: comm rs:main Q:Reg: pblk 25 bad header/extent: extent tree corrupted - magic f30a, entries 12, max 340(340), depth 0(0)
[318719.787192] EXT4-fs error (device zram0): ext4_ext_remove_space:2999: inode #16: comm rs:main Q:Reg: pblk 25 bad header/extent: extent tree corrupted - magic f30a, entries 12, max 340(340), depth 0(0)
```

Dietpi-Drive_manager:
```
┌───────────────────────────────────────────────┤ DietPi-Drive_Manager ├───────────────────────────────────────────────┐
│ Please select a drive to see available options.                                                                      │
│  - User data location: /mnt/exthd01/dietpi_userdata                                                                  │
│                                                                                                                      │
│                                                   ●─ mmcblk0 ──────────────────────────────────────────────●         │
│         /                                         : /dev/mmcblk0p1 | ext4 | Capacity: 15G | Used: 5.7G (40%)         │
│                                                   ●─ zram0 ────────────────────────────────────────────────          │
│         /var/log                                  : /dev/zram0 | ext4 | Capacity: 49M | Used: 4.7M (11%)             │
│                                                   ●─ sda ────────────────────────────────────────────────            │
│         /mnt/exthd01                              : /dev/sda1 | ext4 | Capacity: 1.4T | Used: 1.2T (89%)             │
│                                                   ●─ zram1 ────────────────────────────────────────────────          │
│         /mnt/e67411a6-5eef-41db-b471-5a29f7539e98 : /dev/zram1 | swap | Not mounted                                  │
│                                                   ●─ Global Options ───────────────────────────────────────●         │
│         Idle Spindown                             : Set a global idle duration, before drives power down             │
│                                                   ●─ Add / Refresh Drives ─────────────────────────────────●         │
│         Add network drive                         : Select to mount networked drives                                 │
│         Refresh                                   : Scan for recently added/removed drives                           │
│                                                                                                                      │
│                                                                                                                      │
│                                  <Ok>                                      <Exit>                                    │
│                                                                                                                      │
└──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

MichaIng, 2020-01-02:
@svh1985 
Many thanks for reporting. It's not the RAMdisk, but the conflicting zRam implementation of ARMbian. The forcefully enable this on every root package upgrade, which is why I want to drop it completely in the future. However there are some useful system configs contained as well.

The `dietpi-update` to v6.27 should solve it, it contains a patch to mask the services and a DPKG config to prevent it from installing the conflicting ARMbian package files: https://github.com/MichaIng/DietPi/blob/master/dietpi/patch_file#L2200-L2234

MichaIng, 2020-03-13:
I mark this as closed, solution has already been applied via v6.27

closed 2020-03-13 by MichaIng