MichaIng/DietPi #3304: cannot overwrite non-directory '/var/www/phpmyadmin' with directory pMyAdmin-5.0.0-english'

Nightliss, 2020-01-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
```
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=27
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `buster`
- Kernel version | `Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`
- SBC device | `RPi 4 Model B (armv7l)
` or (EG: RPi3)
- Power supply used | Official
- SDcard used | SanDisk 64GB

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `1b874344-513d-4ce2-82aa-c6e56620d808`

#### Steps to reproduce

1. Tried to update from v6.26 | After error above removed phpmyadmin via dietpi-software and tried to install again - same error

#### Expected behaviour

phpmyadmin should install without error

#### Actual behaviour

phpmadmin fails with `cannot overwrite non-directory '/var/www/phpmyadmin' with directory
pMyAdmin-5.0.0-english`

Lighthttd fails to start because of 50-myphpadmin.conf doesnt exist

#### Extra details

[image]
[image]

I tried to sent report from the error-handler but terminal closed after it did something. Dunno if actually sent anything. First time doing a report, also attached bug report ID, maybe report went through

MichaIng, 2020-01-02:
@Nightliss 
Many thanks for your report.

Hmm strange, I tried fresh install an reinstall and both works very well, hence at least no syntax bug.
It should no be related, but could you try the following:
```
sed -i 's/target="-d \$target"/target="-d\$target"/' /DietPi/dietpi/dietpi-software
dietpi-software reinstall 90
```

Nightliss, 2020-01-03:
@MichaIng 
[image]

Tried to install again 
[image]

but no dice :/

Joulinar, 2020-01-03:
Hi,
just a stupid thing but if I read the error message above, it's stating `cannot overwrite non-directory with directory`. Means `/var/www/phpmyadmin` is a file. Isn't it?

Pls can you past output of 
```
cd /var/www/
ls -l```

Nightliss, 2020-01-03:
Hey @Joulinar, here is the result, seems it's symlinked?

[image]

Joulinar, 2020-01-03:
pls can you try to remove the symlink. it should be a directory
```
root@DietPiVM1:/var/www# ls -l
insgesamt 76
-rw-r--r--  1 root root 37984 Jan  3 11:04 apc.php
-rw-r--r--  1 root root  3388 Jan  3 10:59 index.lighttpd.html
-rw-r--r--  1 root root 22915 Jan  3 11:04 opcache.php
-rw-r--r--  1 root root    20 Jan  3 11:04 phpinfo.php
drwxr-xr-x 11 root root  4096 Jan  3 11:04 phpmyadmin
root@DietPiVM1:/var/www#
```

Nightliss, 2020-01-03:
@Joulinar,

that helped, phpmyadmin installed correctly. Somehow it didn't occur to me to try a remove the symlink. Many thanks for help :)

PS: Had no idea that overwriting a symlink would be a problem for the os.

MichaIng, 2020-01-03:
@Nightliss 
Ah you're right. In case of Nginx and Lighttpd we previously placed symlinks and I guess the installer does not correctly remove them, but only directories. I'll fix that and release a small hotfix release v6.28, as there have been some other small bugs appeared (small code changes, easy to fix).

MichaIng, 2020-01-03:
v6.28 is out, I mark this issue as closed.

closed 2020-01-03 by MichaIng