MichaIng/DietPi #3305: Rockpi4 wont boot DietPi after booting Debian and upgrading the SPI-flash

twerpyfie, 2020-01-02:
Creating a bug report/issue
Required Information

    DietPi version |
    Distro version | Buster
    Kernel version |
    SBC device | ROCKPi4-ARMv8 4GB
    Power supply used | 5V 3A
    SDcard used | SanDisk ultra 64GB

Additional Information

I cant boot up after a fresh flash to the SD-card
I used Rufus, Win32DiskImager, and Etcher
I dont get a picture when i connect the device to a monitor via HDMI (not with an adapter)
I dont see the device in my router when its connected to my LAN
Steps to reproduce

Flash rockpi4-debian-stretch-desktop on SD-card. Boot up and apt-get update && apt-get upgrade
(i dont know if this is the problem:) Type "y" at the question:

"Step one: upgrade bootloader on SPI Flash
One boot device, SPI Flash, is found, would you like to upgrade bootloader on it?
The installation would cost about seven minutes."

shut down the device

take out SD
flash Dietpi from https://dietpi.com/ to Sd
put it back and nothing works
Expected behaviour

Dietpi should boot up and show up in my router or show something on my monitor
Actual behaviour

The green LED lights up constantly
no picture on monitor, not showing up in router.
Extra details

It worked a few days ago. Then i used rockpi4-debian-stretch-desktop with the SD-card booted up and upgradet the bootloader on SPI Flash. I reflashed Dietpi, nothing works now.

I still can Flash rockpi4-debian-stretch-desktop and it will boot up, but dietpi wont.

______________________________________________________________________________

I finally did it after going all the wrong ways. 

The easiest way is to delete the SPI-bootloader with the Debian you upgraded it with:

- 1. boot debian (i used "rockpi4-debian-stretch-desktop-arm64-20190730_2022-gpt")
- 2. sudo apt-get update
- 3. sudo apt-get upgrade
- 4. dont upgrade spi-bootloader, or do it, doesn't really matter
- 5. cd /usr/local/sbin/
- 6. sudo ./rockpi4b_erase_spi_flash.sh
- 7. type YES

And now you can "burn" a DietPi.img to an SD as usual and boot it.

However, I cannot say how it behaves with other storage media and hard disk partitioning.  I will have to try this in the future.

MichaIng, 2020-01-03:
For reference: https://github.com/MichaIng/DietPi/issues/3292

Most likely the SPI bootloader expects certain root/boot partition UUIDs or a certain partitioning in general (while the currently DietPi image, as based on ARMbian, comes with a single partition only), which does not match the DietPi SDcard image. I also see, the Debian image you used is GPT (?), while ours is MBR.

If one does not want to flash the Raxda Debian image to only erase the SPI flash, connecting PIN 23 and 25 disables the SPI flash: https://github.com/MichaIng/DietPi/issues/3292#issuecomment-569961704

Then one can run this script from DietPi, to erase the SPI bootloader which allows to remove the PIN 23-25 wire again.
@twerpyfie 
Could you paste the content of the `rockpi4b_erase_spi_flash.sh` script?
`rkdeveloptool` is required then as well, but maybe we can put everything together so one can use this to erase the SPI image from any currently booted system.

twerpyfie, 2020-01-03:
> > If one does not want to flash the Raxda Debian image to only erase the SPI flash, connecting PIN 23 and 25 disables the SPI flash: #3292 (comment)
> 
> Then one can run this script from DietPi, to erase the SPI bootloader which allows to remove the PIN 23-25 wire again.

I recommend to use this guide: [https://wiki.radxa.com/Rockpi4/dev/spi-install](url) 
at the very bottom

> @twerpyfie
> Could you paste the content of the rockpi4b_erase_spi_flash.sh script?
> rkdeveloptool is required then as well, but maybe we can put everything together so one can use this to erase the SPI image from any currently booted system.

@MichaIng 
Sure, but you don't need the rkdeveloptool for that as far as I know.

```
#!/bin/bash

set -eo pipefail

if [[ "$(id -u)" -ne "0" ]]; then
    echo "This script requires root."
    exit 1
fi

if ! which flash_erase &>/dev/null; then
    echo "Install mtd-utils with 'apt-get install mtd-utils'"
    exit 1
fi

echo "Doing this will overwrite data stored on SPI Flash"
echo "  and it will require that you use eMMC or SD"
echo "  as your boot device."
echo ""

while true; do
    echo "Type YES to continue or Ctrl-C to abort."
    read CONFIRM
    if [[ "$CONFIRM" == "YES" ]]; then
        break
    fi
done

if ! MTD=$(grep \"loader\" /proc/mtd | cut -d: -f1); then
    echo "loader partition on MTD is not found"
    return 1
fi

flash_erase "/dev/$MTD" 0 0

echo Done.

```

MichaIng, 2020-01-03:
@twerpyfie 
Many thanks.

It's [mtd-utils](https://packages.debian.org/buster/amd64/mtd-utils/filelist) which provides the `flash_erase` command to clear SPI flash.