MichaIng/DietPi #3306: DietPi-update fails with SSL_ERROR_SYSCALL

Dmitry-Dee, 2020-01-02:
### DietPi-update fails with SSL_ERROR_SYSCALL in connection to dietpi.com:443

#### Required Information
- DietPi version | v6.25.3
- Distro version | Buster
- Kernel version | 4.19.57-v71
- SBC device | RPi4 Model B (armv71)
- Power supply used | Berryuk 5.1V 3A 
- SDcard used | Kingston class 10

#### Additional Information (if applicable)
- Software title | dietpi-update
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? yes

#### Steps to reproduce
connected with eth0, can ssh to RPi, time synced
Fresh install from dietpi.com/downloads/images/DietPi_RPi-ARMv6-Buster.7z

#### Expected behaviour
should compete required update and installation

#### Actual behaviour
fails to update from any mirror

#### Extra details
Exit code 4
Temporary failure in name resolution
wget: unable to resolve host address raspbian.raspberrypi.org
curl: (28) Operation timed out after 300541 milliseconds with 0 out of 0 bytes received
curl: (35) OpenSSL SSL_connect: SSL_ERROR_SYSCALL in connection to dietpi.com:443
Unable to access any update server
NTPD: time sync | completed
Root access verified
RootFS R/W accesss verified
curl: (6) could not resolve host: raw.githubusercontent.com

Joulinar, 2020-01-02:
looks like your RPi is not able to connect to the Internet as you have DNS resolution issues like `unable to resolve host address raspbian.raspberrypi.org`. Can you try to ping google.com or another web site? For me it's working fine

```
root@DietPi4:~# ping raspbian.raspberrypi.org
PING mirrordirector.raspbian.org (93.93.128.193) 56(84) bytes of data.
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=1 ttl=55 time=34.9 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=2 ttl=55 time=35.9 ms
64 bytes from 93.93.128.193 (93.93.128.193): icmp_seq=3 ttl=55 time=33.8 ms
```

Dmitry-Dee, 2020-01-03:
It pings Google fine, and DietPi's connection test to http://google.com works, however test with secure protocol (https://google.com) fails

Joulinar, 2020-01-03:
But this seems to be a local issue because for me it's working fine

`[  OK  ] DietPi-Config | Connection test: https://google.com`

As well a survey connection test is working well
```
root@DietPi:~# dietpi-survey 1
[  OK  ] DietPi-Survey | Connection test: ssh.dietpi.com
[  OK  ] DietPi-Survey | Successfully sent survey data
```

MichaIng, 2020-01-03:
@Dmitry-Dee 
As Joulinar mentioned, your DNS server seems to work do its job correctly. This should not be related to SSL, but to be true, could you run those and paste result:
```
wget --spider http://google.com
wget --spider https://google.com
```

Are you using DHCP or static IP for your DietPi device? In case try to switch to static IP and set an external DNS server explicitly, e.g. `1.1.1.1` Cloudflare DNS.

Dmitry-Dee, 2020-01-04:
# wget --spider http://google.com                                      Spider mode enabled. Check if remote file exists.
--2020-01-03 21:41:33--  http://google.com/
Resolving google.com (google.com)... 172.217.164.206, 2607:f8b0:400b:808::200e
Connecting to google.com (google.com)|172.217.164.206|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: http://www.google.com/ [following]
Spider mode enabled. Check if remote file exists.
--2020-01-03 21:41:33--  http://www.google.com/
Resolving www.google.com (www.google.com)... 172.217.1.164, 2607:f8b0:400b:809::2004Connecting to www.google.com (www.google.com)|172.217.1.164|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [text/html]
Remote file exists and could contain further links,
but recursion is disabled -- not retrieving.

# wget --spider https://google.com
Spider mode enabled. Check if remote file exists.
--2020-01-03 21:42:23--  https://google.com/
Resolving google.com (google.com)... 172.217.164.206, 2607:f8b0:400b:800::200e
Connecting to google.com (google.com)|172.217.164.206|:443... connected.
HTTP request sent, awaiting response... 301 moved permanently 
Location: https://www.google.com/ [following]
Spider mode enabled. Check if remote file exists.
--2020-01-03 21:42:35-- 
https://www.google.com/
Resolving www.google.com (www.google.com)... 172.217.1.164, 2607:f8b0:400b:802::2004
Connecting to www.google.com (www.google.com)|172.217.1.164|:443... connected.
GnuTLS: Error in the pull function.
Unable to establish SSL connection.

Joulinar, 2020-01-04:
ok HTTP is working fine but seems you have an issue connecting HTTPS because of this error message
```
GnuTLS: Error in the pull function.
Unable to establish SSL connection.
```

Are you behind a Proxy?

probably you can try to run the following to see if the certificate is pulled correctly
`openssl s_client -servername www.google.com -connect www.google.com:443`

Dmitry-Dee, 2020-01-04:
The result of this command shows Connected (00000003)

Dmitry-Dee, 2020-01-04:
I modified /etc/resolvconf.conf to uncomment nameserver 8.8.8.8, it started downloading some updates that it was failing to pull earlier, and prompted to restart Pi, and after restart the issue is back. Tried with static IP and Google DNS - same thing

Dmitry-Dee, 2020-01-04:
Is it possible to skip/postpone updates?

Joulinar, 2020-01-04:
@Dmitry-Dee 

> The result of this command shows Connected (00000003) 

pls can you past the whole output, not just the first line

> Tried with static IP and Google DNS - same thing

I don't think your issue is with DNS because DNS is resolved correctly
```
Resolving google.com (google.com)... 172.217.164.206, 2607:f8b0:400b:800::200e
Connecting to google.com (google.com)|172.217.164.206|:443... connected.
```
Your issue seems to be with https connection. 

For me `/etc/resolvconf.conf` looks like this
```
# Configuration for resolvconf(8)
# See resolvconf.conf(5) for details

resolv_conf=/etc/resolv.conf
# If you run a local name server, you should uncomment the below line and
# configure your subscribers configuration files below.
#name_servers=127.0.0.1

# Mirror the Debian package defaults for the below resolvers
# so that resolvconf integrates seemlessly.
dnsmasq_resolv=/var/run/dnsmasq/resolv.conf
pdnsd_conf=/etc/pdnsd.conf
unbound_conf=/var/cache/unbound/resolvconf_resolvers.conf
```

Dmitry-Dee, 2020-01-05:
Connected (00000003) - this is all it displays for openssl s_client -servername www.google.com -connect www.google.com:443
The command does not produce any other output, though seems stuck and does not return to the terminal prompt - I had to Ctrl+c to end it

Joulinar, 2020-01-05:
puhh this is far to less. It should looks like this

```
root@DietPiV1:~# openssl s_client -servername www.google.com -connect www.google.com:443
CONNECTED(00000003)
depth=2 OU = GlobalSign Root CA - R2, O = GlobalSign, CN = GlobalSign
verify return:1
depth=1 C = US, O = Google Trust Services, CN = GTS CA 1O1
verify return:1
depth=0 C = US, ST = California, L = Mountain View, O = Google LLC, CN = www.google.com
verify return:1
---
Certificate chain
 0 s:C = US, ST = California, L = Mountain View, O = Google LLC, CN = www.google.com
   i:C = US, O = Google Trust Services, CN = GTS CA 1O1
 1 s:C = US, O = Google Trust Services, CN = GTS CA 1O1
   i:OU = GlobalSign Root CA - R2, O = GlobalSign, CN = GlobalSign
---
Server certificate
-----BEGIN CERTIFICATE-----
>> Certificate removed to get it more readable <<
-----END CERTIFICATE-----
subject=C = US, ST = California, L = Mountain View, O = Google LLC, CN = www.google.com

issuer=C = US, O = Google Trust Services, CN = GTS CA 1O1

---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 2632 bytes and written 386 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
Server public key is 256 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
read:errno=0
root@DietPiV1:~#
```
There is definitely something wrong on your system and you are not able to pull SSL certificates. I guess you would need to find out what is blocking you to connect on port 443. It might be firewall or you are behind a proxy

MichaIng, 2020-01-05:
@Dmitry-Dee 
Hmm, either it is something with your local network/router setup, a proxy/VPN (e.g. configured in router) or the openssl or related packages are broken somehow.

You said that after manually setting `nameserver 8.8.8.8` in `/etc/resolv.conf` the firstrun setup finished? When setting static IP with Google DNS, the result should be exactly the same (although 8.8.4.4 added as well as second Google DNS IP), could you verify this? If indeed it works then temporarily I would try to reinstall related packages to rule out any related issue:
```
apt update
apt install --reinstall openssl gnupg resolvconf curl
```

Also check your `dmesg` output for any error messages (red lines), probably some file system corruption is the issue.

Dmitry-Dee, 2020-01-06:
apt update failed:
The repository 'https://archive.raspberrypi.org/debian buster Release' does not have a Release file.
I ran 'apt update --allow-releaseinfo-change', it is running through updates now, will see how it goes

Joulinar, 2020-01-06:
@MichaIng 
regarding the repository error. I have seen a similar issue reported on the forum 
https://dietpi.com/phpbb/viewtopic.php?f=11&t=7119

MichaIng, 2020-01-06:
@Joulinar 
Nope, this is not the same issue. The one from the forum is an expected one-time issue, when users never called `apt update` from official Debian Buster release until now (half a year), on a Buster image installed prior to official release (only VM images marked as stable, AFAIK, RPi and others were always marked as beta/testing until official Buster release). Hence the amount of users running now into this should be countable on one hand. However I'm prepared to prevent this issue in 1.5 years when Debian Bullseye release comes 😉.

@Dmitry-Dee 
> The repository 'https://archive.raspberrypi.org/debian buster Release' does not have a Release file

This error however is a different one, sounds like a temporary server issue, since the Release file of course exists: https://archive.raspberrypi.org/debian/dists/buster/Release
The `--allow-releaseinfo-change` should not have been the solution. `apt update` would, if the Buster testing => stable suite change would not have been accepted, interactively ask to accept it now 😉. However this is already done on our current RPi images.

But back to the originating SSL issue, did the reinstall help to now have the following produce the expected result?
```
openssl s_client -servername www.google.com -connect www.google.com:443
```

Dmitry-Dee, 2020-01-10:
I am still investigating why it didn't work with eth0, though as soon as I switched to WiFi, APT updated successfully and DietPi setup completed as well.

MichaIng, 2020-02-04:
I mark this issue as closed. Repo issue was most likely temporary, still not sure about the SSL error, but probably the last firmware release or some other package updates resolved a related issue meanwhile.

If same issue persists still after updating to lastest APT packages (apt update && apt full-upgrade) please reopen.

closed 2020-02-04 by MichaIng