MichaIng/DietPi #3307: 2x Backports Install zip? No zip? Please clarify

GvY85, 2020-01-03:
I updated to 6.27 1 hour after it was released.
Now i see there is some fix(es) out there?

There are now two close "Backport Download_Install zip to target fix " (#3302 & #3303) but i cant find the zip fix anywhere?

I did apply this fix:
sed -i 's/target="-d \$target"/target="-d\$target"/' /DietPi/dietpi/dietpi-software 

Is that all and will the rest be fixed in 6.28 or? Please clarify a bit

MichaIng, 2020-01-03:
@GvY85 
Those backports are included in current master/release zips. Yes not best practice as early updaters do not have those fixes now. If you have MOTD enabled in your DietPi terminal banner, those are applied for you as well. I just wanted to prevent as much users as possible from running into this bugs quickly.

However I'll wrap up a tiny v6.28 bugfix release now to include all those little fixes for all users and have a consistent code across systems again.

MichaIng, 2020-01-03:
v6.28 is out, I mark this issue as closed.

closed 2020-01-03 by MichaIng