MichaIng/DietPi #3312: Nextcloud | CSRF check failed when logging out

Borotes, 2020-01-03:
Hi guys,

I reinstalled Nextcloud with a fresh DietPi image (v6.27.2) on my Raspberry Pi 3B+, because I updated it to version 17.0.2 and nothing worked anymore.

I saw the problem thread and that you have fixed it.

With my clean install I am getting [this](https://i.imgur.com/ShKVTJO.jpg) error message, when logging out of my browser session.

Could it be related?

MichaIng, 2020-01-03:
@Borotes 
What has been fixed (worked around) is the bug with fresh install of Nextcloud 17.0.2. We install Nextcloud 17.0.1 now which can then be upgraded via internal updater.

But the CSRF error is a different bug on Nextcloud side, see: https://github.com/nextcloud/server/issues/17065

MichaIng, 2020-01-11:
Finally invested some time to debug together with Nextcloud devs. Solution is here: https://github.com/MichaIng/DietPi/commit/2a756ee8c160ea29ad07d0c44681489c0e4439f4
Will be implemented more elegant at a later date. The `url-normalize-required` setting of Lighttpd produces this particular logout issue, but the whole URL parsing options are dangerous for many apps, which do not expect this. As well PHP itself decodes the URL/query string automatically, hence with PHP it is all double-decoded, leading to `%2B` => `+` => `space`.

Lighttpd changelog: https://www.lighttpd.net/2019/5/27/1.4.54/
Docs: https://redmine.lighttpd.net/projects/lighttpd/wiki/Server_http-parseoptsDetails

closed 2020-05-03 by MichaIng