MichaIng/DietPi #3313: DietPi-Banner | Error if "CPU temp" and "MOTD" are not selected

Joulinar, 2020-01-03:
Hi,

I'm getting following error message as soon as `CPU temp` and `MOTD` are not selected within `dietpi-banner`

`/DietPi/dietpi/func/dietpi-banner: line 191: G_TERM_CLEAR: command not found`

It seems the issue was introduced with v6.27.2 as I can reproduce it on my VM as well as on the RPi4B running v6.28 but it's not visible on my RPi3B+ running v6.26.3

**RPi4B**
```
login as: root
root@192.168.0.12's password:
/DietPi/dietpi/func/dietpi-banner: Zeile 191: G_TERM_CLEAR: command not found
 ─────────────────────────────────────────────────────
 DietPi v6.28.0 : 23:45 - Fr 03.01.2020
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (armv7l)
 - Uptime : up 3 hours, 3 minutes
 - LAN IP : 192.168.0.12 (eth0)
 - Info Text : !!! NEW RPi4 System !!!
 ─────────────────────────────────────────────────────
```
**RPi3B+**
```
login as: root
root@192.168.0.11's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v6.26.3 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi 3 Model B+ (armv7l)
 - Uptime : up 10 hours, 49 minutes
 - LAN IP : 192.168.0.11 (eth0)
 - Info Text : !!! PRODUCTION SYSTEM !!!
 ─────────────────────────────────────────────────────
```

As soon as `CPU temp` or `MOTD` are selected, the issue is gone
```
login as: root
root@192.168.0.12's password:
 ─────────────────────────────────────────────────────
 DietPi v6.28.0 : 23:53 - Fr 03.01.2020
 ─────────────────────────────────────────────────────
 - Device model : RPi 4 Model B (armv7l)
 - Uptime : up 3 hours, 11 minutes
 - CPU temp : 40'C : 104'F (Optimal temperature)
 - LAN IP : 192.168.0.12 (eth0)
 - Info Text : !!! NEW RPi4 System !!!
 ─────────────────────────────────────────────────────
```

Looks like you only calling `dietpi-globals` if one of these 2 options is selected.
`[[ ${aENABLED[2]} != 1 && ${aENABLED[12]} != 1 || $G_PROGRAM_NAME ]] || . /DietPi/dietpi/func/dietpi-globals`

MichaIng, 2020-01-03:
@Joulinar 
Many thanks for your report, indeed a bug.

Hmm, that new function assures that previous command outputs are preserved in scrollback buffer since the previous `print '\ec'` purges current screen content on most terminal types. Not yet sure how to handle best. Simplest would be to load DietPi-Globals always, but since it is a bid heavy, I'd like to avoid it if not required 🤔.

Joulinar, 2020-01-03:
I just came across by accident because usually I deselect MOTD and on my VM's I don't have any CPU information. I guess majority of people will have CPU or MOTD selected because usually you will have some CPU information from physical devices. Therefore `dietpi-globals` will be called at this point anyway. Now it would be good to have some statistics which options are used the most, to be able to decide how to deal with it 😆 

however for me it's working if I switch to this
`[[ $G_PROGRAM_NAME ]] || . /DietPi/dietpi/func/dietpi-globals`

MichaIng, 2020-01-04:
@Joulinar 
Yeah that means globals are always leaded. Okay I'll add it the way that this is done which simplifies the code as well. For next release I plan do increase globals load performance anyway by making all scripts source-able, hence no mawk/grep/sed required to read settings from our own files.

MichaIng, 2020-01-04:
Resolved for v6.29 with: https://github.com/MichaIng/DietPi/commit/c77812d8363b58ffe90b5b083687cd7ee14ec280
Changelog: https://github.com/MichaIng/DietPi/commit/8d6cbab03623de8db8cca1f451c522847d845d15
Hacked in a fix via MOTD:
cat /var/www/motd
```
[[ -w '/DietPi/dietpi/func/dietpi-banner' ]] && sed -i '/G_TERM_CLEAR$/s/$/ 2> \/dev\/null || printf "\\ec"/' /DietPi/dietpi/func/dietpi-banner
motd='★☆ ✿¸.•*¨★*☆*★`*•..¸✿ ☆¯`★
          ★[̲̅̅H̲̅][̲̅̅A̲̅][̲̅̅P̲̅][̲̅̅P̲̅][̲̅̅Y̲̅]★[̲̅̅N̲̅][̲̅̅E̲̅][̲̅̅W̲̅]★[̲̅̅Y̲̅][̲̅̅E̲̅][̲̅̅A̲̅][̲̅̅R̲̅]★
          *”˜˜”*°•.♥ღϠ₡ღ♥♥ღϠ₡ღ♥♥ღϠ₡ღ*”˜”*'
```
- Easy safe one which does least changes/overhead, only suppressing the error and calls the old screen clear method in case.

closed 2020-05-03 by MichaIng