MichaIng/DietPi #3314: dietpi odroid c2 (pivpn issue )i think

drieksje, 2020-01-04:
hi, new here (complete noob on linux)
i installed dietpi on a odroid c2, after installing pivpn dietpi cannot access the revis server anymore, so no software can be installed! can enybody help please?#### Details:
- Date           | Sat  4 Jan 12:07:35 CET 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (Fourdee/master)
- Image creator  | DietPi Core Team
- Pre-image      | Meveric
- SBC device     | Odroid C2 (aarch64) (ID=12)
- Kernel version | Linux DietPi 3.16.57+ #1 SMP PREEMPT Sun Aug 19 15:27:56 CEST 2018 aarch64 GNU/Linux
- Distro         | stretch (ID=4)
- Command        | systemctl restart redis-server
- Exit code      | 1
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Job for redis-server.service failed because the control process exited with error code.
See "systemctl status redis-server.service" and "journalctl -xe" for details.

MichaIng, 2020-01-04:
@drieksje 
Many thanks for your report.

Could you run the following command in console and past its output here:
```
journalctl -u redis-server
```

drieksje, 2020-01-04:
here is the log-- Logs begin at Thu 2016-11-03 18:16:42 CET, end at Sat 2020-01-04 13:39:21 CET
. --
jan 04 11:06:19 DietPi systemd[1]: Starting Advanced key-value store...
jan 04 11:06:20 DietPi run-parts[13058]: run-parts: executing /etc/redis/redis-s
erver.pre-up.d/00_example
jan 04 11:06:20 DietPi run-parts[13064]: run-parts: executing /etc/redis/redis-s
erver.post-up.d/00_example
jan 04 11:06:20 DietPi systemd[1]: Started Advanced key-value store.
jan 04 11:13:31 DietPi systemd[1]: Stopping Advanced key-value store...
jan 04 11:13:31 DietPi run-parts[26706]: run-parts: executing /etc/redis/redis-s
erver.pre-down.d/00_example
jan 04 11:13:32 DietPi run-parts[26716]: run-parts: executing /etc/redis/redis-s
erver.post-down.d/00_example
jan 04 11:13:32 DietPi systemd[1]: Stopped Advanced key-value store.
jan 04 11:13:51 DietPi systemd[1]: Starting Advanced key-value store...
jan 04 11:13:51 DietPi run-parts[30952]: run-parts: executing /etc/redis/redis-s
erver.pre-up.d/00_example
jan 04 11:13:52 DietPi redis-server[30958]: *** FATAL CONFIG FILE ERROR ***
jan 04 11:13:52 DietPi redis-server[30958]: Reading the configuration file, at l
ine 1053
jan 04 11:13:52 DietPi redis-server[30958]: >>> 'always-show-logo no'
jan 04 11:13:52 DietPi redis-server[30958]: Bad directive or wrong number of arg
uments

MichaIng, 2020-01-04:
@drieksje 
Dammit, this setting was introduced with Redis 4 which is not yet shipped with Debian Stretch:
- https://github.com/antirez/redis/blob/4.0/00-RELEASENOTES
- https://packages.debian.org/stretch/redis-server

Please do:
```
sed -i '/always-show-logo/d' /etc/redis/redis.conf
systemctl restart redis-server
```

MichaIng, 2020-01-04:
Fixed with: https://github.com/MichaIng/DietPi/commit/357c929b46882cc6eababb72f019b4e4502259bb
Changelog: https://github.com/MichaIng/DietPi/commit/ecec709856625d8be7aa1f81f8d760f1d1aaf5ec
Fixed on live systems via MOTD:
```
root@dietpi:~# cat /var/www/motd
...
[[ $G_DIETPI_VERSION_SUB == 2[78] && $G_DISTRO == 4 && -w '/DietPi/dietpi/dietpi-software' ]] && sed -i '/always-show-logo/d' /DietPi/dietpi/dietpi-software
...
```

drieksje, 2020-01-04:
hi, back again
thanks for the quick replay but still no luck with the commands you give me

drieksje, 2020-01-04:
hi Michalng,
so i enterred the commands in putty, so nothing happends, then tried to install kodi and jackpot, many thanks , thanks.....................................

MichaIng, 2020-01-05:
@drieksje 
Jep sorry, the commands did only fix the bug in `dietpi-software` but did not include the reinstall. Glad that it worked.

drieksje, 2020-01-05:
hi, back again,
so i have VPN but i have to use username and password, so is there a openvpn client possible in dietpi-debian?????NEEDED badly,ISP is blocking torrents
thanks in advance

MichaIng, 2020-01-05:
@drieksje 
Of course, you can install OpenVPN the regular way as on default Debian and invoke e.g. some .ovpn file given by your provider. If provided by your VPN provider, WireGuard can be installed via dietpi-software as well and configured (meaning left unconfigured) to be used as client with a config file by your provider.

If you use or want to use NordVPN, you can as well install DietPi-NordVPN, which provides some easy-to-use GUI for server selection, credentials input etc.

closed 2020-05-03 by MichaIng