MichaIng/DietPi #3315: DietPi-Software | OctoPrint | Can't access backups

ModischFabrications, 2020-01-04:
### Creating a bug report/issue

Accessing backups created from Octoprint is impossible due to "data/" being located in /root/*. This makes automated backups from cron or similar difficult and dangerous, as it requires using root cron tabs. 

#### Additional Information (if applicable)
- Software title | OctoPrint
- installed freshly, imported backup
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

https://docs.octoprint.org/en/latest/bundledplugins/backup.html
```
OCTOPRINT_BACKUP_DIR="/root/.octoprint/data/backup/"
1. octoprint plugins backup:backup
2. cp -rfL "$OCTOPRINT_BACKUP_DIR" "$BACKUP_DATA_DIR"
```

#### Expected behaviour
Successful copy of backup files to designated folder. 

#### Actual behaviour
`cp: cannot stat '/root/.octoprint/data/backup/': Permission denied`

#### Extra details
This problems stems in the fact that the application was installed into a folder with "root" as owner. 

[The documentation](https://dietpi.com/phpbb/viewtopic.php?f=8&t=5&p=7958#p7958) is not that helpful, as the mounted directory has nothing to do with the necessary "data/" directory.

I created an issue to pass a path (https://github.com/foosel/OctoPrint/issues/3415) in the OctoPrint Repository.

Somewhat related to https://github.com/MichaIng/DietPi/issues/854.

MichaIng, 2020-01-04:
@ModischFabrications 
Many thanks for your report. I'll have a look, it would be anyway better to have it run as own user with own dedicated home dir, as long as required permissions can be granted via system groups.

MichaIng, 2020-08-13:
Okay, changing the backup path is now supported: https://github.com/OctoPrint/OctoPrint/issues/3415#issuecomment-646026215
Works great, however this is not a nice long-term solution for our implementation since all the other data+configs remain in `/root/.octoprint`.

When running as own user the issue is solved a better way. I'm just thinking where to put user home. `/mnt/dietpi_userdata/octoprint` would be an idea, but `.octoprint` data/config dir is not ignored by `git` < `.gitignore`. Best would be to have the git/install dir in `/opt/octoprint` and the user home at `/mnt/dietpi_userdata/octoprint` so that data would be stored in `/mnt/dietpi_userdata/octoprint/.octoprint`. The problem is that the CLI needs to be executed as `octoprint` user as well.  The issue is the same with some other software titles that run as system service (with e.g. web UI) and have a CLI as well.

Handy solution for this: `alias octoprint='sudo -u octoprint octoprint'`
Can be added to `/etc/bashrc.d/dietpi-octoprint.sh` so that octoprint binary calls from any user end up calling it as user `octoprint`. If it is wanted to run it indeed as a different user, one would need to call the binary with full path directly: `/usr/local/bin/octoprint`

@ModischFabrications 
Any suggestions about nice clean setup to combine non-root system service with flexible UNIX user CLI calls? I always wanted to find a generic approach for this for all our software implementations. Of course one could always run the service as own UNIX login user but that IMO does not satisfy general security considerations where you want to separate long-term running (system-)services from login users.

ModischFabrications, 2020-08-13:
Is there any configuration that is not inside the backup but still relevant? From my understanding backing up contains everything relevant and can be pointed wherever we want , so the "real" path is not relevant for most cases. 

To be honest, I am at most a user, if not a complete amateur regarding UNIX scripting. I think the alias works, but it's still kind of a workaround. I'm afraid I can't think of a better way than that, your guess is as good (or probably even better) than mine.

MichaIng, 2020-08-13:
> Is there any configuration that is not inside the backup but still relevant?

I bet all configuration that you do on web UI is stored in `/root/.octoprint`. Probably this is not something that one wants to access manually often, but generally such data produced by a system service should not be stored in the root users home dir, IMO.

Same is basically true for the CLI. I guess you can do most things usually needed in web UI, but if it has a CLI already, its nice to reduce the overhead of using it to a minimum. I just remember ownCloud/Nextclouds `sudo -u www-data php /var/www/nextcloud/occ` that on DietPi is `occ`/`ncc` via alias. But at least this command fails with reasonable message when being invoked with a different user then the webserver runs as.

MichaIng, 2020-09-18:
I implemented the above ideas: https://github.com/MichaIng/DietPi/commit/c97d02de75aa1396b6c60332469db4995c407421
Migration and update needs to be done on DietPi update, working on this now.

closed 2020-10-17 by MichaIng

ModischFabrications, 2020-12-13:
Sorry for necroing, but I think migration does not work properly. The old installation does not seem to be copied over properly. 

My steps: 
1. dietpi-update
2. did not succeed, futures version missing
3. exit and reboot
4. octoprint has been reset, completely missing everything

Looking into the old /root/.octoprint everything seems to still be there, so I will try to recover manually. Still, this issue might hit more people and it's scary to lose the whole installation.

MichaIng, 2020-12-13:
Sadly, quickte shortly after having this released, OctoPrint release a new major version (1.5) with a complete update method rework that practically broke the internal updater in our implementation: #3940

> 2. did not succeed, futures version missing

What do you mean by this? `dietpi-update` did not succeed or the included OctoPrint reinstall?

> Looking into the old /root/.octoprint everything seems to still be there

Then indeed the the update itself stopped before even reaching the OctoPrint reinstall since that directory is moved to `/mnt/dietpi_userdata/octoprint/.octoprint`: https://github.com/MichaIng/DietPi/blob/master/dietpi/patch_file#L2534

OctoPrint is reinstalled another time with this update, migrating `/mnt/dietpi_userdata/octoprint/.octoprint` or `/root/.octoprint` if the first does not exist. It should be a long term solution then so that installs+reinstalls with dietpi-software work fine with the OctoPrint internal updater an plugin manager, but I found just recently that there are still bugs in OctoPrint that break the way we configure it... In the end it might be impossible to have a global OctoPrint instance (that we and everyone can install and update via `sudo pip3 install -U OctoPrint`) that is able to update itself and install plugins, when running as "octoprint" service user. I'll see if a solution can be found until DietPi release, otherwise it will be a local install that can only be maintained via web UI or `sudo -u octoprint pip3 --user install -U ...`, not great but that is currently the only invocation that can reliably match what OctoPrint web UI does internally.

ModischFabrications, 2020-12-13:
> What do you mean by this? dietpi-update did not succeed or the included OctoPrint reinstall?

Probably Octoprint, but I don't recall. I will simply refrain from updating until a solution is found and just stay on my current, half transferred setup. 

Thanks for the extensive explanation!