MichaIng/DietPi #3316: Jackett download issue

snoodler, 2020-01-04:
download stopped at 84% with this error:

2020-01-03 12:21:43 (6.52 KB/s) - Read error at byte 38593082/45736441 (Error in the pull function.). Retrying.

--2020-01-03 12:21:44-- (try: 2) https://github-production-release-asset-2e65be.s3.amazonaws.com/45195614/456e7f80-2d29-11ea-9430-f0240eb4472b?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20200103%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20200103T104507Z&X-Amz-Expires=300&X-Amz-Signature=5c487379b690f1774dd403b35e7aba55bf7950aff1400906bfddabe77f4bca23&X-Amz-SignedHeaders=host&actor_id=0&response-content-disposition=attachment%3B%20filename%3DJackett.Binaries.LinuxARM32.tar.gz&response-content-type=application%2Foctet-stream
Connecting to github-production-release-asset-2e65be.s3.amazonaws.com (github-production-release-asset-2e65be.s3.amazonaws.com)|52.216.206.3|:443... connected.
HTTP request sent, awaiting response... 403 Forbidden
2020-01-03 12:21:46 ERROR 403: Forbidden.

Joulinar, 2020-01-05:
Hi,
did you used `dietpi-software` to install jackett?? I checked it on my DietPiVM and it was working fine.

MichaIng, 2020-01-05:
@snoodler 
Many thanks for your report. For reference: https://github.com/MichaIng/DietPi/issues/3029#issuecomment-570365895

The following commands is what failed on Jackett archive download at 84%:
```
cd /tmp
wget https://github.com/Jackett/Jackett/releases/download/v0.12.1398/Jackett.Binaries.LinuxARM32.tar.gz
tar xf Jackett.Binaries.LinuxARM32.tar.gz --one-top-level=/opt
```

`wget` seems to do a second attempt against the redirected URL which is only allowed/valid for a single connection I guess, e.g. single request token or something like that. Hence the second failure is expected.

Searching around a bid leads to several different possible causes, e.g.: https://github.com/curl/curl/issues/1707 (declared as network-related issue)

It goes a bid in the same direction as: https://github.com/MichaIng/DietPi/issues/3306
Steps to debug and possibly solve would be similar:
```
dmesg # Check for any file system errors (red lines)
apt update
apt install --reinstall openssl gnupg wget
```
Verify that downloading the same resource from other machines on the same local network and gateway (router) work fine to rule out any network-related/proxy/VPN issues and such, e.g. you can simply open the URL above from a regular desktop browser and see if it downloads fine there.

snoodler, 2020-01-06:
Hi,
Just re tried to install with dietpi-software and this time I had no errors. Before it was taking a few hours to get the files from github , this time it did it in a few minutes. I guess something was wrong with the download and taking too long timed out the install.

Thanks for the help

closed 2020-01-06 by snoodler

MichaIng, 2020-01-06:
@snoodler 
Great to here. Yeah sometimes the servers or global CDNs simply have temporary issues.