MichaIng/DietPi #3317: DietPi-Login Failed

acryllect, 2020-01-05:
- Date           | Sun 14 Jul 17:26:01 BST 2019
- Bug report     | N/A
- DietPi version | v6.25.3 (MichaIng/master)
- Img creator    | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi  (armv7l) (index=-1)
- Kernel version | #1244 SMP Thu Jul 4 18:48:07 BST 2019
- Distro         | buster (index=5)
- Command        | Connection test: http://raspbian.raspberrypi.org/raspbian/
- Exit code      | 4
- Software title | DietPi-Login

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- I honestly have no idea, this is my first time using DietPi
- Probably its just meant to boot normally

#### Actual behaviour:

- I connect to my RBPi 4 with a fresh install of DietPi (after following the Starter Guide) through SSH
- It immediately starts to load DietPi-Login
- DietPi errors out
#### Extra details:

- ...

#### Additional logs:
```
Log file contents:
Spider mode enabled. Check if remote file exists.
--2019-07-14 17:25:37--  http://raspbian.raspberrypi.org/raspbian/
Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... failed: Connection timed out.
wget: unable to resolve host address ‘raspbian.raspberrypi.org’
```

acryllect, 2020-01-05:
I noticed that anything trying to access ```http://raspbian.raspberrypi.org/raspbian/raspbian ``` will not work. I can access it through my computer though.

MichaIng, 2020-01-05:
@acryllect 
Many thanks for your report.

If SSH works, the WiFi connection is fine at least.
> Resolving raspbian.raspberrypi.org (raspbian.raspberrypi.org)... failed: Connection timed out.

This means DNS resolving failed. As you set up a static IP address, which DNS server did you choose? Generally I would use DHCP first and apply the given IP, gateway and DNS server, received by DHCP, as static entries afterwards via dietpi-config > Network Settings: Adapters. However you can change this now and try some generic functional solution like Cloudflare or Google DNS. In most cases simply using the IP of your router/gateway works as well and best, since most routers allow to resolve DNS for the local network and have a cache etc included.

acryllect, 2020-01-05:
Also, tried modifying the timeout max time to 20 seconds and nothing has changed

acryllect, 2020-01-05:
My current DNS is 8.8.8.8. I don't fully understand what you mean. I could try changing it to 1.1.1.1 but if I run ```/etc/resolv.conf``` i get a permission denied so I'm guessing I'll have to manually change it in the SD card

acryllect, 2020-01-05:
Thanks for the suggestion, but changing the DNS (8.8.8.8 Google -> 1.1.1.1 Cloudflare) did not fix the issue.

Joulinar, 2020-01-05:
can you try to ping google.com? just to check if DNS resolution is working.

acryllect, 2020-01-05:
Tried pinging google.com, errored out the same way.
I'm using the root user

Joulinar, 2020-01-05:
using `root` should not prevent you from changing the file. Pls run `nano /etc/resolv.conf`

MichaIng, 2020-01-05:
@acryllect 
I meant to edit the file via `nano /etc/resolv.conf`, which must be done as root user, or `sudo nano /etc/resolv.conf`. However you tried that without success, as far as I understand.

> Edit: Just wanted to add that I did try making the max timeout 20 seconds as opposed to 5 but nothing changed.

Sorry, I've overseen the timeout, hence raising it from default 5 to 20 seconds was a good idea. However if ping fails as well, then most likely it is not a DNS issue but a connectivity issue (connection the www only, since connection to local network is fine). To test that, run: `ping 1.1.1.1`

acryllect, 2020-01-05:
If I run ```nano /etc/resolv.conf``` I only see one line which is ```nameserver 1.1.1.1```.
However when running ```ping 1.1.1.1``` I got the following error over and over until I just decided to hit ctrl+c (to interrupt the command):
```
root@DietPi:~# ping 1.1.1.1
PING 1.1.1.1 (1.1.1.1) 56(84) bytes of data.
From 192.168.1.20 icmp_seq=1 Destination Host Unreachable
From 192.168.1.20 icmp_seq=2 Destination Host Unreachable
From 192.168.1.20 icmp_seq=3 Destination Host Unreachable
etc... for all other icmp_seq=number
```
So this is the likely issue @MichaIng pointed out in the latest comment but as a new Pi user myself, I have close to no clue how to fix this.

Joulinar, 2020-01-05:
looks like your Gateway settings are not correct. Therefore your Pi is not able to connect to the internet. pls go to 

1. run `dietpi-config` on command line
2. Goto number 7 ` Network Options: Adapters`
3. Goto `WiFi`
4. Check your `Static Gateway` IP address. Usually it should point to your Internet Router
5. In your case it might be better die switch to DHCP in first place to get your system working. It can be reverted alter on.

acryllect, 2020-01-06:
### Issue has been fixed! What I did: (everything was done through SSH)
1. Ran ```dietpi-config```
2. Went to n. 7 ```Network Options: Adapters```
3. Went to ```WiFi```
4. Fixed my ```Static Gateway``` IP address (192.168.0.1 -> 192.168.1.1 which was my router's private IP, checked it before changing)
5. Hit ```Apply``` and restarted networking
6. Waited for completion, then rebooted
Didn't even have to try ```ping 1.1.1.1```, as soon as I connected to my Pi it started ```DietPi-Login``` which gave no error whatsoever. I'm now running a fully functional, minimal install of DietPi! Thank you @MichaIng and @Joulinar for helping me out :)

closed 2020-01-06 by acryllect

MichaIng, 2020-01-06:
@acryllect 
Great, sometimes it is that simple. Yeah this is why I always recommend to use DHCP first and apply received values as static IP afterwards, so one can be sure they are correct 😉.

acryllect, 2020-01-06:
Unfortunately I kind of had no option, I had to set a static IP in the dietpi first boot config file because I have a headless setup. I'm having other issues now but it's software related so we'll see if I can figure it out myself or I'll have to post another thread (hopefully not). Cheers!

Joulinar, 2020-01-06:
well there is no need for Static IP address, even on headless systems DHCP is working fine. You would just need to check on your DHCP Server/ Internet Router, which IP was assigned. Usually all my systems running in headless mode right from the beginning.

acryllect, 2020-01-06:
Yes, I was aware of that. Forgot to say that I cannot access my router's panel as my whole family has forgotten the password AND username. And we really don't want to reset.

Joulinar, 2020-01-06:
you could try running `nslookup dietpi` on your workstation computer to check the assigned IP address.

MichaIng, 2020-01-06:
Jep or check the assigned IPs on other hosts and test ping or SSH connection to IPs, counting up the last triple. Usually DHCP servers just use the one higher than the highest already reserved/assigned IP. However most importantly its working now 😄.

Joulinar, 2020-01-06:
@acryllect 
>I cannot access my router's panel as my whole family has forgotten the password AND username. And we really don't want to reset.

Not sure how old your internet router is or how long it is already running, but might be that your vendor has released some updates/security patches in meantime. So probably you should consider to reset to get it updated. 😉