MichaIng/DietPi #3318: Airsonic | Fails to start upon fresh install

tenryou-17, 2020-01-06:
#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | stretch
- Kernel version | Linux DietPi 4.14.66+ #1 SMP PREEMPT Thu Aug 23 05:59:33 UTC 2018 armv7l GNU/Linux
- SBC device | Odroid XU4
- Power supply used | 5V 4A US Plug
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
- Software title | Airsonic
- Was the software title installed freshly or updated/migrated? Freshly
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Install Airsonic via Dietpi-Software

#### Expected behaviour

- Airsonic service starts

#### Actual behaviour

- Airsonic service fails to start

#### Extra details

The following is a list of other dietpi-services on the system:

- proftpd      
- nfs-kernel-server      
- php7.3-fpm          
- lighttpd 
- minidlna     
- cron         
- dropbear                
- pihole-FTL 

The following is the status log for when the Airsonic service tries to start:

 ● airsonic.service - Airsonic Media Server (DietPi)                                                                  │          
          │    Loaded: loaded (/etc/systemd/system/airsonic.service; disabled; vendor preset: enabled)                           │          
          │    Active: failed (Result: exit-code) since Mon 2020-01-06 01:47:48 GMT; 3s ago                                      │          
          │   Process: 2095 ExecStart=/usr/bin/java -Xmx398m -Dairsonic.home=/mnt/dietpi_userdata/airsonic                       │          
          │ -Dserver.context-path=/airsonic -Dserver.port=8080 -jar /mnt/dietpi_userdata/airsonic/airsonic.war (code=exited,     │          
          │ status=1/FAILURE)                                                                                                    │          
          │  Main PID: 2095 (code=exited, status=1/FAILURE)                                                                      │          
          │                                                                                                                      │          
          │ Jan 06 01:47:48 DietPi systemd[1]: Started Airsonic Media Server (DietPi).                                           │          
          │ Jan 06 01:47:48 DietPi java[2095]: /usr/lib/jvm/java-8-openjdk-armhf/jre/lib/rt.jar: invalid LOC header (bad         │          
          │ signature)                                                                                                           │          
          │ Jan 06 01:47:48 DietPi java[2095]: Error: Could not find or load main class                                          │          
          │ org.springframework.boot.loader.WarLauncher                                                                          │          
          │ Jan 06 01:47:48 DietPi systemd[1]: airsonic.service: Main process exited, code=exited, status=1/FAILURE              │          
          │ Jan 06 01:47:48 DietPi systemd[1]: airsonic.service: Unit entered failed state.

Joulinar, 2020-01-06:
@tenryou-17 

I tried it on my VM and a fresh installation was working fine. Could be that there is something corrupt on your JAVA installation

 pls can you past the following as well 
```
journalctl -u airsonic
java -version
javac -version
```

MichaIng, 2020-01-06:
@tenryou-17 
Many thanks for your report.
> /usr/lib/jvm/java-8-openjdk-armhf/jre/lib/rt.jar: invalid LOC header (bad signature)

From what I found this file must be corrupt. Please try to reinstall it and check dmesg for file system errors, in case those were the initial reason:
```
apt install --reinstall openjdk-8-jre-headless
dmesg
```

tenryou-17, 2020-01-07:
@Joulinar Here is the response line-by-line:
```
--No entries--
```
```
openjdk version "1.8.0_232"
OpenJDK Runtime Environment (build 1.8.0_232-8u232-b09-1~deb9u1-b09)
OpenJDK Client VM (build 25.232-b09, mixed mode)
```
```
/usr/lib/jvm/java-8-openjdk-armf/jre/lib/rt.jar: invalid LOC header (bad signature)
Error: Could not find or load main class com.sun.tools.javac.Main
```

@MichaIng 
Java was reinstalled with no issues. Here is the output of dmesg (apologies for the length):

```
[    0.000000] Booting Linux on physical CPU 0x100
[    0.000000] Linux version 4.14.66+ (root@loki) (
gcc version 4.9.2 (Debian 4.9.2-10+deb8u1)) #1 SMP
PREEMPT Thu Aug 23 05:59:33 UTC 2018
[    0.000000] CPU: ARMv7 Processor [410fc073] revi
sion 3 (ARMv7), cr=10c5387d
[    0.000000] CPU: div instructions available: pat
ching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data ca
che, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Hardkernel O
droid XU4
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0xb6800000
[    0.000000] Samsung CPU ID: 0xe5422001
[    0.000000] On node 0 totalpages: 518656
[    0.000000] free_area_init_node: node 0, pgdat c
0d8e980, node_mem_map eee25000
[    0.000000]   Normal zone: 1728 pages used for m
emmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO ba
tch:31
[    0.000000]   HighMem zone: 322048 pages, LIFO b
atch:31
[    0.000000] Running under secure firmware.
[    0.000000] random: get_random_bytes called from
 start_kernel+0x8c/0x420 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu @eed65
000 s37568 r8192 d23872 u69632
[    0.000000] pcpu-alloc: s37568 r8192 d23872 u696
32 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[0] 4 [0] 5 [0] 6 [0] 7
[    0.000000] Built 1 zonelists, mobility grouping
 on.  Total pages: 516928
[    0.000000] Kernel command line: console=tty1 co
nsole=ttySAC2,115200n8 root=UUID=3032ee2a-4efc-40b6
-9c1f-2839a840144d rootwait ro fsck.repair=yes net.
ifnames=0 drm_kms_helper.edid_firmware=edid/1920x10
80.bin smsc95xx.macaddr=00:1e:06:61:7a:72 governor=
performance hdmi_tx_amp_lvl=31 hdmi_tx_lvl_ch0=3 hd
mi_tx_lvl_ch1=3 hdmi_tx_lvl_ch2=3 hdmi_tx_emp_lvl=6
 hdmi_clk_amp_lvl=31 hdmi_tx_res=0 HPD=true vout=hd
mi  s5p_mfc.mem=16M
[    0.000000] hdmi: using HDMI mode
[    0.000000] PID hash table entries: 4096 (order:
 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131
072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 6553
6 (order: 6, 262144 bytes)
[... 754 lines cut ...]
are load for s5p-mfc-v8.fw failed with error -2
[   19.454443] s5p_mfc_load_firmware:73: Firmware i
s not present in the /lib/firmware directory nor co
mpiled in kernel
[   19.608995] sd 0:0:0:0: Attached scsi generic sg
0 type 0
[   22.030799] IPv6: ADDRCONF(NETDEV_UP): eth0: lin
k is not ready
[   22.052489] r8152 6-1:1.0 eth0: carrier on
[   22.055316] IPv6: ADDRCONF(NETDEV_CHANGE): eth0:
 link becomes ready
[   27.987551] Installing knfsd (copyright (C) 1996
 okir@monad.swb.de).
[   28.168027] EXT4-fs (sda1): mounted filesystem w
ith ordered data mode. Opts: (null)
[   28.421117] NFSD: starting 90-second grace perio
d (net c0d83e80)
[138659.259406] nfsd: last server has exited, flush
ing export cache
```
After I reinstalled Java, Airsonic still failed to start. I reinstalled it, rebooted the XU4, and it seems to work now. 

When I looked at the log under the `service airsonic status` command, I could tell that there was a problem with Java. My question is how did you know to try reinstalling  `openjdk-8-jre-headless` specifically? Just for my own knowledge:)

Thanks!

Joulinar, 2020-01-07:
@tenryou-17 
> /usr/lib/jvm/java-8-openjdk-armf/jre/lib/rt.jar: invalid LOC header (bad signature)

This is indicating the issue of your JAVA installation. Basically your issue is described in this forum 
https://askubuntu.com/questions/869723/javac-exception-java-8

But it's not exactly the same as we did not use the mentioned JAVA package. But @MichaIng exactly knows which package to resetup as he is the main developer 😄 Next to that, the JAVA package is listed on the `dietpi-software` file

`packages="ca-certificates-java openjdk-$version-jre-headless openjdk-$version-jdk-headless"`

https://github.com/MichaIng/DietPi/blob/fea785249696f1d51ab21c5b046a471abff64c4a/dietpi/dietpi-software#L6447

tenryou-17, 2020-01-07:
Thanks @Joulinar ! Sometimes the most obvious solution is right in front of me. Also thanks @MichaIng !

closed 2020-01-07 by tenryou-17