MichaIng/DietPi #3319: Home Assistant silently died

Phil1988, 2020-01-06:
Hello,

I installed HA in middle of september 2019 and everything worked as I wanted.
I am using a vitual machine x86_64.

It worked for a few weeks, then I lost attention and maintaining because of less time :)

Since 3+ weeks I received a beacon from my android smartphone, that HA ist not availible.

A few days ago I had time and took a look and saw that HA is not reachable anymore via IP:8123.

I saw a note at the commandline that a new python version is nessesary, so I started it with an installation of a "precreated Python 3.6"
[image]

Which ended in installing Phython 3.8.0.
[image]

I also updated dietpi and today updated from
[ INFO ] DietPi-Update | Current version : v6.27.2
to
[ INFO ] DietPi-Update | Latest version  : v6.28.0

I also saw, that the installation and/or access notes have been changed here:
https://dietpi.com/phpbb/viewtopic.php?p=10416#p10416

Now, I am completely lost, why HA stopped working and is not reachable any more.
I even have no idea how I can access it or make it work again... or how do get the source of the issues.

The updates itself did run without any problems and restarted everything without any issues:
```
DietPi-Update
─────────────────────────────────────────────────────
 Phase: Installing new DietPi code

######################################################################## 100.0%
[  OK  ] DietPi-Update | Unpacking update archive
[  OK  ] DietPi-Update | Installing new DietPi scripts to RAMdisk
[  OK  ] DietPi-Update | Installing new DietPi system files to disk
[  OK  ] DietPi-Update | Setting execute permissions for DietPi scripts
[ SUB1 ] DietPi-Set_software > verify_dietpi.txt ()
[  OK  ] DietPi-Set_software | Connection test: https://raw.githubusercontent.co                                                                                                                                                             m/MichaIng/DietPi/master/dietpi.txt
[  OK  ] DietPi-Set_software | Downloading current dietpi.txt
[  OK  ] DietPi-Set_software | dietpi.txt verification completed
[  OK  ] verify_dietpi.txt  | Completed

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Applying incremental patches

[ INFO ] DietPi-Update | Current version : v6.27.2
[ INFO ] DietPi-Update | Latest version  : v6.28.0
[ INFO ] DietPi-Update | Patching 6.27 to 6.28
[  OK  ] DietPi-Update | Incremental patching to v6.28.0 completed

 DietPi-Update
─────────────────────────────────────────────────────
 Phase: Completed

[ INFO ] DietPi-Update | Current version : v6.28.0
[ INFO ] DietPi-Update | Latest version  : v6.28.0
[  OK  ] DietPi-Survey | Purging survey data
[  OK  ] DietPi-Update | Syncing new DietPi scripts to disk
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : redis-server
[  OK  ] DietPi-Services | restart : mariadb
[  OK  ] DietPi-Services | restart : apache2
[  OK  ] DietPi-Services | restart : mosquitto
[  OK  ] DietPi-Services | restart : node-red
[  OK  ] DietPi-Services | restart : blynkserver
[  OK  ] DietPi-Services | restart : home-assistant
[  OK  ] DietPi-Services | restart : cron
[  OK  ] DietPi-Services | ncc maintenance:mode --off
```

The only issue is, that HA is not accessable ;)
I hope you can help me, helping you, to help me :D

MichaIng, 2020-01-06:
@Phil1988 
Many thanks for your report.

Did you follow the instructions to wait for compete service start via htop, restart the service again and access the web UI one time?

If so please paste the output of: `journalctl -u home-assistant`

Phil1988, 2020-01-06:
Thanks for you reply @MichaIng !

Yes I did so. The update of python was days ago (but really needed a lot of time ;) ).

`journalctl -u home-assistant` gives me

```
--Logs begin at Fri 2019-12-27 19:45:01 CET, end at Mon 2020-01-06 19:20:01 CET. --
Jan 02 18:36:35 DietPi systemd[1]: Stopping Home Assistant (DietPi)...
Jan 02 18:36:36 DietPi systemd[1]: Stopped Home Assistant (DietPi).
Jan 02 18:39:31 DietPi systemd[1]: Started Home Assistant (DietPi).
Jan 02 18:39:34 DietPi homeassistant-start.sh[8103]: pyenv-virtualenv: prompt changing will be removed from future release. configure `export PYENV_VIRTUALENV_DISABLE_PROMPT=1' to simulate the behavior.
Jan 02 18:39:38 DietPi homeassistant-start.sh[8103]: 2020-01-02 18:39:38 ERROR (MainThread) [homeassistant.bootstrap] Error loading /home/homeassistant/.homeassistant/configuration.yaml:   in "/home/homeassistant/.homeassistant/configura
tion.yaml", line 14, column 11: Unable to read file /home/homeassistant/.homeassistant/frontend.yaml.
Jan 02 18:39:38 DietPi homeassistant-start.sh[8103]: 2020-01-02 18:39:38 INFO (MainThread) [homeassistant.core] Starting Home Assistant
Jan 02 18:39:38 DietPi homeassistant-start.sh[8103]: 2020-01-02 18:39:38 INFO (MainThread) [homeassistant.core] Timer:starting
Jan 02 18:40:14 DietPi systemd[1]: Stopping Home Assistant (DietPi)...
Jan 02 18:40:14 DietPi systemd[1]: Stopped Home Assistant (DietPi).
Jan 03 09:13:43 DietPi systemd[1]: Started Home Assistant (DietPi).
Jan 03 09:13:53 DietPi homeassistant-start.sh[10282]: pyenv-virtualenv: prompt changing will be removed from future release. configure `export PYENV_VIRTUALENV_DISABLE_PROMPT=1' to simulate the behavior.
Jan 03 09:14:07 DietPi systemd[1]: Stopping Home Assistant (DietPi)...
Jan 03 09:14:07 DietPi systemd[1]: Stopped Home Assistant (DietPi).
Jan 03 09:18:20 DietPi systemd[1]: Started Home Assistant (DietPi).
Jan 03 09:18:24 DietPi homeassistant-start.sh[24954]: 2020-01-03 09:18:24 INFO (SyncWorker_0) [homeassistant.config] Upgrading configuration directory from 0.98.5 to 0.103.5
Jan 03 09:18:24 DietPi homeassistant-start.sh[24954]: 2020-01-03 09:18:24 ERROR (MainThread) [homeassistant.bootstrap] Error loading /mnt/dietpi_userdata/homeassistant/configuration.yaml:   in "/mnt/dietpi_userdata/homeassistant/configur
ation.yaml", line 14, column 11: Unable to read file /mnt/dietpi_userdata/homeassistant/frontend.yaml.
Jan 03 09:18:24 DietPi homeassistant-start.sh[24954]: 2020-01-03 09:18:24 INFO (MainThread) [homeassistant.core] Starting Home Assistant
Jan 03 09:18:24 DietPi homeassistant-start.sh[24954]: 2020-01-03 09:18:24 INFO (MainThread) [homeassistant.core] Timer:starting
Jan 06 16:56:29 DietPi systemd[1]: Stopping Home Assistant (DietPi)...
Jan 06 16:56:29 DietPi systemd[1]: Stopped Home Assistant (DietPi).
Jan 06 16:56:29 DietPi systemd[1]: Started Home Assistant (DietPi).
Jan 06 16:56:30 DietPi homeassistant-start.sh[82406]: 2020-01-06 16:56:30 ERROR (MainThread) [homeassistant.bootstrap] Error loading /mnt/dietpi_userdata/homeassistant/configuration.yaml:   in "/mnt/dietpi_userdata/homeassistant/configur
ation.yaml", line 14, column 11: Unable to read file /mnt/dietpi_userdata/homeassistant/frontend.yaml.
Jan 06 16:56:30 DietPi homeassistant-start.sh[82406]: 2020-01-06 16:56:30 INFO (MainThread) [homeassistant.core] Starting Home Assistant
Jan 06 16:56:30 DietPi homeassistant-start.sh[82406]: 2020-01-06 16:56:30 INFO (MainThread) [homeassistant.core] Timer:starting
Jan 06 17:04:35 DietPi systemd[1]: Stopping Home Assistant (DietPi)...
Jan 06 17:04:35 DietPi systemd[1]: Stopped Home Assistant (DietPi).
Jan 06 17:04:49 DietPi systemd[1]: Started Home Assistant (DietPi).
Jan 06 17:04:49 DietPi homeassistant-start.sh[83795]: 2020-01-06 17:04:49 ERROR (MainThread) [homeassistant.bootstrap] Error loading /mnt/dietpi_userdata/homeassistant/configuration.yaml:   in "/mnt/dietpi_userdata/homeassistant/configur
ation.yaml", line 14, column 11: Unable to read file /mnt/dietpi_userdata/homeassistant/frontend.yaml.
Jan 06 17:04:49 DietPi homeassistant-start.sh[83795]: 2020-01-06 17:04:49 INFO (MainThread) [homeassistant.core] Starting Home Assistant
Jan 06 17:04:49 DietPi homeassistant-start.sh[83795]: 2020-01-06 17:04:49 INFO (MainThread) [homeassistant.core] Timer:starting
```

looks like there is (was) an issue with /home/homeassistant/.homeassistant/frontend.yaml.
I have not removed it but I can see it is gone via WinSCP.
I created a new frontend.yaml and uploaded it with correct rights and owners.
This is a bit strange, as there must have been some kind of auto deleting.
But I will have a close look at it and try to get the source of this problem.

now `journalctl -u home-assistant` tells me:
```
Jan 06 19:34:14 DietPi homeassistant-start.sh[86184]: 2020-01-06 19:34:14 INFO (MainThread) [homeassistant.core] Starting Home Assistant
Jan 06 19:34:14 DietPi homeassistant-start.sh[86184]: 2020-01-06 19:34:14 INFO (MainThread) [homeassistant.core] Timer:starting

```

But its still not accessible via IP:8123

Phil1988, 2020-01-08:
OK I will close this issue because dietpi was not the cause.

It was just a failed backup and then reconfigure home assistant.
I Can also report back, that the python update did not cause any further problems with the system.

Sorry for posting here, thanks for your help and thank you for all the great work you put into this great project!
I really appreciate it!

closed 2020-01-08 by Phil1988