MichaIng/DietPi #3320: Installation of No-Ip.com fails

ben0rism, 2020-01-06:
#### Details:
- Date           | Mon  6 Jan 18:19:27 GMT 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux DietPi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | unzip -o 67.zip 
- Exit code      | 11
- Software title | DietPi-Software

#### Steps to reproduce:
1. Installation DietPi
2. DHCP ON - ( connected via eth)
3. 

#### Expected behaviour:
Installation No-Ip.com dac

#### Actual behaviour:

#### Extra details:
Nothing special 
#### Additional logs:
```
Archive:  67.zip
caution: filename not matched:  
```

MichaIng, 2020-01-08:
@ben0rism 
Many thanks for your report. I'll try to replicate.

MichaIng, 2020-01-09:
Dammit, this bug has been introduced with this fix: https://github.com/MichaIng/DietPi/issues/3300
That fixed extraction of zip archives to a target dir, but when an empty argument is given, unzip interprets this as contained file name and of course does not find an empty string file name... Not smart at all.

Hence it is not possible to to pass a variable to the unzip command to conditionally add an argument. I tested all variations:
```
2020-01-09 13:03:32 root@VM-Buster:/tmp/DietPi-Software$ unzip -o 67.zip ''
Archive:  67.zip
caution: filename not matched:
2020-01-09 13:03:43 root@VM-Buster:/tmp/DietPi-Software$ unzip '' -o 67.zip
unzip:  cannot find or open , .zip or .ZIP.
2020-01-09 13:03:47 root@VM-Buster:/tmp/DietPi-Software$ unzip -o '' 67.zip
unzip:  cannot find or open , .zip or .ZIP.
```

MichaIng, 2020-01-09:
Affected software titles:
- No-IP: All archs/models
- WebIOPi: BPi Pro only
- WiFi Hotspot: RTL8188C on non-Armbian based ARMs
- EmonHub
- Gogs ARMv6/7
- Ampache

One liner hotfix via MOTD hack required.

MichaIng, 2020-01-09:
Ah perfect, there is a way:
```
2020-01-09 13:50:25 root@VM-Buster:/tmp$ var='dir with spaces'
2020-01-09 13:50:53 root@VM-Buster:/tmp$ unzip -o test.zip ${var:+"$var"}
Archive:  test.zip
caution: filename not matched:  dir with spaces
2020-01-09 13:50:56 root@VM-Buster:/tmp$ var='-ddir with spaces'
2020-01-09 13:51:13 root@VM-Buster:/tmp$ unzip -o test.zip ${var:+"$var"}
Archive:  test.zip
  inflating: dir with spaces/index.php
  inflating: dir with spaces/readme.txt
2020-01-09 13:51:15 root@VM-Buster:/tmp$ unset var
2020-01-09 13:51:27 root@VM-Buster:/tmp$ unzip -o test.zip ${var:+"$var"}
Archive:  test.zip
  inflating: index.php
  inflating: readme.txt
```
2020-01-09 13:51:29 root@VM-Buster:/tmp$ l
drwxr-xr-x 2 root root   80 Jan  9 13:51 'dir with spaces'
-rw-r--r-- 1 root root 106K Dec 28  2016  index.php
-rw-r--r-- 1 root root  70K Dec 28  2016  readme.txt
-rw-r--r-- 1 root root  38K Dec 29  2016  test.zip

Joulinar, 2020-01-09:
@MichaIng 
stupid question from my side. For the MOTD hack. Does MOTD needs to be activated or will it work even if MOTD is not enabled within `dietpi-banner`

MichaIng, 2020-01-09:
@Joulinar 
It needs to be activated. Actually I want to implement this separately as "live patches" or something, so less sneaky and more transparent. Users can then enable or disable it or check which patches would be done and decide themselves.

This is not only useful where me made a mistake/too few testing, and bugs slipped through beta, but as well when external changes happen, e.g. download URLs change, repo keys break/expire, new software versions break our installer/config files and all such. For this to keep up with, a very regular release schedule of 1 per month would be required, but the last releases showed that I am not able to implement all I want in one month, including beta phase etc. and by times new issues arise faster then I am able to fix. Hence such hotfixes would be able to prevent users from running into simple-to-fix (one-liner) bugs quickly while we are able to take the time a new release and wanted features/changes + testing take.

MichaIng, 2020-01-09:
Fixed with (allowed some code simplification as well, this time tested up and down 😉): https://github.com/MichaIng/DietPi/commit/9b3df750352aa3706e1be51f50b38a720cb729eb
Changelog: https://github.com/MichaIng/DietPi/commit/ebc7b20a6fde99da5b0f4cca318c3fa8c71c96d8

Quick fix:
```
sed -i 's/G_RUN_CMD unzip -o $file "$target"/G_RUN_CMD unzip -o $file ${target:+"$target"}/' /DietPi/dietpi/dietpi-software
```

Joulinar, 2020-01-09:
@MichaIng 
>Actually I want to implement this separately as "live patches"

sounds great, like that idea. I hope it will not cause to much overhead to keep the track between the releases and the applied HotFix

>This is not only useful where I made a mistake or too few testing

We are all humans and these thinks could happen. And you are doing such a great job. 👍 
Unfortunately I don't understand that much from this shell coding stuff. It would make things easier to assist you. (not just spamming around on the board)

MichaIng, 2020-01-09:
Live path, for now:
```
# Live patches
# - https://github.com/MichaIng/DietPi/issues/3313
[[ $G_DIETPI_VERSION_SUB == 2[78] && -w '/DietPi/dietpi/func/dietpi-banner' ]] && sed -i '/G_TERM_CLEAR$/s/$/ 2> \/dev\/null || printf "\\ec"/' /DietPi/dietpi/func/dietpi-banner
# - https://github.com/MichaIng/DietPi/issues/3314
[[ $G_DIETPI_VERSION_SUB == 2[78] && $G_DISTRO == 4 && -w '/DietPi/dietpi/dietpi-software' ]] && sed -i '/always-show-logo/d' /DietPi/dietpi/dietpi-software
# - https://github.com/MichaIng/DietPi/issues/3320
[[ $G_DIETPI_VERSION_SUB == 2[78] && -w '/DietPi/dietpi/dietpi-software' ]] && sed -i 's/G_RUN_CMD unzip -o $file "$target"/G_RUN_CMD unzip -o $file ${target:+"$target"}/' /DietPi/dietpi/dietpi-software

# MOTD
motd='★☆ ✿¸.•*¨★*☆*★`*•..¸✿ ☆¯`★
          ★[̲̅̅H̲̅][̲̅̅A̲̅][̲̅̅P̲̅][̲̅̅P̲̅][̲̅̅Y̲̅]★[̲̅̅N̲̅][̲̅̅E̲̅][̲̅̅W̲̅]★[̲̅̅Y̲̅][̲̅̅E̲̅][̲̅̅A̲̅][̲̅̅R̲̅]★
          *”˜˜”*°•.♥ღϠ₡ღ♥♥ღϠ₡ღ♥♥ღϠ₡ღ*”˜”*'
```

MichaIng, 2020-01-09:
@Joulinar 
> Unfortunately I don't understand that much from this shell coding stuff. It would make things easier to assist you. (not just spamming around on the board)

I am very thankful for your support on issues/reports on forum and here. You debug steps are great and having a lower issue reply time is great anyway ❤️.

I was thinking to add you as contributor here and moderator in the forum, so you're able to edit/flag/move threads/issues. This enables you push commits as well, however sticking with PRs for now make sense.

Joulinar, 2020-01-09:
@MichaIng 
I con confirm the fix working

>however sticking with PRs for now make sense. 

what do you mean with `PRs` :see_no_evil:

MichaIng, 2020-01-10:
@Joulinar 
Pull requests 😉.

ben0rism, 2020-01-10:
Cool that was fast!  Thank you guys! :) :+1:

closed 2020-05-03 by MichaIng