MichaIng/DietPi #3322: Chromium Kiosk - Permission Denied

lasers, 2020-01-07:
### Creating a bug report/issue

#### Required Information
```
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- `buster 10.2`
- `Linux dietpi-leaf 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`
- `RPi 4 Model B (armv7l)`
- `Raspberry Pi 15.3W USB-C Power Supply`
- `SanDisk ultra`

#### Steps to reproduce

1. Install chromium kiosk on fresh install.
2. Use user `dietpi`.

#### Expected behaviour

- Start chromium kiosk.

#### Actual behaviour
- Received `/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh Permission denied` on the screen.

Joulinar, 2020-01-07:
Hi,
pls try to run chromium as user `root` within the autostart option

lasers, 2020-01-07:
This works fine. Thanks. I'm filing a bug because it doesn't work out of box with non-root. Also, `chromium-browser` takes up some of the screen similar to https://github.com/MichaIng/DietPi/issues/2306. I'm just trying things right now.

Joulinar, 2020-01-07:
hi, you would need to adjust Chromium resolution settings as follow `nano /DietPi/dietpi.txt`. Close to the bottom of the file you will find the following.
```
# Chromium options
SOFTWARE_CHROMIUM_RES_X=1280
SOFTWARE_CHROMIUM_RES_Y=720
SOFTWARE_CHROMIUM_AUTOSTART_URL=https://dietpi.com
```

lasers, 2020-01-07:
Ah I see. That's it. Thank you very much. :blush: I'll opt for minimal WM instead.  I'll leave this issue open since I'm not sure if it's an issue that `chromium-browser` must be used with user `root`. Cheers.

Joulinar, 2020-01-07:
you can fix the access to the file. that's not the problem, but you will create new `issues` like `Cannot open /dev/tty0 (Permission denied)`

MichaIng, 2020-01-08:
@lasers 
Many thanks for your report. Indeed the option to choose autologin user is relatively new and we did not yet auto-apply to required permissions to start certian applications, as also noted in dietpi-autostart GUI, AFAIK.

I'll check the required permissions for Chromium kiosk mode + GPU acceleration and implement/post here. Until then, choosing root will work.

theiconrichie, 2020-01-10:
I have root set, and i get no such file found. doesnt work for me and the pi zero is so slow it cant run chromium on a X desktop at all.

Joulinar, 2020-01-10:
@theiconrichie 
pls can you past `ls -l /var/lib/dietpi/dietpi-software/installed/`

promofu, 2020-02-12:
I just had a similar issue and looked for file permissions along the path and
found that /var/lib/dietpi allthough being a directory was set to 660 instead of at least 770.

Maybe that helps ?

MichaIng, 2020-02-13:
@promofu 
Many thanks for reporting.
You're right, we assign this wrong during image creation 😓: https://github.com/MichaIng/DietPi/commit/8f28f231cbbbae7659005020dbf20d9cb86cdd38
I'll add this to v6.29 patch + MOTD live patch until then.

MichaIng, 2020-03-18:
Fixed: https://github.com/MichaIng/DietPi/commit/d6cfa300305b356d1f5e4bc9c4da582771153b15
MOTD hack:
```
# - https://github.com/MichaIng/DietPi/issues/3322
[[ $UID == 0 ]] && chmod -f +x $(find /var/lib/dietpi -type d) /var/lib/dietpi/services/*sh /var/lib/dietpi/dietpi-software/installed/*sh
```
Changelog: https://github.com/MichaIng/DietPi/commit/363533b003fa9a6789204c9e621d0f571b1e3e84

closed 2020-05-03 by MichaIng