MichaIng/DietPi #3323: my nerxcloud.

ludji49, 2020-01-07:
helloI'm no longer at my nerxcloud.
https://nextcloud-myweb.ddns.net/index.php/login
[image]

Joulinar, 2020-01-07:
pls can you check in your device if all the required services are still running like redis-server, mariadb, php and webser (eg lighttpd). This can be done using `dietpi-services` and should looks like this

```
●─ Single Service Options ─────────────●
redis-server : active | Affinity 0-0
mariadb      : active | Affinity 0-0
php7.3-fpm   : active | Affinity 0-0
lighttpd     : active | Affinity 0-0
```
If one if the services is down, pls try to restart

ludji49, 2020-01-07:
I get that. 

[image]

Joulinar, 2020-01-07:
As far as I can see you are using a DDNS service to connect. pls can you try to open NextCloud from your local network using the local IP for your DietPi device

`http://192.168.0.13/nextcloud`

as well you could try to restart all services.

ludji49, 2020-01-07:
no it's impossible to open

Joulinar, 2020-01-07:
usually you should be able to reach the default page of your Apache web server like `http://<yourDietPiAddress>`

if not pls can you try to restart Apache `systemctl restart apache2` and past the output of 

```
systemctl status apache2
journalctl -u apache2
```

ludji49, 2020-01-07:
still unable to open the page
[image]

Joulinar, 2020-01-07:
@ludji49 
hmm strange. Apache seems to be starting but your are not able to connect. Pls can you past output of `lsof -i -P -n | grep LISTEN`. This will show us if Apache is listen on Port 80 as it should be. As well pls can you check availabile content within the webserver root dir `ls -la /var/www/`

ludji49, 2020-01-08:
here's the result
[image]

Joulinar, 2020-01-08:
hmm there are quite a lot of Apache processes. Did you tried to restart your system in meantime to clean it up? Because it's quite strange as your webserver is running and listen on port 80 but you still did not get anything if you open the following web page `http://192.168.0.47`

ludji49, 2020-01-08:
I get nothing from the address

Joulinar, 2020-01-08:
what about the restart of the whole system?

ludji49, 2020-01-08:
> qu'en est-il du redémarrage de l'ensemble du système?

how does that

Joulinar, 2020-01-08:
type `reboot` on OS layer. after 1-2 minutes you should be able to login again

ludji49, 2020-01-08:
after the reboot I still can't open the page.
[image]

Joulinar, 2020-01-08:
pls can you try using the IP Address of you DietPi System and not use the DDNS name

MichaIng, 2020-01-08:
@ludji49 
Also, the systemctl/journal logs does not contain all binary error messages, please to the following to get binary errors, in case:
```
systemctl restart apache2 # Force fresh error messages, in case RAMlog got cleared meanwhile
cat /var/log/apache2/error.log
```

ludji49, 2020-01-08:
I don't have access 

[image]
[image]

ludji49, 2020-01-08:
I have access to the website 
thanks for your support
[image]

ludji49, 2020-01-08:
I've got this mistake

[image]

MichaIng, 2020-01-08:
@ludji49 
Great, though not sure what changed? The `curl` was successful, since SSL cert is only valid for real domain, the SSL error is expected.
First Nextcloud UI screen looks like you removed the asterisk `*` from valid domains in `config.php`, which is generally a good idea but breaks access via any other domain or IP as the explicitly listed ones.

Whatever changed, great it works again, hence I mark this issue as closed. Feel free to reopen if required.
___________
> I've got this mistake

Please run the following two commands in console:
```
ncc db:add-missing-indices
ncc db:convert-filecache-bigint
```

Generally just follow what those warnings suggest, but use `ncc` command instead of `occ`, which we implemented as alias to invoke php with the correct user to call `/var/www/nextcloud/occ`.

closed 2020-01-08 by MichaIng

Joulinar, 2020-01-08:
probably I misunderstood the entire issue as I was at the impression that the websever was somehow running but there where an issue to get something displayed while calling the website. But good that it is solved for now, even we don't know how 🙈

MichaIng, 2020-01-08:
@Joulinar 
> Damn I was fighting with the issue and you solved it in an hour

No solution provided by me either, probably the forth and back with Apache restarts solved something we could not see in the logs anymore. Sometimes its just playing around a bid and magically things are working again... Network not 100% predictable :slightly_smiling_face:.