MichaIng/DietPi #3324: A very strange question, I have a headache..

hifitime, 2020-01-07:
### Creating a bug report/issue

**In this way, no error will be reported**

`root@DietPi:/etc/systemd/system# echo '  192.168.1.111    ' | sed 's/ //g'
192.168.1.111
root@DietPi:/etc/systemd/system#`

`root@DietPi:/etc/systemd/system# echo '  192.168.1.111    ' | awk '{sub(/^ */, "");sub(/ *$/, "")}1'
192.168.1.111
root@DietPi:/etc/systemd/system# 
`

**But this operation will report an error**

[image]

[image]

The same problem occurs with "cat" or "sed" or "awk" commands，The same command can be executed in the console

**Required Information**
DietPi version | cat /DietPi/dietpi/.version
#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=26
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version | echo $G_DISTRO_NAME or cat /etc/debian_version
10.2

Kernel version | uname -a
Linux DietPi 4.19.84-mvebu64 #5.99 SMP PREEMPT Mon Nov 18 22:13:49 CST 2019 aarch64 GNU/Linux

SBC device | echo $G_HW_MODEL_DESCRIPTION or (EG: RPi3)
Generic Device (aarch64)

MichaIng, 2020-01-08:
@hifitime 
Generally I would use `[[:blank:]]` in sed/grep/awk etc to match any kind of spaces., e.g.
```
echo ' 192.168.1.111 ' | sed 's/[[:blank:]]//g'
```

Next if its in a variable already, simplify it by avoiding external binaries like sed/grep/awk and use bash internal string maipulation instead:
```
echo ${IP//[[:blank:]]/}
IP=${IP//[[:blank:]]/}
```

The same is what I would use with awk, to assure correct free space handling:
```
echo '  192.123.123.43     ' | awk '{sub(/^[[:blank:]]*/,"");sub(/[[:blank:]]*$/,"")}1'
```

In your systemd unit, the issue is that you use cascaded single quotes, which AFAIK is not possible. Either use double quotes to wrap the shell commands or to wrap the internal command arguments. In your case, the dash command ends with the single quite after echo, which creates a syntax error, and an additional argument starts with the 3rd single quote etc.

With double quotes, cascading can be done (although not sure if it works well in systemd units as well, but in shell scripts+console at least), when having command substitutions, e.g.: `var="123$(grep "pattern with spaces" /path/to/file)"`
But with single quotes this does not work, since everything is taken literally. In the systemd unit the shell code is not itself wrapped, but systemd interprets it a shell-like way, hence as well takes things in single quotes literally and hence does not know about command substitutions or contained syntax but simple ends a string/argument with the next single quote.

hifitime, 2020-01-15:
I want to add this run instruction to / etc / SYSTEMd / system /, but always report an error... I tried many times, many ways，My goal is to point to removing the space before and after

ExecStart=/usr/bin/123 "$(cat /xxxxx/xxxxxxx | grep ABCD | sed "s/.*ABCD -//;s/ //")"
OR
ExecStart=/usr/bin/123 "$(cat /xxxxx/xxxxxxx | grep ABCD | sed "s/.*ABCD -//;s/[[:blank:]]//g")"
OR
ExecStart=/usr/bin/123 "$(cat /xxxxx/xxxxxxx | grep ABCD | sed "s/.*ABCD -//" | sed 's/[[:blank:]]//g')"

MichaIng, 2020-01-15:
@hifitime 
Systemd units are no shells, command substitution and such does not work. You can either wrap it into a dash or bash session or you apply the result as environment variable, which would be my preference to avoid the shell overhead. Or is `/usr/bin/123` a oneshot command and no daemon?

Env var method:
```
ExecStartPre=/bin/dash -c 'systemctl set-environment IP=$(sed -nE "/ABCD/{s/^.*ABCD -[[:blank:]]*([^[:blank:]]*).*$/\1/p;q}" /xxxxx/xxxxxxx)'
ExecStart=/usr/bin/123 $IP
```
This way the shell is only opened at service start but does not stay open as part of the main process. Variables applied this way are available throughout the systemd unit and its processes. Scraping file is done more effectively via single sed which can be done one a file directly (no cat required) and can be apply on certain patterns only as well. Probably the pattern needs to be adjusted to your use case.

MichaIng, 2020-03-11:
I mark this issue as closed, feel free to reopen if required.

closed 2020-03-11 by MichaIng