MichaIng/DietPi #3325: Any software install fails on Plex

barkusmasta, 2020-01-08:
For some reason the software installs are failing due to an invalid signature for Plex but funny thing is, I didn't select Plex to be installed.

#### Details:
- Date           | Tue  7 Jan 17:19:12 MST 2020
- Bug report     | N/A
- DietPi version | v6.28.0 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- SBC device     | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux PlexPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- Distro         | buster (ID=5)
- Command        | G_AGUP
- Exit code      | 100
- Software title | DietPi-Software

#### Steps to reproduce:

1. ...
2. ...

#### Expected behaviour:

- ...

#### Actual behaviour:

- ...

#### Extra details:

- ...

#### Additional logs:
```
Ign:1 https://downloads.plex.tv/repo/deb public InRelease
Get:2 https://downloads.plex.tv/repo/deb public Release [5,815 B]
Get:3 https://downloads.plex.tv/repo/deb public Release.gpg [821 B]
Ign:3 https://downloads.plex.tv/repo/deb public Release.gpg
Hit:4 https://archive.raspberrypi.org/debian buster InRelease
Hit:5 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists...
W: GPG error: https://downloads.plex.tv/repo/deb public Release: The following signatures were invalid: BADSIG 97203C7B3ADCA79D Plex Inc.
E: The repository 'https://downloads.plex.tv/repo/deb public Release' is not signed.
```

warm--tape, 2020-01-08:
Came here to report the same issue.

barkusmasta, 2020-01-08:
> +1
> 
> Came here to report the same issue.

I ended up removing the Plex repo all together. I’ll fiddle with it later but for now I just need to get through updates.

UNiXMIT, 2020-01-08:
I think this is an issue with Plex and not DietPi - https://forums.plex.tv/t/again-failed-to-fetch-https-downloads-plex-tv-repo-deb-dists-public-release-gpg/510793/26

MichaIng, 2020-01-08:
@barkusmasta 
Many thanks for your report. I'll check, damn thing if their repo key is outdated/invalid... Yes workaround is do remove the remove for now and re-add it when this has been fixed.

What I am confused about:
> Kernel version | Linux PlexPi 4.19.75-v7l+

PlexPi Linux? This does not match our image, although all the other meta info do. I hope Plex does not ship any matching kernel packages with their repo, that would be horrible...

barkusmasta, 2020-01-08:
> I think this is an issue with Plex. Not sure why it's affecting DietPi if you don't have it installed though - https://forums.plex.tv/t/again-failed-to-fetch-https-downloads-plex-tv-repo-deb-dists-public-release-gpg/510793/26

You're right... This is the same issue.  I promise I searched prior to posting but apparently my search query needed improvement.  Thanks for pointing this out.

barkusmasta, 2020-01-08:
> @barkusmasta
> Many thanks for your report. I'll check, damn thing if their repo key is outdated/invalid... Yes workaround is do remove the remove for now and re-add it when this has been fixed.
> 
> If you never selected Plex Media Server for install, then indeed the repo should not be there, actually it should not be possible. Please verify when running `dietpi-software` > Uninstall that Plex is indeed not in the list.

"What I am confused about:

Kernel version | Linux PlexPi 4.19.75-v7l+"

I believe there was a previous regarding "PlexPi" and there might have been some confusion.  "PlexPi" is the hostname I gave this board when I installed.  SD card imaged from downloaded "DietPi_v6.25_RPi-ARMv6-Buster.img" file.

MichaIng, 2020-01-08:
@barkusmasta 
> What I am confused about:
> ...
> "PlexPi" is the hostname

Jep, my bad, this is why I edited that part out of my comment, everything is fine.

> I promise I searched prior to posting but apparently my search query needed improvement.

It is perfectly fine that you reported the issue here, which is the way me and other users get aware of this and how to work around or wait for solution on Plex side 👍. Many thanks as well to @UNiXMIT for linking the related issue from Plex forum!

warm--tape, 2020-01-09:
> @barkusmasta
> Many thanks for your report. I'll check, damn thing if their repo key is outdated/invalid... Yes workaround is do remove the remove for now and re-add it when this has been fixed.
>
> If [...]

Repo key appears to be fixed, I was able to successfully install Plex just now.

UNiXMIT, 2020-01-09:
> Repo key appears to be fixed, I was able to successfully install Plex just now.

Same here. Thanks for the heads up.

MichaIng, 2020-01-09:
Yes, thanks for heads up all, I'll mark this issue as closed.

closed 2020-01-09 by MichaIng