MichaIng/DietPi #3326: Automated kiosk mode blocked by GPLv2 accept screen

mrbluecoat, 2020-01-08:
### Creating a bug report/issue

#### Required Information
- DietPi version | 6.28.0
- Distro version | buster
- Kernel version | 4.19.75-v7+
- SBC device | RPi 3 Model B (armv7l)
- Power supply used | RPi3 plug
- SDcard used | SanDisk ultra 8 GB

#### Additional Information (if applicable)
- Software title | Chromium
- Was the software title installed freshly or updated/migrated?  Fresh install
- Can this issue be replicated on a fresh installation of DietPi?  Yes

#### Steps to reproduce

1. On a Mac, flashed official DietPi Raspberry Pi image onto SD card
2. Updated dietpi-wifi.txt with Wi-Fi credentials
3. Updated dietpi.txt:

```
sed -i '' 's/AUTO_SETUP_LOCALE=.*/AUTO_SETUP_LOCALE=en_US.UTF-8/' dietpi.txt
sed -i '' 's/AUTO_SETUP_KEYBOARD_LAYOUT=.*/AUTO_SETUP_KEYBOARD_LAYOUT=us/' dietpi.txt
sed -i '' 's/AUTO_SETUP_TIMEZONE=.*/AUTO_SETUP_TIMEZONE=America\/Phoenix/' dietpi.txt
sed -i '' 's/AUTO_SETUP_NET_WIFI_ENABLED=.*/AUTO_SETUP_NET_WIFI_ENABLED=1/' dietpi.txt
sed -i '' 's/AUTO_SETUP_AUTOSTART_TARGET_INDEX=.*/AUTO_SETUP_AUTOSTART_TARGET_INDEX=11/' dietpi.txt
sed -i '' 's/AUTO_SETUP_AUTOMATED=.*/AUTO_SETUP_AUTOMATED=1/' dietpi.txt
sed -i '' 's/#AUTO_SETUP_INSTALL_SOFTWARE_ID=44.*/AUTO_SETUP_INSTALL_SOFTWARE_ID=113    # Chromium browser/' dietpi.txt
sed -i '' 's/CONFIG_BOOT_WAIT_FOR_NETWORK=.*/CONFIG_BOOT_WAIT_FOR_NETWORK=2/' dietpi.txt
sed -i '' 's/CONFIG_AUTO_DIETPI_UPDATES=.*/CONFIG_AUTO_DIETPI_UPDATES=1/' dietpi.txt
sed -i '' 's/CONFIG_WIFI_COUNTRY_CODE=.*/CONFIG_WIFI_COUNTRY_CODE=US/' dietpi.txt
sed -i '' 's/CONFIG_SERIAL_CONSOLE_ENABLE=.*/CONFIG_SERIAL_CONSOLE_ENABLE=0/' dietpi.txt
sed -i '' 's/CONFIG_ENABLE_IPV6=.*/CONFIG_ENABLE_IPV6=0/' dietpi.txt
sed -i '' 's/SOFTWARE_CHROMIUM_RES_X=.*/SOFTWARE_CHROMIUM_RES_X=1920/' dietpi.txt
sed -i '' 's/SOFTWARE_CHROMIUM_RES_Y=.*/SOFTWARE_CHROMIUM_RES_Y=1080/' dietpi.txt
```
4. I put the SD card in the Raspberry Pi 3, plugged in the HDMI cable, and the power cable.
5. DietPi started, updated, and eventually displayed the blue GPLv2 ncurses whiptail accept screen

#### Expected behaviour
I expected the [SOFTWARE_CHROMIUM_AUTOSTART_URL](https://github.com/MichaIng/DietPi/blob/master/dietpi.txt#L236) to be displayed in full-screen Chromium kiosk mode on the screen.

#### Actual behaviour
Displays the blue GPLv2 ncurses whiptail accept screen

#### Extra details
When I SSH into the unit I can accept the GPLv2 OK prompt screen but it still displays on the HDMI TV source.   Then, if I then run `reboot` or `/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh` it will display the kiosk mode output as expected, except:

1.  The mouse is still visible (similar to #2298).  `G_AGI unclutter && echo '/usr/bin/unclutter -idle 0.1 &' > /etc/chromium-browser/customizations/99-dietpi-unclutter` from #2575 and a `reboot` fixed that.
2.  I get Chromium nag popups saying it didn't shut down properly.   Changing `--homepage ` to `--app=` in */var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh* fixed that (see https://raspberrypi.stackexchange.com/a/85827).

P.S. https://github.com/MichaIng/DietPi/blob/master/dietpi.txt#L48-L49 is confusing because the comment says "Disable" but the default value is enabled.  Perhaps change to something like "Set to '1' to disable HDMI output..."

Joulinar, 2020-01-08:
Hi,
this is all as expected I would say. On first initial boot you would need to run the inital setup process once. This will update you to latest DietPi version. So you need to do the following

1. flash your SD card
2. change your settings in dietpi.txt
3. don't enable autostart for Chromium yet
4. plug in SD card into your SBC
5. connect via SSH to your SBC and run first initial set up
6. once done and all required software was installed you can enable autostart for Chromium
7. install unclutter to hide the mouse
8. change chromium-autostart.sh to enable --app mode to hide error messages during boot

MichaIng, 2020-01-08:
@mrbluecoat 
Many thanks for your report.

Indeed the GPLv2 license prompt is skipped if automated firstrun setup was chosen. But it then prompts on the next "interactive" login instead. If this login is not actually interactive (we check shell parameters, not for an actually attached keyboard, which would be totally not trivial or impossible, especially when considering SSH or serial console), then this causes an issue. So yeah current workaround is given by @Joulinar :+1:.

I am thinking about how to reasonably prevent this issue. Most autostart options are anyway only usable interactively, so we could simply hide/skip the Chromium kiosk autostart option until GPLv2 license has been accepted. But this info must already been given in `config.txt`, else users select automated firstrun setup + Chromium kiosk autostart and expect the website to show up on screen/LCD without any need for interactivity.
We could of course skip the license prompt if Chromium kiosk is selected, but this is actually nothing I want to do. The license should be visible and accepted as fast as DietPi is actively used in any way.

mrbluecoat, 2020-01-08:
Since our use case required a fully automated installation at the destination location, with some help from #2938 and a bit more trial and error, here's the solution that finally worked for me:

```
sed -i '' 's/AUTO_SETUP_LOCALE=.*/AUTO_SETUP_LOCALE=en_US.UTF-8/' dietpi.txt
sed -i '' 's/AUTO_SETUP_KEYBOARD_LAYOUT=.*/AUTO_SETUP_KEYBOARD_LAYOUT=us/' dietpi.txt
sed -i '' 's/AUTO_SETUP_TIMEZONE=.*/AUTO_SETUP_TIMEZONE=America\/Phoenix/' dietpi.txt
sed -i '' 's/AUTO_SETUP_NET_WIFI_ENABLED=.*/AUTO_SETUP_NET_WIFI_ENABLED=1/' dietpi.txt
sed -i '' 's/AUTO_SETUP_AUTOSTART_TARGET_INDEX=.*/AUTO_SETUP_AUTOSTART_TARGET_INDEX=11/' dietpi.txt
sed -i '' 's/AUTO_SETUP_AUTOMATED=.*/AUTO_SETUP_AUTOMATED=1/' dietpi.txt
sed -i '' "s/AUTO_SETUP_GLOBAL_PASSWORD=.*/AUTO_SETUP_GLOBAL_PASSWORD=$SSH_PASSWORD/" dietpi.txt
sed -i '' 's/#AUTO_SETUP_INSTALL_SOFTWARE_ID=44.*/AUTO_SETUP_INSTALL_SOFTWARE_ID=113    # Chromium browser/' dietpi.txt
sed -i '' 's/CONFIG_BOOT_WAIT_FOR_NETWORK=.*/CONFIG_BOOT_WAIT_FOR_NETWORK=2/' dietpi.txt
sed -i '' 's/CONFIG_AUTO_DIETPI_UPDATES=.*/CONFIG_AUTO_DIETPI_UPDATES=1/' dietpi.txt
sed -i '' 's/CONFIG_WIFI_COUNTRY_CODE=.*/CONFIG_WIFI_COUNTRY_CODE=US/' dietpi.txt
sed -i '' 's/CONFIG_SERIAL_CONSOLE_ENABLE=.*/CONFIG_SERIAL_CONSOLE_ENABLE=0/' dietpi.txt
sed -i '' 's/CONFIG_ENABLE_IPV6=.*/CONFIG_ENABLE_IPV6=0/' dietpi.txt
sed -i '' 's/SOFTWARE_CHROMIUM_RES_X=.*/SOFTWARE_CHROMIUM_RES_X=1920/' dietpi.txt
sed -i '' 's/SOFTWARE_CHROMIUM_RES_Y=.*/SOFTWARE_CHROMIUM_RES_Y=1080/' dietpi.txt
sed -i '' 's/SOFTWARE_CHROMIUM_AUTOSTART_URL=.*/SOFTWARE_CHROMIUM_AUTOSTART_URL=https:\/\/example.com/' dietpi.txt

cat > Automation_Custom_Script.sh <<EOF
#!/bin/bash
if [[ -f '/var/lib/dietpi/license.txt' ]]; then mv /var/lib/dietpi/license.txt /var/lib/dietpi/license.accepted ; fi
sed -i 's/--homepage\s/--app=/' /var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh
sed -i 's/xinit.*/xinit \\\$FP_CHROMIUM \\\$CHROMIUM_OPTS -- -nocursor/' /var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh
EOF
```

MichaIng, 2020-01-08:
@mrbluecoat 
> [[ -f '/var/lib/dietpi/license.txt' ]]; then mv /var/lib/dietpi/license.txt /var/lib/dietpi/license.accepted

Jep this is what works around the issue. Probably it is easiest if we add an `AUTO_SETUP_ACCEPT_LICENSE` option (with comment that `dietpi-LICENSE.md` should be read), which can be set to 1 to skip the prompt. When users change/set this actively, this can be taken as sufficient step, comparable with hitting `<Ok>` once on interactive prompt.

MichaIng, 2020-03-18:
I'll postpone this, together with all those Chromium kiosk/app mode tasks, to v6.30. Need to get v6.29 finished and this is more a special case with existing workaround IMO.

MichaIng, 2020-06-25:
#### Done:
- New setting: https://github.com/MichaIng/DietPi/commit/1a3dcac35379b91bddbb760ecc02010db177d118
- Skips dialog: https://github.com/MichaIng/DietPi/commit/adf1409c5a6a2f0584e3412a5bfaff76881be334
- Changelog: https://github.com/MichaIng/DietPi/commit/3d3dcfddd05709ff7a888de17ca52e4685488163
- Clarified wording on `AUTO_SETUP_HEADLESS`: https://github.com/MichaIng/DietPi/commit/dfbc502b415a754d6870ec226531ed96b5353b21

closed 2020-06-30 by MichaIng

Velociraptor85, 2021-10-04:
Is it possible to use this with an RPi 4B with 2 monitors and 2 different websites ?

MichaIng, 2021-10-04:
At least not with a single kiosk mode instance. Probably it is possible via an adjusted copy of the `/var/lib/dietpi/dietpi-software/installed/chromium-autostart.sh` script which is then started (separately) on the other screen attached to a different TTY.

I'm actually not sure how the second HDMI is used OOTB on RPI4. If anything, what do you actually see by default on the second monitor? If nothing, does a login prompt show up when you run:
```sh
systemctl start getty@tty2
```

Velociraptor85, 2021-10-17:
The command launched an console on the monitor which the chromium was displayed on.

Velociraptor85, 2021-11-13:
> 
> 
> Try:
> 
> ```shell
> G_CONFIG_INJECT 'max_framebuffers=' 'max_framebuffers=2' /boot/config.txt
> reboot
> ```

So sadly this did not help -_- its still only one framebuffer

MichaIng, 2021-11-13:
Do you have a KMS driver enabled?
```sh
grep 'dtoverlay=v3d' /boot/config.txt
```

Velociraptor85, 2021-11-13:
So with both monitors attached, both `max_framebuffers=2 ` and `dtoverlay=v3d` set and updated to 7.8.2 I have now two framebuffer devices =) nice

MichaIng, 2021-11-13:
Whoops, I made a typo above, it should be `dtverlay=vc4-kms-v3d` (full KMS driver) or `dtverlay=vc4-fkms-v3d` (fake KMS driver). I was thinking whether this has somehow an impact on the framebuffer devices. At least for GUI applications, those are then not used anymore, but `/dev/dri/card0` resp. `/dev/dri/card1` instead.

Velociraptor85, 2021-11-13:
- so for both `vc4-kms-v3d` & `vc4-fkms-v3d` I only get one Framebuffer device but I have the card0/1 devices
- with `vc4` or `v3d` I get both framebuffer deivces but no `/dev/dri`
- neither configuration starts a shell on the second screen via the mentioned command `systemctl start getty@tty2`

MichaIng, 2021-11-13:
> with vc4 or v3d I get both framebuffer deivces but no /dev/dri

These overlays do not exist, hence it's like you simply remove the line. Okay so enabling KMS disables the second framebuffer, ignoring `max_framebuffers=2`. This looks like a bug to me since with this it is impossible to show two consoles on both screens, AFAIK. But it is possible to start a second X server and hence second Chromium, since it uses DRM/KMS automatically, when available. Not sure whether the second DRI is the second screen or whether something needs to be configured for that. To be tested:
```sh
xinit "$(command -v chromium-browser)" --kiosk -- :1
```
This can be called on any console or SSH session.

Velociraptor85, 2021-11-13:
So I removed the framebuffer setting and set `vc4-kms-v3d` - sadly I could not start another chromium browser with that command on the other screen. The first screen blanked out to black and I got following message: 
```
root@dietpi:~# xinit "$(command -v chromium-browser)" --kiosk -- :1

X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Raspbian
Current Operating System: Linux halle 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1080M@60 video=HDMI-A-2:1920x1200M@60 smsc95xx.macaddr=DC:A6:32:49:CA:1E vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=PARTUUID=0e2887fb-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=ttyS0,115200 console=tty1
Build Date: 05 August 2021  11:35:55AM
xorg-server 2:1.20.11-1+rpt1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.1.log", Time: Sat Nov 13 23:22:32 2021
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
error: XDG_RUNTIME_DIR not set in the environment.
Xorg: ../../../../dix/privates.c:384: dixRegisterPrivateKey: Assertion `!global_keys[type].created' failed.
(EE)
(EE) Backtrace:
(EE)
(EE)
Fatal server error:
(EE) Caught signal 6 (Aborted). Server aborting
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE) Please also check the log file at "/var/log/Xorg.1.log" for additional information.
(EE)
(EE) Server terminated with error (1). Closing log file.

xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
```

MichaIng, 2021-11-13:
Can you check the mentioned log file:
```sh
cat /var/log/Xorg.1.log
```
The same works with the main screen, does it?
```sh
xinit "$(command -v chromium-browser)" --kiosk -- :0
```

Velociraptor85, 2021-11-13:
```
[    48.673]
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[    48.673] Build Operating System: linux Raspbian
[    48.673] Current Operating System: Linux halle 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l
[    48.673] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1200M@60 smsc95xx.macaddr=DC:A6:32:49:CA:1E vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=PARTUUID=0e2887fb-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=ttyS0,115200 console=tty1
[    48.673] Build Date: 05 August 2021  11:35:55AM
[    48.673] xorg-server 2:1.20.11-1+rpt1 (https://www.debian.org/support)
[    48.673] Current version of pixman: 0.40.0
[    48.673]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    48.674] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    48.674] (==) Log file: "/var/log/Xorg.1.log", Time: Sat Nov 13 23:42:04 2021
[    48.674] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    48.674] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    48.674] (==) No Layout section.  Using the first Screen section.
[    48.675] (==) No screen section available. Using defaults.
[    48.675] (**) |-->Screen "Default Screen Section" (0)
[    48.675] (**) |   |-->Monitor "<default monitor>"
[    48.675] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    48.675] (**) Option "BlankTime" "0"
[    48.675] (==) Automatically adding devices
[    48.675] (==) Automatically enabling devices
[    48.675] (==) Automatically adding GPU devices
[    48.675] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    48.675] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    48.675]    Entry deleted from font path.
[    48.675] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    48.676]    Entry deleted from font path.
[    48.676] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    48.676]    Entry deleted from font path.
[    48.676] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    48.676]    Entry deleted from font path.
[    48.676] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    48.676]    Entry deleted from font path.
[    48.676] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    48.676]    Entry deleted from font path.
[    48.676] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[    48.676] (==) ModulePath set to "/usr/lib/xorg/modules"
[    48.676] (**) Extension "DPMS" is disabled
[    48.676] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    48.676] (II) Loader magic: 0x20cf40
[    48.676] (II) Module ABI versions:
[    48.676]    X.Org ANSI C Emulation: 0.4
[    48.676]    X.Org Video Driver: 24.1
[    48.676]    X.Org XInput driver : 24.1
[    48.676]    X.Org Server Extension : 10.0
[    48.677] (--) using VT number 3

[    48.677] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    48.679] (II) xfree86: Adding drm device (/dev/dri/card1)
[    48.679] (EE) /dev/dri/card1: failed to set DRM interface version 1.4: Permission denied
[    48.680] (II) xfree86: Adding drm device (/dev/dri/card0)
[    48.681] (II) no primary bus or device found
[    48.681]    falling back to /sys/devices/platform/v3dbus/fec00000.v3d/drm/card0
[    48.681] (II) LoadModule: "glx"
[    48.681] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    48.684] (II) Module glx: vendor="X.Org Foundation"
[    48.684]    compiled for 1.20.11, module version = 1.0.0
[    48.684]    ABI class: X.Org Server Extension, version 10.0
[    48.684] (==) Matched modesetting as autoconfigured driver 0
[    48.684] (==) Matched fbdev as autoconfigured driver 1
[    48.684] (==) Assigned the driver to the xf86ConfigLayout
[... 175 lines cut ...]
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[    49.757] (EE) Please also check the log file at "/var/log/Xorg.1.log" for additional information.
[    49.758] (EE)
[    49.780] (EE) Server terminated with error (1). Closing log file.
```
starting the browser on :0 gives me following error
```
root@halle:~# xinit "$(command -v chromium-browser)" --kiosk -- :0

(EE)
Fatal server error:
(EE) Server is already active for display 0
        If this server is no longer running, remove /tmp/.X0-lock
        and start again.
(EE)
(EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
(EE)
```

only after killing X I can restart the browser

MichaIng, 2021-11-13:
Strange. I loads KMS with `/dev/dri/card1` correctly, but then also tries to load the framebuffer module, which then fails 🤔. Can you compare with the log for screen `:0` (makes sense that an already running X server on that screen needs to be stopped first 😉):
```sh
cat /var/log/Xorg.0.log
```

Velociraptor85, 2021-11-13:
after a fresh reboot the file shows
```
root@dietpi:~# cat /var/log/Xorg.0.log
[     8.461]
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[     8.462] Build Operating System: linux Raspbian
[     8.462] Current Operating System: Linux halle 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l
[     8.462] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:1920x1200M@60 smsc95xx.macaddr=DC:A6:32:49:CA:1E vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  root=PARTUUID=0e2887fb-02 rootfstype=ext4 rootwait fsck.repair=yes net.ifnames=0 logo.nologo quiet console=ttyS0,115200 console=tty1
[     8.462] Build Date: 05 August 2021  11:35:55AM
[     8.462] xorg-server 2:1.20.11-1+rpt1 (https://www.debian.org/support)
[     8.462] Current version of pixman: 0.40.0
[     8.462]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[     8.462] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[     8.463] (==) Log file: "/var/log/Xorg.0.log", Time: Sat Nov 13 23:50:49 2021
[     8.472] (==) Using config directory: "/etc/X11/xorg.conf.d"
[     8.472] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[     8.474] (==) No Layout section.  Using the first Screen section.
[     8.474] (==) No screen section available. Using defaults.
[     8.474] (**) |-->Screen "Default Screen Section" (0)
[     8.474] (**) |   |-->Monitor "<default monitor>"
[     8.477] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[     8.477] (**) Option "BlankTime" "0"
[     8.477] (==) Automatically adding devices
[     8.477] (==) Automatically enabling devices
[     8.477] (==) Automatically adding GPU devices
[     8.481] (==) Max clients allowed: 256, resource mask: 0x1fffff
[     8.489] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[     8.489]    Entry deleted from font path.
[     8.489] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[     8.489]    Entry deleted from font path.
[     8.489] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[     8.489]    Entry deleted from font path.
[     8.489] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[     8.489]    Entry deleted from font path.
[     8.489] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[     8.489]    Entry deleted from font path.
[     8.489] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[     8.489]    Entry deleted from font path.
[     8.489] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[     8.489] (==) ModulePath set to "/usr/lib/xorg/modules"
[     8.489] (**) Extension "DPMS" is disabled
[     8.489] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[     8.489] (II) Loader magic: 0x20cf40
[     8.489] (II) Module ABI versions:
[     8.489]    X.Org ANSI C Emulation: 0.4
[     8.489]    X.Org Video Driver: 24.1
[     8.489]    X.Org XInput driver : 24.1
[     8.489]    X.Org Server Extension : 10.0
[     8.491] (--) using VT number 2

[     8.491] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[     8.493] (II) xfree86: Adding drm device (/dev/dri/card1)
[     8.505] (II) xfree86: Adding drm device (/dev/dri/card0)
[     8.506] (II) no primary bus or device found
[     8.506]    falling back to /sys/devices/platform/gpu/drm/card1
[     8.506] (II) LoadModule: "glx"
[     8.508] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[     8.541] (II) Module glx: vendor="X.Org Foundation"
[     8.541]    compiled for 1.20.11, module version = 1.0.0
[     8.541]    ABI class: X.Org Server Extension, version 10.0
[     8.541] (==) Matched modesetting as autoconfigured driver 0
[     8.541] (==) Matched fbdev as autoconfigured driver 1
[... 191 lines cut ...]
[    11.033] (II) event1  - vc4: is tagged by udev as: Keyboard Pointingstick
[    11.034] (II) event1  - vc4: device is a pointer
[    11.034] (II) event1  - vc4: device is a keyboard
[    11.057] (**) vc4: Applying InputClass "libinput keyboard catchall"
[    11.057] (II) Using input driver 'libinput' for 'vc4'
[    11.057] (**) vc4: always reports core events
[    11.057] (**) Option "Device" "/dev/input/event0"
[    11.057] (**) Option "_source" "_driver/libinput"
[    11.057] (II) libinput: vc4: is a virtual subdevice
[    11.057] (**) Option "config_info" "udev:/sys/devices/platform/soc/fef00700.hdmi/rc/rc0/input1/event0"
[    11.057] (II) XINPUT: Adding extended input device "vc4" (type: KEYBOARD, id 8)
[    11.057] (**) Option "xkb_model" "pc105"
[    11.057] (**) Option "xkb_layout" "de"
[    11.112] (**) vc4: Applying InputClass "libinput keyboard catchall"
[    11.112] (II) Using input driver 'libinput' for 'vc4'
[    11.112] (**) vc4: always reports core events
[    11.112] (**) Option "Device" "/dev/input/event1"
[    11.112] (**) Option "_source" "_driver/libinput"
[    11.112] (II) libinput: vc4: is a virtual subdevice
[    11.113] (**) Option "config_info" "udev:/sys/devices/platform/soc/fef05700.hdmi/rc/rc1/input2/event1"
[    11.113] (II) XINPUT: Adding extended input device "vc4" (type: KEYBOARD, id 9)
[    11.113] (**) Option "xkb_model" "pc105"
[    11.113] (**) Option "xkb_layout" "de"
[    15.900] (II) modeset(0): Disabling kernel dirty updates, not required.
```

Velociraptor85, 2021-11-13:
seems like it gets both HDMI connections but thinks that one is not connected
```
[     9.975] (II) modeset(0): Output HDMI-1 disconnected
[     9.975] (II) modeset(0): Output HDMI-2 connected
```

MichaIng, 2021-11-13:
Somehow. Also on screen `:0` uses the same `/dev/dri/card1`, so none is using `/dev/dri/card0`. Of course both cannot use the same. At least the `fb` sub module seems to be expected, probably this doesn't mean that a framebuffer device is required, like the `fbdev` module does.

Btw, is one a 4k screen?

Linking the official docs HDMI section: https://www.raspberrypi.com/documentation/computers/configuration.html#hdmi-configuration
Probably it's indeed about the firmware/hardware parts and not (only) about the display driver and X server.

Velociraptor85, 2021-11-13:
Both are faily "old" standard full HD screens one has max 1920*1200 the other 1920*1080 and both work separately so 🤷

Velociraptor85, 2022-01-07:
so after some more hours of tinkering I found out that my displays where announing itselfe in different modes DMT/CEA so I tested with two identical displays both CEA - sadly - only in hdmi_safe mode it showed both displays via tvservice 
... later I found out that with `dtoverlay=vc4-fkms-v3d` mode it creates one screen
... with `DISPLAY=:0 xrandr` it shows both HDMI devices and I can start an chromium with the windows-position 1920,0 on the second screen 
... but I cant figure out how to start two instances of chromium with xinit 
... `xinit chromium-browser -- :1` does not work since it is found to be one screen with 3840 width -_-