MichaIng/DietPi #3327: Issues after updates

canabino, 2020-01-08:
@MichaIng HI there I will post the logs here for you for the RockPi4 with issues after the update with software titles Deluge Sonarr Radarr Sabnzbd and Jackett

canabino, 2020-01-08:
I just did the update. In dietpi-services everything is running, but I can't access Deluge web or daemon.
SabNzbd was running fine when I add a new file to download through Sonarr SabNzbd crashes and I can't access the webui even though it is running.
 
[image]

I will send the logs now.

canabino, 2020-01-08:
```
root@DietPi:~# cat /var/tmp/dietpi/logs/dietpi-update.log
cat: /var/tmp/dietpi/logs/dietpi-update.log: No such file or directory
```

canabino, 2020-01-08:
These for Deluge and Sabnzbd

```
root@DietPi:~# journalctl -u deluged
-- Logs begin at Wed 2020-01-08 19:42:42 GMT, end at Wed 2020-01-08 19:51:51 GMT. --
Jan 08 19:44:16 DietPi systemd[1]: Started Deluge Daemon (DietPi).
root@DietPi:~# journalctl -u sabnzbd
-- Logs begin at Wed 2020-01-08 19:42:42 GMT, end at Wed 2020-01-08 19:53:06 GMT. --
Jan 08 19:44:16 DietPi systemd[1]: Started SABnzbd (DietPi).

```

canabino, 2020-01-08:
These for Sonarr and Radarr
```
root@DietPi:~# journalctl -u sonarr
-- Logs begin at Wed 2020-01-08 19:42:42 GMT, end at Wed 2020-01-08 19:55:19 GMT. --
Jan 08 19:44:16 DietPi systemd[1]: Started Sonarr (NzbDrone) Daemon (DietPi).
Jan 08 19:44:18 DietPi mono[983]: [Info] Bootstrap: Starting Sonarr - /opt/NzbDrone/NzbDrone.exe - Version 2.0.0.5337
Jan 08 19:44:19 DietPi mono[983]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/sonarr
]
Jan 08 19:44:20 DietPi mono[983]: [Info] Router: Application mode: Interactive
Jan 08 19:44:20 DietPi mono[983]: [Info] MigrationLogger: *** Migrating data source=/mnt/dietpi_userdata/sonarr/nzbdrone.d
b;cache size=-10000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3 ***
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** Migrating data source=/mnt/dietpi_userdata/sonarr/logs.db;ca
che size=-10000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3 ***
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** VersionMigration migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** VersionUniqueMigration migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** VersionDescriptionMigration migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 1: InitialSetup migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] InitialSetup: Starting migration to 1
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 2: remove_tvrage_imdb_unique_constraint migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] remove_tvrage_imdb_unique_constraint: Starting migration to 2
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 3: remove_renamed_scene_mapping_columns migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] remove_renamed_scene_mapping_columns: Starting migration to 3
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 4: updated_history migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] updated_history: Starting migration to 4
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 5: added_eventtype_to_history migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] added_eventtype_to_history: Starting migration to 5
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 6: add_index_to_log_time migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] add_index_to_log_time: Starting migration to 6
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 7: add_renameEpisodes_to_naming migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] add_renameEpisodes_to_naming: Starting migration to 7
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 8: remove_backlog migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] remove_backlog: Starting migration to 8
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 9: fix_rename_episodes migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] fix_rename_episodes: Starting migration to 9
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 10: add_monitored migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] add_monitored: Starting migration to 10
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 11: remove_ignored migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] remove_ignored: Starting migration to 11
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 12: remove_custom_start_date migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] remove_custom_start_date: Starting migration to 12
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 13: add_air_date_utc migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] add_air_date_utc: Starting migration to 13
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 14: drop_air_date migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] drop_air_date: Starting migration to 14
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 15: add_air_date_as_string migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] add_air_date_as_string: Starting migration to 15
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 16: updated_imported_history_item migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] updated_imported_history_item: Starting migration to 16
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 17: reset_scene_names migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] reset_scene_names: Starting migration to 17
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 18: remove_duplicates migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] remove_duplicates: Starting migration to 18
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 19: restore_unique_constraints migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] restore_unique_constraints: Starting migration to 19
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 20: add_year_and_seasons_to_series migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] add_year_and_seasons_to_series: Starting migration to 20
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 21: drop_seasons_table migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] drop_seasons_table: Starting migration to 21
Jan 08 19:44:21 DietPi mono[983]: [Info] MigrationLogger: *** 22: move_indexer_to_generic_provider migrating ***
Jan 08 19:44:21 DietPi mono[983]: [Info] move_indexer_to_generic_provider: Starting migration to 22
[... cut ...]
Jan 08 19:54:47 DietPi mono[983]:   at NzbDrone.Core.Download.Clients.Deluge.Deluge.GetItems () [0x00029] in <96f52d5e5b95
442a89ae3bbf62e59664>:0
Jan 08 19:54:47 DietPi mono[983]:   at NzbDrone.Core.Download.TrackedDownloads.DownloadMonitoringService.ProcessClientDown
loads (NzbDrone.Core.Download.IDownloadClient downloadClient) [0x0000c] in <96f52d5e5b95442a89ae3bbf62e59664>:0
Jan 08 19:55:10 DietPi mono[983]: [Warn] Torznab: Indexer RARBG rss sync didn't cover the period between 01/08/2020 02:03:
59 and 01/08/2020 10:01:14 UTC. Search may be required.
```

canabino, 2020-01-08:
This is Jackett
root@DietPi:~# journalctl -u jackett
-- Logs begin at Wed 2020-01-08 19:42:42 GMT, end at Wed 2020-01-08 20:01:30 GMT. --
Jan 08 19:44:16 DietPi systemd[1]: Started Jackett (DietPi).
Jan 08 19:44:19 DietPi jackett[981]: 01-08 19:44:19 Info Starting Jackett v0.12.1494.0
Jan 08 19:44:22 DietPi jackett[981]: 01-08 19:44:22 Info Environment version: 3.0.1 (/opt/jackett/)
Jan 08 19:44:22 DietPi jackett[981]: 01-08 19:44:22 Info OS version: Unix 5.4.0.1 (64bit OS) (64bit process)
Jan 08 19:44:22 DietPi jackett[981]: 01-08 19:44:22 Info Jackett variant: CoreLinuxArm64
Jan 08 19:44:22 DietPi jackett[981]: 01-08 19:44:22 Info ThreadPool MaxThreads: 32767 workerThreads, 1000 completionPortTh
reads
Jan 08 19:44:22 DietPi jackett[981]: 01-08 19:44:22 Info App config/log directory: /opt/jackett/Jackett
Jan 08 19:44:22 DietPi jackett[981]: 01-08 19:44:22 Info issue: Debian Buster with Armbian Linux \l
Jan 08 19:44:22 DietPi jackett[981]: 01-08 19:44:22 Info Using HTTP Client: HttpWebClient2NetCore
Jan 08 19:44:23 DietPi jackett[981]: 01-08 19:44:23 Info Loading Cardigann definitions from: /opt/jackett/cardigann/defini
tions/, /etc/xdg/cardigan/definitions/, /opt/jackett/Definitions
Jan 08 19:44:23 DietPi jackett[981]: 01-08 19:44:23 Info Loading Cardigann definition /opt/jackett/Definitions/pthome.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/xwtclassics
.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/zamundanet.
yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/glodls.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/crnaberza.y
ml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/wihd.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/cili180.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/diablotorre
nt.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/ilcorsarone
ro.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/chdbits.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/kapaki.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/trezzor.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/pornbay.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/lesaloon.ym
l
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/acgrip.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/cinematik.y
ml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/sporthd.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/audiobookto
rrents.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/kickasstorr
ent.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/torrentdown
load.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/gay-torrent
s.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/cpasbienclo
ne.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/cinemageddo
n.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/seedfile.ym
l
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/mononokebt.
yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/ourbits.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/tfile.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/retroflix.y
ml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/crazyspirit
s.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/uniongang.y
ml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/racing4ever
yone.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/3evils.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/exttorrents
.yml
Jan 08 19:44:24 DietPi jackett[981]: 01-08 19:44:24 Info Loading Cardigann definition /opt/jackett/Definitions/onlineselfe
ducation.yml
[... cut ...]
Jan 08 19:55:31 DietPi jackett[981]: 01-08 19:55:31 Info Found 100 (0 new) releases from LimeTorrents
Jan 08 19:55:33 DietPi jackett[981]: 01-08 19:55:33 Info Found 100 (0 new) releases from LimeTorrents
Jan 08 19:55:35 DietPi jackett[981]: 01-08 19:55:35 Info Found 100 (0 new) releases from LimeTorrents
Jan 08 19:55:36 DietPi jackett[981]: 01-08 19:55:36 Info Found 100 (0 new) releases from LimeTorrents
Jan 08 19:55:39 DietPi jackett[981]: 01-08 19:55:39 Info Found 100 (0 new) releases from LimeTorrents

canabino, 2020-01-08:
This is for `dmesg | tail`

root@DietPi:~# dmesg | tail
[ 1246.868600]  __switch_to+0xc8/0x218
[ 1246.868615]  __schedule+0x23c/0x600
[ 1246.868626]  schedule+0x38/0xd0
[ 1246.868642]  futex_wait_queue_me+0xc0/0x138
[ 1246.868656]  futex_wait+0x118/0x200
[ 1246.868668]  do_futex+0x568/0xc60
[ 1246.868682]  __arm64_sys_futex+0x114/0x150
[ 1246.868698]  el0_svc_common.constprop.1+0x88/0x178
[ 1246.868713]  el0_svc_handler+0x20/0x80
[ 1246.868725]  el0_svc+0x8/0xc

canabino, 2020-01-08:
Sorry here is Radarr

root@DietPi:~# journalctl -u radarr
-- Logs begin at Wed 2020-01-08 19:42:42 GMT, end at Wed 2020-01-08 20:06:13 GMT. --
Jan 08 19:44:16 DietPi systemd[1]: Started Radarr Daemon (DietPi).
Jan 08 19:44:18 DietPi mono[986]: [Info] Bootstrap: Starting Radarr - /opt/Radarr/Radarr.exe - Version 0.2.0.1358
Jan 08 19:44:20 DietPi mono[986]: [Info] AppFolderInfo: Data directory is being overridden to [/mnt/dietpi_userdata/radarr
]
Jan 08 19:44:20 DietPi mono[986]: [Info] Router: Application mode: Interactive
Jan 08 19:44:21 DietPi mono[986]: [Info] MigrationLogger: *** Migrating data source=/mnt/dietpi_userdata/radarr/nzbdrone.d
b;cache size=-20000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3 ***
Jan 08 19:44:21 DietPi mono[986]: [Info] MigrationLogger: *** Migrating data source=/mnt/dietpi_userdata/radarr/logs.db;ca
che size=-20000;datetimekind=Utc;journal mode=Wal;pooling=True;version=3 ***
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** VersionMigration migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** VersionUniqueMigration migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** VersionDescriptionMigration migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 1: InitialSetup migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] InitialSetup: Starting migration to 1
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 2: remove_tvrage_imdb_unique_constraint migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] remove_tvrage_imdb_unique_constraint: Starting migration to 2
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 3: remove_renamed_scene_mapping_columns migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] remove_renamed_scene_mapping_columns: Starting migration to 3
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 4: updated_history migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] updated_history: Starting migration to 4
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 5: added_eventtype_to_history migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] added_eventtype_to_history: Starting migration to 5
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 6: add_index_to_log_time migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] add_index_to_log_time: Starting migration to 6
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 7: add_renameEpisodes_to_naming migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] add_renameEpisodes_to_naming: Starting migration to 7
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 8: remove_backlog migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] remove_backlog: Starting migration to 8
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 9: fix_rename_episodes migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] fix_rename_episodes: Starting migration to 9
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 10: add_monitored migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] add_monitored: Starting migration to 10
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 11: remove_ignored migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] remove_ignored: Starting migration to 11
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 12: remove_custom_start_date migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] remove_custom_start_date: Starting migration to 12
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 13: add_air_date_utc migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] add_air_date_utc: Starting migration to 13
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 14: drop_air_date migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] drop_air_date: Starting migration to 14
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 15: add_air_date_as_string migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] add_air_date_as_string: Starting migration to 15
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 16: updated_imported_history_item migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] updated_imported_history_item: Starting migration to 16
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 17: reset_scene_names migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] reset_scene_names: Starting migration to 17
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 18: remove_duplicates migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] remove_duplicates: Starting migration to 18
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 19: restore_unique_constraints migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] restore_unique_constraints: Starting migration to 19
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 20: add_year_and_seasons_to_series migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] add_year_and_seasons_to_series: Starting migration to 20
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 21: drop_seasons_table migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] drop_seasons_table: Starting migration to 21
Jan 08 19:44:22 DietPi mono[986]: [Info] MigrationLogger: *** 22: move_indexer_to_generic_provider migrating ***
Jan 08 19:44:22 DietPi mono[986]: [Info] move_indexer_to_generic_provider: Starting migration to 22
[... cut ...]
.Core.Download.Clients.Deluge.DelugeSettings settings, System.String method, System.Object[] arguments) [0x00008] in <2807
1e62f2f449e682962e05af20a3d7>:0
Jan 08 20:06:13 DietPi mono[986]:   at NzbDrone.Core.Download.Clients.Deluge.DelugeProxy.GetTorrentsByLabel (System.String
 label, NzbDrone.Core.Download.Clients.Deluge.DelugeSettings settings) [0x00012] in <28071e62f2f449e682962e05af20a3d7>:0
Jan 08 20:06:13 DietPi mono[986]:   at NzbDrone.Core.Download.Clients.Deluge.Deluge.GetItems () [0x00029] in <28071e62f2f4
49e682962e05af20a3d7>:0

MichaIng, 2020-01-08:
@canabino 
So Sonarr and Radarr and Jackett web UIs are all accessible and working fine right? Hence its both download backends Deluge and SABnzbd which have issues, sadly exactly those two services which do not have journal log entries. Let me check which log file they use 😉.

MichaIng, 2020-01-08:
@canabino 
Okay, please paste:
```
cat /var/log/deluged/daemon.log
````
If it's empty (hourly RAMlog clearing), `systemctl restart deluged` to create new entries.

For SABnzbd:
```
cat /var/log/sabnzbd/*
```
This is the log dir, not sure about the file name, hence cat everything inside. The same about restarting the service, if file has been cleared meanwhile.

canabino, 2020-01-08:
I actually downloaded and installed the image again and it is busy installing the software now. Last time the image didn't want to work but you said you fixed it. It seems to be working better and actually got past the initial setup after boot. 
I wanted to start fresh anyway. Let me test this and see if it works before we dive into something that could have been the matter with my previous setup.

MichaIng, 2020-01-09:
@canabino 
Okay makes sense. Probably it makes even more sense if I create the new image first?

canabino, 2020-01-09:
Well let me test this one for now, and If it works well, we can wait for an even newer version of Armbian to do a new image?

canabino, 2020-01-11:
@MichaIng 
Hi Micha I did a fresh install of the Dietpi System and started to install and configure all of the apps Sonarr, Radarr, SABnzbd, Deluge .  Deluge and SABnzbd still crash. According to `htop` and` dietpi-services` they are running but I can't access their webui.

If I run
```
root@DietPi:~# cat /var/log/deluged/daemon.log
[WARNING ] 19:38:49 preferencesmanager:528 GeoIP Unavailable
```
I then do 
`root@DietPi:~# systemctl restart deluged`
but it gets stuck and nothing happens for a long time then I ctrl-c to get out
If I use `dietpi-services restart` it also gets stuck at Deluge and can't restart.

For SABnzbd I ran
```
root@DietPi:~# systemctl restart sabnzbd
root@DietPi:~# cat /var/log/sabnzbd/*
root@DietPi:~#
```
I opened the log file in  cat /var/log/sabnzbd/sabnzbd.log
But it's empty.🤔 

If I restart the system the same thing happens.

Joulinar, 2020-01-11:
Would that be possible to do a fresh install and setup just 1 of the problematic software title like `deluged` alone. Just to avoid side effects and to figure out one by one?

MichaIng, 2020-01-11:
@canabino 
> I unplug then reconnect the LAN cable I can't access it with putty SSH

That is expected since by default there is no network adapter event detection or such. The `allow-hotplug` entries in /etc/network/interfaces wait for an initial cable plug or adapter init and then configure the network automatically, but when you unplug and re-plug the cable afterwards, it is expected to stay unconfigured, AFAIK. But haven't tested it for a while, actually. ifplugd would be a solution to allow it: https://packages.debian.org/buster/ifplugd

> but it gets stuck and nothing happens for a long time then I ctrl-c to get out

This will take 90 seconds (by default) before systemd gives up/times out. If you wait that time, there is a small chance that SABnzbd creates a log entry, but I don't guess so. All this sounds more like a system overload actually. Did you check RAM usage in `htop`, probably extensive swap file usage, or CPUs being on 100% for a long while by one of these services?

All all does not give any hint, then I agree with @Joulinar to start with only installing a single software title (or disabling/masking all other services for debugging).

canabino, 2020-01-12:
Hi @MichaIng 
SABnzbd, Deluge, Sonarr and Radarr had no problems after I installed them and I was able to access the webuis without problem. Deluge and SABnzbd started to crash after I configured everything and  added a new download to them. From there on SABnzbd I am able to access it's ui and it runs for a short time after I reboot but it crashes after a a few minutes, Deluge I am not able to access it's ui at all.

> This will take 90 seconds (by default) before systemd gives up/times out.
I have waited longer than 10 minutes but it gets stuck here
[image]
The three dots go from one side to the other then eventually stops. Then I ctrl-c to get out.

`htop` doesn't show a lot system resources used
[image]
When I stopped all the other services, (still can't stop deluge and SAB) then `htop` shows a lot of swap activity and memory usage and terminal is very slow when I input commands.
[image]

I ran
[image]
SABnzbd log is still empty.

Next step I will remove other software packages completely and report back.

canabino, 2020-01-12:
@MichaIng I think I found the culprit. 
I removed all software packages except for Deluge and SABnzbd. After a reboot, SAB and Deluge ran fine and I could access them. I then installed Jackett and Deluge did the same again. couldn't access it.
I then removed Jackett and insatalled Sonarr and Radarr everything runs fine without Jackett. 
I saw the following on Sonarr's update page
[image]
So something the matter with Jackett?

MichaIng, 2020-01-13:
@canabino
> So something the matter with Jackett?
 
We install mono v6, so nope the issue is not related.

The crash is due to full memory, I guess. See your second last screen, physical RAM and swap are both full. Since no active service is actually eating that much RAM, something seems to full some tmpfs dir or similar. Could you paste the output of `df` command, when replicating the issue? Maybe also when enabling kernel and user threads list in `htop` or using top could give a clue what uses 2G RAM there.

paliasp, 2020-01-17:
I am also having some memory and load average issues on a RockPi4 with 4GB ram v6.29.
What I notice is memory being consumed in about 6 hours, then system becomes unresponsive, and memory is released to become functional again, without reboot.
The culprit seems to be Jackett.

df looks good.
```
dietpi@rockpi:[~]$ df -h
Filesystem                                        Size  Used Avail Use% Mounted on
udev                                              1,9G     0  1,9G   0% /dev
tmpfs                                             387M   45M  343M  12% /run
/dev/mmcblk2p1                                     29G  1,5G   28G   6% /
tmpfs                                             1,9G   20K  1,9G   1% /dev/shm
tmpfs                                             5,0M     0  5,0M   0% /run/lock
tmpfs                                             1,9G     0  1,9G   0% /sys/fs/cgroup
tmpfs                                             1,9G   57M  1,9G   3% /tmp
tmpfs                                              50M     0   50M   0% /var/log
tmpfs                                              10M  1,4M  8,7M  14% /DietPi
```
Under stress I can see this
```
Load average: 48.00 47.72 46.44
USER       NI CPU% MEM%  VIRT   TIME+  Command
root        0  4.6  0.0     0  1h21:53 kswapd0
root        0  1.6  0.1  7580  0:00.31 htop -d 30 --sort-key=VIRT
jackett     0  1.0  1.8 3170M  1:20.44 /opt/jackett/jackett --NoRestart
Debian-sn   0  0.7  0.1 26888  8:36.91 /usr/sbin/snmpd -Lsd -Lf /dev/null -u Debian-snmp -g Debian-snmp -I -smux mteTrigger mteTriggerConf -f -p /run/snmpd.pid
root        0  0.7  0.1  162M  4:05.92 /sbin/init
root        0  0.7  0.0  8200  0:00.93 /usr/sbin/cron -f
jackett     0  0.3  1.8 3170M  0:00.05 /opt/jackett/jackett --NoRestart
jackett     0  0.3  1.8 3170M  0:06.62 /opt/jackett/jackett --NoRestart
root        0  0.3  0.0     0  0:00.04 kworker/5:2-events
root        0  0.3  0.0     0  0:17.01 kworker/0:2-events
dietpi      0  0.3  0.0 12100  0:00.79 sshd: dietpi@pts/0
root        0  0.3  0.0     0  0:20.44 kworker/1:1-events
root        0  0.0  0.0     0  9:49.35 rcu_preempt
root      -20  0.0  0.0     0  0:00.68 kworker/0:1H-kblockd
root        0  0.0  0.0     0  0:11.69 kworker/4:1-events_power_efficient
root      -20  0.0  0.0     0  0:02.36 kworker/4:2H-kblockd
root        0  0.0  0.0     0  0:21.20 kworker/2:0-events
root      -20  0.0  0.0     0  0:01.92 kworker/1:2H-kblockd
root        0  0.0  0.0 34908  0:14.82 /usr/sbin/nmbd --foreground --no-process-group
dietpi      0  0.0  0.0 21000  0:10.92 proftpd: (accepting connections)
jackett     0  0.0  1.8 3170M  0:00.41 /opt/jackett/jackett --NoRestart
```
This is when memory got released back and load avg started falling
```
top - 12:20:46 up 2 days, 11:48,  1 user,  load average: 2,03, 4,83, 20,91
Tasks: 139 total,   1 running, 138 sleeping,   0 stopped,   0 zombie
%Cpu(s):  0,1 us,  0,1 sy,  0,0 ni, 99,8 id,  0,0 wa,  0,0 hi,  0,0 si,  0,0 st
MiB Mem :   3867,7 total,   3228,3 free,    374,0 used,    265,5 buff/cache
MiB Swap:      0,0 total,      0,0 free,      0,0 used.   3304,1 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND                                                                                                                         
 2050 jackett   20   0 3245408 122920  63912 S   0,3   3,1   0:12.12 jackett                                                                                                                         
  581 www-data  20   0  286044  17560    244 S   0,0   0,4   0:01.67 mono                                                                                                                            
    1 root      20   0  166184   6056   3500 S   0,0   0,2   4:06.05 systemd                                                                                                                         
26235 statd     20   0   58308  49372      0 S   0,0   1,2   0:00.03 rpc.statd                                                                                                                       
24240 root      20   0   54028   4284   1648 S   0,0   0,1   0:00.54 smbd                                                                                                                            
24244 root      20   0   50728   2540     16 S   0,0   0,1   0:00.16 cleanupd                                                                                                                        
24243 root      20   0   50712   2532      8 S   0,0   0,1   0:00.16 smbd-notifyd                                                                                                                    
 6234 root      20   0   43264   5140   3996 S   0,0   0,1   0:08.60 systemd-journal                                                                                                                 
24227 root      20   0   34908   3764   1884 S   0,0   0,1   0:14.87 nmbd                                                                                                                            
  754 Debian-+  20   0   26888   6320   2872 S   0,0   0,2   8:38.65 snmpd                                                                                                                           
25286 dietpi    20   0   21000   1908     92 S   0,0   0,0   0:10.96 proftpd                                                                                                                         
[... cut ...]
  538 _rpc      20   0    7100    444      0 S   0,0   0,0   0:00.69 rpcbind                                                                                                                         
  903 root      20   0    5300    104      0 S   0,0   0,0   0:00.01 agetty                                                                                                                          
 2049 jackett   20   0    2036    468    404 S   0,0   0,0   0:00.04 dash                         
```

From packages I have installed sonarr, radarr, jackett, samba, proftpd, and replaced dropbear with sshd.
Let me know if more troubleshooting is needed.

Joulinar, 2020-01-17:
@paliasp 
Many thanks for your report. Pls can you verfiy your DietPi version because there is no v6.29 out. The actual version is v6.28.

pls can you past `htop` under full load.

paliasp, 2020-01-17:
That's what it says:
```
 ─────────────────────────────────────────────────────
 DietPi v6.29.0 (dev) : 12:41 - Παρ 17/01/2020
 ─────────────────────────────────────────────────────
 - Device model : ROCK Pi 4 (aarch64)
 - Uptime : up 2 days, 12 hours, 9 minutes
 - CPU temp : 39'C : 102'F (Cool runnings)
```
```
dietpi@rockpi:[~]$ cat /etc/debian_version 
10.2
dietpi@rockpi:[~]$ uname -a
Linux rockpi 5.4.0-rc1-rockchip64 #19.11.8.014 SMP PREEMPT Tue Jan 14 03:50:12 CET 2020 aarch64 GNU/Linux
dietpi@rockpi:[~]$ mono -V
Mono JIT compiler version 6.6.0.161 (tarball Tue Dec 10 11:01:57 UTC 2019)
Copyright (C) 2002-2014 Novell, Inc, Xamarin Inc and Contributors. www.mono-project.com
        TLS:           __thread
        SIGSEGV:       normal
        Notifications: epoll
        Architecture:  arm64
        Disabled:      none
        Misc:          softdebug 
        Interpreter:   yes
        LLVM:          yes(610)
        Suspend:       preemptive
        GC:            sgen (concurrent by default)
```
Although it is not so terrible, load avg and memory are rising even though there is no stressful cpu process.
```
  1  [                                                                                     0.0%]   4  [|                                                                                    0.7%]
  2  [|                                                                                    0.7%]   5  [|||                                                                                  2.6%]
  3  [                                                                                     0.0%]   6  [                                                                                     0.0%]
  Mem[||||||||||||||||||||||||                                                       752M/3.78G]   Tasks: 27, 18 thr; 1 running
  Swp[                                                                                    0K/0K]   Load average: 5.91 5.23 7.35 
  Cpu Temp:   37 C                                                                                 Uptime: 2 days, 12:15:33

  PID USER      PRI  NI  VIRT   RES   SHR S CPU% MEM%   TIME+  Command               
 2050 jackett    20   0 3169M  120M 63912 S  0.4  3.1  0:17.17 /opt/jackett/jackett --NoRestart
 2058 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.76 /opt/jackett/jackett --NoRestart
 3683 jackett    20   0 3169M  120M 63912 S  0.2  3.1  0:00.54 /opt/jackett/jackett --NoRestart
 2061 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.39 /opt/jackett/jackett --NoRestart
 2056 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.03 /opt/jackett/jackett --NoRestart
 2051 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart
 2052 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart
 2053 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart
 2054 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart
 2055 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart
 2057 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart
 2062 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart
 2065 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart               
 2066 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart
 3827 jackett    20   0 3169M  120M 63912 S  0.0  3.1  0:00.00 /opt/jackett/jackett --NoRestart
  581 www-data   20   0  279M 17560   244 S  0.0  0.4  0:01.67 /usr/bin/mono /usr/lib/mono/4.5/xsp4.exe --port 8084 --address 0.0.0.0 --appconfigdir /etc/xsp4 --nonstop
  793 www-data   20   0  279M 17560   244 S  0.0  0.4  0:00.06 /usr/bin/mono /usr/lib/mono/4.5/xsp4.exe --port 8084 --address 0.0.0.0 --appconfigdir /etc/xsp4 --nonstop
  586 www-data   20   0  279M 17560   244 S  0.0  0.4  0:00.00 /usr/bin/mono /usr/lib/mono/4.5/xsp4.exe --port 8084 --address 0.0.0.0 --appconfigdir /etc/xsp4 --nonstop
  595 www-data   20   0  279M 17560   244 S  0.0  0.4  0:00.00 /usr/bin/mono /usr/lib/mono/4.5/xsp4.exe --port 8084 --address 0.0.0.0 --appconfigdir /etc/xsp4 --nonstop
  797 www-data   20   0  279M 17560   244 S  0.0  0.4  0:00.00 /usr/bin/mono /usr/lib/mono/4.5/xsp4.exe --port 8084 --address 0.0.0.0 --appconfigdir /etc/xsp4 --nonstop
 2593 root       20   0  162M  7068  4496 D  0.0  0.2  0:00.01 (imesyncd)
 2889 root       20   0  162M  7068  4496 D  0.0  0.2  0:00.01 (imesyncd)
 3309 root       20   0  162M  7320  4748 D  0.0  0.2  0:00.01 (imesyncd)
 3760 root       20   0  162M  7068  4496 D  0.0  0.2  0:00.01 (imesyncd)
    1 root       20   0  162M  6364  3808 S  0.0  0.2  4:07.90 /sbin/init
26235 statd      20   0 58308 49372     0 S  0.0  1.2  0:00.03 /sbin/rpc.statd --no-notify
24240 root       20   0 54028  5332  2696 S  0.0  0.1  0:00.55 /usr/sbin/smbd --foreground --no-process-group
24244 root       20   0 50728  2540    16 S  0.0  0.1  0:00.17 /usr/sbin/smbd --foreground --no-process-group
24243 root       20   0 50712  2532     8 S  0.0  0.1  0:00.16 /usr/sbin/smbd --foreground --no-process-group
 6234 root       20   0 43264  5404  4260 S  0.0  0.1  0:08.89 /lib/systemd/systemd-journald
24227 root       20   0 34908  4164  2284 S  0.0  0.1  0:14.96 /usr/sbin/nmbd --foreground --no-process-group
[... cut ...] ian-snmp -I -smux mteTrigger mteTriggerConf -f -p /run/snmpd.pid
25286 dietpi     20   0 21000  3376  1556 S  0.0  0.1  0:11.03 proftpd: (accepting connections)
  460 root       20   0 18456  1128    16 S  0.0  0.0  0:00.91 /lib/systemd/systemd-udevd
 3425 dietpi     20   0 12100  3372  2492 S  0.0  0.1  0:00.24 sshd: dietpi@pts/0
23157 root       20   0 12100   852    16 S  0.0  0.0  0:00.22 /usr/sbin/sshd -D
 3423 root       20   0 12100  6364  5488 S  0.0  0.2  0:00.06 sshd: dietpi [priv]
```

MichaIng, 2020-01-17:
@paliasp 
I guess you updated to dev branch. This should be usually only done for development or when a certain issues that you faced was solved there. Afterwards it makes sense to revert to master branch:
```
G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /DietPi/dietpi.txt
```

Could you please hide threads in your htop? I first thought there are multiple Jackett processes running, but since this affects other processes as well I guess you show all threads of the same process, right?

Joulinar, 2020-01-17:
Doesn't looks like a physical memory issue. There are used 750M out of 4GB only. No swapping at all. But yeah you are running a Development Version.

MichaIng, 2020-01-17:
@Joulinar 
> But yeah you are running a Development Version.

But this should not be an issue currently, since besides bug fixes there have no related changes been done so far.

DietPi v6.27 included a Jackett reinstall which included a fix for Jackett failing after every internal update. The Jackett devs wrapped the binary call into a helper script, which waits for the updater to finish, before allowing systemd to restart the process. I didn't make a long-term test with this, just forced an update one or two times and checked whether the helper shell does its job. Since all this thing is still a workaround (currently C# programmers are missing at Jackett project to solve this binary-wise), and there is no signal handling available, to know for sure that any forks have finished, I is at least possible that under certain circumstances old forked processes are still running while systemd pushes new ones by restarting the service.

paliasp, 2020-01-17:
> Doesn't looks like a physical memory issue. There are used 750M out of 4GB only. No swapping at all. 
> But yeah you are running a Development Version.

I have disabled swapping.
Memory keeps raising, it just happened to take a snapshot when it was freed.
Right now, almost 1 hour after my previous post I have these:
```

  0  [                                                                                     0.0%]   Tasks: 29; 1 running                  
  1  [||                                                                                   1.1%]   Load average: 4.44 3.87 3.06 
  2  [                                                                                     0.0%]   Uptime: 2 days, 13:06:20
  3  [|                                                                                    0.4%]   Hostname: rockpi
  4  [||||||||                                                                             7.0%]   Hostname: rockpi
  5  [||                                                                                   1.8%]   Cpu Temp:   37 C
  Mem[|||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||||               1.25G/3.78G]   Time: 13:38:36
  Swp[                                                                                    0K/0K]

USER       NI CPU% MEM%  VIRT   TIME+  Command
sonarr      0  0.3  4.1 1430M  1:06.64 /usr/bin/mono -O=-aot /opt/NzbDrone/NzbDrone.exe -nobrowser -data=/mnt/dietpi_userdata/sonarr
jackett     0  0.7  4.0 3434M  0:29.84 /opt/jackett/jackett --NoRestart
radarr      0  0.0  3.1 1147M  0:16.45 /usr/bin/mono -O=-aot /opt/Radarr/Radarr.exe -nobrowser -data=/mnt/dietpi_userdata/radarr
statd       0  0.0  1.2 58308  0:00.03 /sbin/rpc.statd --no-notify
www-data    0  0.0  0.4  279M  0:01.67 /usr/bin/mono /usr/lib/mono/4.5/xsp4.exe --port 8084 --address 0.0.0.0 --appconfigdir /etc/xsp4 --nonstop
root        0  0.0  0.3 54004  0:00.12 /usr/sbin/smbd --foreground --no-process-group
root        0  0.0  0.2  162M  0:00.01 (imesyncd)
root        0  0.0  0.2  162M  0:00.01 (imesyncd)
root        0  0.0  0.2 34900  0:00.19 /usr/sbin/nmbd --foreground --no-process-group
root        0  0.0  0.2  162M  0:00.01 (imesyncd)
root        0  0.0  0.2 43264  0:09.70 /lib/systemd/systemd-journald
root        0  0.0  0.2  162M  4:10.28 /sbin/init
Debian-sn   0  0.0  0.2 26888  8:49.53 /usr/sbin/snmpd -Lsd -Lf /dev/null -u Debian-snmp -g Debian-snmp -I -smux mteTrigger mteTriggerConf -f -p /run/snmpd.pid
dietpi      0  0.0  0.1  8576  0:01.82 -bash
root        0  0.0  0.1 12100  0:00.06 sshd: dietpi [priv]
dietpi      0  0.0  0.1 20872  0:00.13 proftpd: (accepting connections)
root        0  1.3  0.1  7500  0:01.72 htop
root        0  0.0  0.1 50704  0:00.00 /usr/sbin/smbd --foreground --no-process-group
root        0  0.0  0.1 50712  0:00.01 /usr/sbin/smbd --foreground --no-process-group
root        0  0.0  0.1  8048  0:00.01 sudo htop
dietpi      0  0.0  0.1 12100  0:01.68 sshd: dietpi@pts/0
root        0  0.0  0.1  7784  2:39.26 /usr/sbin/haveged --Foreground --verbose=1 -w 1024
root        0  0.0  0.0  8200  0:00.01 /usr/sbin/cron -f
root        0  0.0  0.0  9040  0:00.09 /sbin/dhclient -4 -v -i -pf /run/dhclient.eth0.pid -lf /var/lib/dhcp/dhclient.eth0.leases -I -df /var/lib/dhcp/dhclient6.eth0.leases eth0
root        0  0.0  0.0 18456  0:00.91 /lib/systemd/systemd-udevd
root        0  0.0  0.0 12100  0:00.22 /usr/sbin/sshd -D
_rpc        0  0.0  0.0  7100  0:00.70 /sbin/rpcbind -f -w
jackett     0  0.0  0.0  2036  0:00.06 /bin/dash -c /opt/jackett/jackett --NoRestart; while pgrep -u jackett JackettUpdater; do sleep 1; done
root        0  0.0  0.0  5300  0:00.01 /sbin/agetty -o -p -- \u --noclear tty1 linux
```
This is how memory is used:
[image]

> 
> 
> @paliasp
> I guess you updated to dev branch. This should be usually only done for development or when a certain issues that you faced was solved there. Afterwards it makes sense to revert to master branch:
> 
> ```
> G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /DietPi/dietpi.txt
> ```
Not intentionally! Maybe the image is shipped with such settings? Because I definitely didn't fiddle with that file manually. I only did the upgrades when advertised in banner as well as firmware upgrade.
I'll change it to master.
 
> Could you please hide threads in your htop? I first thought there are multiple Jackett instances running, but since this affects other processes as well I guess you show all threads of the same process, right?

Sorry for that, hope it is better now
> Also please paste:
> 
> ```
> systemctl status jackett
> journalctl -u jackett # If updates were done, this can be very much, hence the last ~20 lines should be enough 😉.
[... 52 lines cut ...]
, Func`3 nestedObjectDeserializer, Object& value)
Ιαν 17 12:55:59 rockpi jackett[7411]:    at YamlDotNet.Serialization.ValueDeserializers.NodeValueDeserializer.DeserializeValue(IParser parser, Type expectedType, SerializerState state, IValueDeseria
lizer nestedObjectDeserializer)
Ιαν 17 12:55:59 rockpi jackett[7411]:    at YamlDotNet.Serialization.ValueDeserializers.AliasValueDeserializer.DeserializeValue(IParser parser, Type expectedType, SerializerState state, IValueDeseri
alizer nestedObjectDeserializer)
Ιαν 17 12:55:59 rockpi jackett[7411]:    at YamlDotNet.Serialization.NodeDeserializers.ObjectNodeDeserializer.YamlDotNet.Serialization.INodeDeserializer.Deserialize(IParser parser, Type expectedType
, Func`3 nestedObjectDeserializer, Object& value)
[... cut ...]
Ιαν 17 12:56:03 rockpi jackett[7411]: 01-17 12:56:03 Info Loading Cardigann definition /opt/jackett/Definitions/amigosshare.yml
Ιαν 17 12:56:03 rockpi jackett[7411]: 01-17 12:56:03 Info Adding aggregate indexer
Ιαν 17 12:56:04 rockpi jackett[7411]: Hosting environment: Production
Ιαν 17 12:56:04 rockpi jackett[7411]: Content root path: /opt/jackett/Content
Ιαν 17 12:56:04 rockpi jackett[7411]: Now listening on: http://[::]:9117
Ιαν 17 12:56:04 rockpi jackett[7411]: Application started. Press Ctrl+C to shut down.
```

canabino, 2020-01-17:
This happened to me on the old DietPi image for RockPi4 also. If I I updated it updated to 6.29 dev and Jackett caused issues for other programs. Installing the latest version of Micha's new build, fixed everything for me.

paliasp, 2020-01-17:
Thank you for the update @canabino !
This file doesn't exist, so I guess I'll use the one without the "_new" suffix from the same directory.

Joulinar, 2020-01-17:
Hi,
I guess @MichaIng copied already to the `stable` folder. At least time stamps are fitting.

https://dietpi.com/downloads/images/?C=M;O=D

canabino, 2020-01-17:
Yeah, the dates for the files are new. So he definitely updated the image at the stable location.

> Okay I moved the new image in stable place but left a symlink to testing dir in place, so all but the _new.7z links stay valid for a while.

MichaIng, 2020-01-17:
@paliasp @Joulinar @canabino 
Yes, after positive real machine test by @canabino I moved the now image to stable.

> Not intentionally! Maybe the image is shipped with such settings? Because I definitely didn't fiddle with that file manually. I only did the upgrades when advertised in banner as well as firmware upgrade.
I'll change it to master.

Ah yes, the old one was based on dev code, where the ROCK Pi 4 hardware definition has just been added. So indeed good to revert to master now.
AFAIK the old image as well based based on development firmware packages, while there are stable ones available (and installed on the new image). To assure that it is no development firmware/kernel issue, @paliasp you could try change those packages. Could you paste:
```
dpkg -l | grep '^linux-*'
```

Based on your htop, actually no process eats that memory, although together with the few hundred MiBs tmpfs usage it fits. Please keep an eye on both, `htop` and `df` to check if any process or tmpfs grows further.

Jackett logs look fine. The error is most likely due to an internal update. The JackettUpdater process updates the files while the service stays active. When a certain file is reached, the Jackett process cashes with an error message. The `while pgrep ... sleep` loop in the systemd unit assures that systemd is not restarting the service while the updater has not yet finished, but after it has. After the error message you should see some "service has stopped" "service is restarted" or similar.

MichaIng, 2020-01-20:
Okay I lost track a bid, but the issues are solved with the new stable image, right? I mark this issue as closed then, feel free to reopen if required.

closed 2020-01-20 by MichaIng

paliasp, 2020-01-20:
I didn't report anything else because I flashed the master version and was busy restoring and testing. So far it seems stable to me. Thank you very much!