MichaIng/DietPi #3328: Synthing fails to start due to wrong folder name for the synthing data

rondadon, 2020-01-10:
### Creating a bug report/issue

Hi there!

I installed dietpi onto my RPi4 4GB and updated to DietPi v6.28.0 . When installing Synthing it installs flawelessly but it wont start.

#### Required Information
- DietPi version | 
`#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`

- Distro version | 
`buster 10.2`
- Kernel version | 
`Linux DietPi4 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`
- SBC device | 
`RPi 4 Model B (armv7l)`
- Power supply used | 
`5V 4A Meanwell PSU`
- SDcard used | 
`SanDisk ultra 32GB for Boot` | `SSD with USB<->SATA Adapter for whole System`

#### Additional Information (if applicable)
- Software title | `Syncthing`
- Was the software title installed freshly or updated/migrated? `Freshly Installed`
- Can this issue be replicated on a fresh installation of DietPi? `It is a fresh Dietpi Installation`

#### Steps to reproduce

1. Update to DietPi v6.28.0
2. Install Syncthing
3. Look in "dietpi-services" for the status of Syncthing. It should show "failed" or run systemctl status syncthing

#### Expected behaviour

- syncthing should start normally and provide the Webui

#### Actual behaviour

- It seems that synthing was searching for /mnt/dietpi_userdata/syncthing instead for /mnt/dietpi_userdata/syncthing_data which is created under /mnt/dietpi_userdata/ and put in place there for syncthing. Therefore it looks for a path/folder which dosent exist and fails to start.

#### Extra details

`systemctl status syncthing` shows following:

```
syncthing.service - Syncthing (DietPi)
   Loaded: loaded (/etc/systemd/system/syncthing.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2020-01-10 03:43:48 CET; 6min ago
  Process: 1069 ExecStart=/opt/syncthing/syncthing -no-browser -no-restart -logflags=0 -home=/mnt/dietpi_userdata/syncthing (code=exited, status=1/FAILURE)
 Main PID: 1069 (code=exited, status=1/FAILURE)

Jan 10 03:43:48 DietPi4 systemd[1]: syncthing.service: Service RestartSec=100ms expired, scheduling restart.
Jan 10 03:43:48 DietPi4 systemd[1]: syncthing.service: Scheduled restart job, restart counter is at 5.
Jan 10 03:43:48 DietPi4 systemd[1]: Stopped Syncthing (DietPi).
Jan 10 03:43:48 DietPi4 systemd[1]: syncthing.service: Start request repeated too quickly.
Jan 10 03:43:48 DietPi4 systemd[1]: syncthing.service: Failed with result 'exit-code'.
Jan 10 03:43:48 DietPi4 systemd[1]: Failed to start Syncthing (DietPi).
```
So it terminates after `ExecStart=/opt/syncthing/syncthing -no-browser -no-restart -logflags=0 -home=/mnt/dietpi_userdata/syncthing`. 

I looked in the /mnt/dietpi_userdata/ folder to see if there is a syncthing folder present. It is but the folder is named `syncthing_data` instead of only `syncthing`. So I created the folder named `syncthing` , chmod/chown, and started syncthing with `systemctl start syncthing` . It the started without a failure. 

After that i ran `systemctl status syncthing` again and it showed following:

```
root@DietPi4:~# systemctl status syncthing
● syncthing.service - Syncthing (DietPi)
   Loaded: loaded (/etc/systemd/system/syncthing.service; disabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-01-10 19:50:50 CET; 2h 42min ago
 Main PID: 9811 (syncthing)
    Tasks: 15 (limit: 4915)
   Memory: 23.6M
   CGroup: /system.slice/syncthing.service
           └─9811 /opt/syncthing/syncthing -no-browser -no-restart -logflags=0 -home=/mnt/dietpi_userdata/syncthing

Jan 10 19:50:51 DietPi4 syncthing[9811]: [USJ72] INFO: QUIC listener ([::]:22000) starting
Jan 10 19:50:51 DietPi4 syncthing[9811]: [USJ72] INFO: TCP listener ([::]:22000) starting
Jan 10 19:50:51 DietPi4 syncthing[9811]: [USJ72] INFO: Completed initial scan of sendreceive folder "Default Folder" (default)
Jan 10 19:50:52 DietPi4 syncthing[9811]: [USJ72] INFO: GUI and API listening on 127.0.0.1:8384
Jan 10 19:50:52 DietPi4 syncthing[9811]: [USJ72] INFO: Access the GUI via the following URL: http://127.0.0.1:8384/
Jan 10 19:50:52 DietPi4 syncthing[9811]: [USJ72] INFO: My name is "DietPi4"
Jan 10 19:51:11 DietPi4 syncthing[9811]: [USJ72] INFO: quic://0.0.0.0:22000 detected NAT type: Port restricted NAT
Jan 10 19:51:11 DietPi4 syncthing[9811]: [USJ72] INFO: quic://0.0.0.0:22000 resolved external address quic://87.122.240.114:1024 (via stun.syncthing.net:3478)
Jan 10 19:51:11 DietPi4 syncthing[9811]: [USJ72] INFO: Detected 1 NAT service
Jan 10 19:51:17 DietPi4 syncthing[9811]: [USJ72] INFO: Joined relay relay://95.244.243.241:22067
```

I can discover the server from other devices running Syncthing but now my problem is that I can not connect to the WebUI. I think I need to Change the GUI and API Listening address from 127.0.0.1:8384 to 0.0.0.0:8384 ith the config for syncthing. But thats a different issue. I think the configuration File for syncthing in the DietPi v6.28.0 is not customized/adjusted propperly for dietpi.

Thank you for your help and have a nice weekend! :)

Joulinar, 2020-01-10:
@rondadon 
I did a small quick test on my RPi4 and Synthing was runnig out of the box without any manual changes. as well I have both folders availible right after setup finished.

```
root@DietPi4:/mnt/dietpi_userdata# ls -la
insgesamt 32
drwxrwxr-x 8 dietpi dietpi 4096 Jan 10 23:04 .
drwxr-xr-x 6 root   root   4096 Jan  7 23:35 ..
drwxrwxr-x 2 dietpi dietpi 4096 Jan 10 14:51 downloads
drwxrwxr-x 2 dietpi dietpi 4096 Jan 10 14:51 Music
drwxrwxr-x 2 dietpi dietpi 4096 Jan 10 14:51 Pictures
drwx------ 3 dietpi dietpi 4096 Jan 10 23:07 syncthing
drwxr-xr-x 3 dietpi dietpi 4096 Jan 10 23:05 syncthing_data
drwxrwxr-x 2 dietpi dietpi 4096 Jan 10 14:51 Video
root@DietPi4:/mnt/dietpi_userdata#
```

The ports LISTEN can be checked as follow `lsof -i -P -n | grep LISTEN`

MichaIng, 2020-01-11:
@rondadon 
Many thanks for your report. I just rechecked why we actually have three different directories for Syncthing, but that is all correct:
- /opt/syncthing for the binary and non-variable files, hence the part that can be removed and updated without loosing any user config/data.
- /mnt/dietpi_userdata/syncthing for config
- /mnt/dietpi_userdata/syncthing_data for the user data handled by Syncthing, which is also observed by the internal file watcher

The config dir is actually created here: https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L10775
- `sudo -u dietpi /opt/syncthing/syncthing -generate=$G_FP_DIETPI_USERDATA/syncthing`
- Execute binary to pre-create the config directory and defaults
- Below this line many commands rely on this being created to apply our default settings, e.g. that `syncthing_data` should be used as data dir. So if the dir creation failed, you must have received a bunch of other error messages. But those probably moved to quickly off of the screen 😉.

To debug why this failed in your case, could you simply reinstall Syncthing (with `/mnt/dietpi_userdata/syncthing` non-existent): `dietpi-software reinstall 50`
Then when you reach the configuration part, keep eyes open and please paste the output here if error messages are present and `/mnt/dietpi_userdata/syncthing` is accordingly still missing.

MichaIng, 2020-02-04:
I mark this issue as closed, due to missing response and as I could not replicate. Feel free to reopen if issue perists.

closed 2020-02-04 by MichaIng