MichaIng/DietPi #3329: Syncthing: Can not connect to WebUI. Listening Address is 127.0.0.1:8384 insted of 0.0.0.0:8384

rondadon, 2020-01-10:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
`#!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'`

- Distro version | 
`buster 10.2`
- Kernel version | 
`Linux DietPi4 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux`
- SBC device | 
`RPi 4 Model B (armv7l)`
- Power supply used | 
`5V 4A Meanwell PSU`
- SDcard used | 
`SanDisk ultra 32GB for Boot` | `SSD with USB<->SATA Adapter for whole System`

#### Additional Information (if applicable)
- Software title | `Syncthing`
- Was the software title installed freshly or updated/migrated? `Freshly Installed`
- Can this issue be replicated on a fresh installation of DietPi? `It is a fresh Dietpi Installation`

#### Steps to reproduce

1. Update to DietPi v6.28.0
3. Install syncthing
4. start syncthing
5. Try to connect to the IP:PORT (eg. 192.168.1.100:8384) of your syncthing Instance (WebUI) on your DietpiServer with a WebBrowser.

#### Expected behaviour

- The WebUI of syncthing should load and show up in your WebBrowser

#### Actual behaviour

- It fails to connect. The Internet Browser can not load the WebUI of syncthing. 

#### Extra details

- 
I Changed `<address>127.0.0.1:8384</address>` to `<address>0.0.0.0:8384</address>` in  `/mnt/dietpi_userdata/syncthing/config.xml`.
Started syncthing with `systemctl status syncthing` and now the WebUI of the syncthing instance on my Dietpi is accessible and the internetBrowser is showing the WebUI.

Have a nice weekend
Ron!

Joulinar, 2020-01-10:
@rondadon 
I'm not able to confirm this. My `config.xml` looks good right from the beginning.

```
 <gui enabled="true" tls="false" debugging="false">
        <address>0.0.0.0:8384</address>
        <apikey>yTTYJzxxxx</apikey>
        <theme>default</theme>
    </gui>
```

MichaIng, 2020-01-11:
@rondadon 
Please see my reply in your other issue, since both issues are related: #3328
- Since config dir pre-creation failed, applying this setting with 0.0.0.0 fails.
- https://github.com/MichaIng/DietPi/blob/master/dietpi/dietpi-software#L10778

I mark this as closed since we only need to find out why the config dir pre-creation failed.

closed 2020-01-11 by MichaIng