MichaIng/DietPi #3331: WIFI Hotpot created networt cannot connect to the internet

Crasher124, 2020-01-11:
#### Required Information
- DietPi version |G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- SBC device | RPi 4 Model B (armv7l)
- Power supply used |  5V 1A
- SDcard used | Toshiba 4GB

#### Additional Information (if applicable)
- Software title | WIFI Hotspot
- Was the software title installed freshly or updated/migrated? no
- Can this issue be replicated on a fresh installation of DietPi? yes
30e3b511-95ac-4b3d-8465-6b9df23603d3
- Bug report ID | `30e3b511-95ac-4b3d-8465-6b9df23603d3`

#### Steps to reproduce
1.Install Dietpi
2. Install WIFI Hotpost and configure it
3. Notice that Going online on the Network it created does not work

#### Expected behaviour
- Be able to go online through the Network it created

#### Actual behaviour
- Im not able to go to the Internet through the Network it created

#### Extra details
- Ethernet is connected

Joulinar, 2020-01-12:
@Crasher124 
Many thanks for your report.

Some more questions on your issue:
1. once your are connect to your DietPi system using SSH, you are able to connect to the internet and `ping google.com` (using Ethernet)?
2. are you able to connect to the WiFi HotSpot SSID?
3. pls can you past output of following command's
```
systemctl status isc-dhcp-server
systemctl status hostapd
journalctl -u isc-dhcp-server
journalctl -u hostapd
```

Crasher124, 2020-01-16:
I dont know what happed but now it works

closed 2020-01-16 by Crasher124

symbios24, 2020-09-18:
I have a similar problem with latest dietpi on a raspberry pi in hotspot mode,it connects to the internet but sometimes in randomly moments there is no internet and i have to reboot the raspberry pi to have internet.

Joulinar, 2020-09-18:
what does it mean you don't have internet connection? Is this valid for Hotspot users only? Can you still login to your RPi via SSH and try to ping anything on the web?

symbios24, 2020-09-18:
Yes i can login everything works
It has internet but in random moments it seems that there is no internet i thing this happened after i updated dietpi to the latest version
When this hsppens i have to reboot the Rpi for the internet to come back on

Joulinar, 2020-09-18:
My question was related to the situation where you lost Internet connection > Can you still login to your RPi via SSH and try to ping anything on the web, if you have lost internet connection on the hotspot?

symbios24, 2020-09-18:
I can login with ssh what is the ping command?

symbios24, 2020-09-18:
I dont thing is my modem problem coz when i reboot the rpi the Internet works again

Joulinar, 2020-09-18:
do you have issues right now? Because this is something to test as soon as the Internet connection fails. If Internet connection is working currently, this will work as well. Important to test once your connection failed. Probably you are loosing WiFi connection between your RPi and your local network.

Ping command would be `ping google.com` 😉

symbios24, 2020-09-18:
Thanks i will try when it happens again 🙂
What about the up down to the speed,when i download something the speed is going up/down its not really stable.