MichaIng/DietPi #3334: [FAILED] No password entered, please try again...

lasers, 2020-01-12:
### Creating a bug report/issue

#### Required Information
- (Unrelated/Suggestion) Tell users to execute a script that prints things so we can copy and paste everything once (e.g.`DietPi version`, `Distro version`, `Kernel version`, `SBC device` and others).
- Maybe you could prompt users to manually enter power supply + SDcard in the script too?

#### Information
- Software title | `DietPi-Set_software`
- Was the software title installed freshly or updated/migrated? `Installed freshly`
- Can this issue be replicated on a fresh installation of DietPi? `Yes`

#### Steps to reproduce

1. Accepting changing a password by accident.
2. You already did this in `/Dietpi/dietpi.txt`
3. Wanting to cancel changing a password.
4. Infinite loop.

#### Expected behaviour

- Permit users to cancel changing a password or parsing `/Dietpi/dietpi.txt` to skip prompting user to change a password.

#### Actual behaviour

- Infinite loop of prompting user to enter a new password.

MichaIng, 2020-01-12:
@lasers 
Many thanks for your report.

Yeah indeed this due is the way the loop for checking identical inputs is currently coded,  not ideal. We'll enhance this to allow cancel from the first input field at least.

Workaround is obvious (I hope) to retype the same password you chose before.

MichaIng, 2025-05-23:
Took a _little_ while, but I finally applied this in combination with enforcing a password chance once on first boot, if the default is still applied.

closed 2025-05-25 by MichaIng