MichaIng/DietPi #3335: RetroPie Dualbooted/On Top with/of DietPi

Proxyl3ss, 2020-01-12:
So I got the new Raspberry pi 4B 2GB Ram Version and I want to use it as a file server and play retrogames on it too. So I have dietpi setup on a 32GB SD Card running docker with 3 containers. Now I want to play retrogames. I tried installing retropie on top of dietpi but I don't think that's possible. (if it is tell me). Is it any way I like dual boot retropie working buster image with dietpi without losing all my containers?

MichaIng, 2020-01-13:
@Proxyl3ss 
Actually RetroPie should work on DietPi. But probably some manual adjustment needs to be done, see: https://github.com/MichaIng/DietPi/issues/3090
When I find time, I want to re-test and in case re-add RetroPie to DietPi-Software. We had this inside before but, due to quite intrusive and Raspbian-only oriented installer removed it: https://github.com/MichaIng/DietPi/issues/1408#issuecomment-359196336

Proxyl3ss, 2020-01-13:
Well, I do the regular setup, I download retropie_setup I do the main core install and when it's done, the error I get is this:

```
root@DietPi:~# emulationstation
lvl0:   Error creating SDL window!
        Could not create GLES window surface
lvl0:   Renderer failed to initialize!
lvl0:   Window failed to initialize!
root@DietPi:~#
```

I have lxde open in front of me if that's a useful thing to add. Tell me if u need more info. I can test anything for you.

Joulinar, 2020-01-13:
I found something on the web on this error. so you are not the first one :)

https://www.raspberrypi.org/forums/viewtopic.php?t=171446#p1098020

not sure if this will help in your case but at least it seems the issue could be fixed following this thread

MichaIng, 2020-01-13:
Not sure what the installer does or expects yet. However as SDL with GLES is used, assure the following:
- `apt install libsdl2-2.0-0 libgles2 libgl1-mesa-dri`
- Enable OpenGL fKMS driver, e.g. on DietPi: `/DietPi/dietpi/func/dietpi-set_hardware rpi-opengl vc4-fkms-v3d`
- Also I am not sure which user/group emulationstation invokes, if any, but it needs to be member of the render group since Buster AFAIK: `usermod -aG render <emulationstation_user>`

Proxyl3ss, 2020-01-13:
Not sure about the last command you said, but I still get the same error.

MichaIng, 2020-01-13:
@Proxyl3ss 
Also did you raise the GPU memory split? RPi4 does not require that much anymore, but 16M default is definitely too low.

MichaIng, 2020-12-19:
I mark this as closed. Testing and in case re-adding RetroPie is one here: #3090

closed 2020-12-19 by MichaIng