MichaIng/DietPi #3336: Error installing nextcloud

lacedup, 2020-01-12:
ithis is a reinstall, I previously uninstalled nextcloud. I am attempting to install on a mounted hdd

  - Command: systemctl restart mariadb                                                    
                         │  - Exit code: 1                                                                          
                         │  - DietPi version: v6.28.0 (MichaIng/master) | HW_MODEL:4 | HW_ARCH:2 | DISTRO:5         
                         │  - Image creator: DietPi Core Team                                                       
                         │  - Pre-image: Raspbian Lite                                                              
                         │                                                                                          
                         │ Log file content:                                                                        
                         │ Job for mariadb.service failed because the control process exited with error code.       
                         │ See "systemctl status mariadb.service" and "journalctl -xe" for details.                 
                         │                                                                                          
                         │ If problems persist, please report at https://github.com/MichaIng/DietPi/issues for      
                         │ investigation.                                                                           
                         │                                                                                          
                         │ Unable to continue, DietPi-Software will now terminate.

Joulinar, 2020-01-12:
@lacedup 
Pls can you past the following 2 commands

```
systemctl status mariadb.service
journalctl -u mariadb.service
```

lacedup, 2020-01-12:
12 15:50:27 DietPi mysqld[28285]: 2020-01-12 15:50:27 0 [Warning] Can't create test file /var/lib/mysql/DietPi.lower-test
Jan 12 15:50:28 DietPi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Jan 12 15:50:28 DietPi systemd[1]: mariadb.service: Failed with result 'exit-code'.
Jan 12 15:50:28 DietPi systemd[1]: Failed to start MariaDB 10.3.17 database server.
--More--

Sent from my iPad

> On Jan 12, 2020, at 11:12 AM, Joulinar <notifications@github.com> wrote:
> 
> journalctl -u mariadb.service

lacedup, 2020-01-12:
Is there a way to just factory restore diet pi from the device?

Joulinar, 2020-01-12:
@lacedup 
pls can you past following command `ls -la /var/lib/`

> Is there a way to just factory restore diet pi from the device?

well you could reset your entire device be re-imaging your SD card with a new DietPi installation

MichaIng, 2020-01-13:
@lacedup 
> Can't create test file /var/lib/mysql/DietPi.lower-test
I have guess: Did you move or link your dietpi_userdata into some users /home/ directory?
See: https://github.com/MichaIng/DietPi/issues/3249#issuecomment-561867451

lacedup, 2020-01-13:
> See: #3249 (comment)

I just reinstalled dietpi fresh. Fixed

MichaIng, 2020-01-20:
Okay, I mark this as closed then. Feel free to reopen if required.

closed 2020-01-20 by MichaIng