MichaIng/DietPi #3338: ping: one.one.one.one: Temporary failure in name resolution

bgant, 2020-01-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | v6.28.0 (MichaIng/master)
- Distro version | bullseye (ID=6)
- Kernel version | Linux DietPi 5.5.0-rc2-g5704b3cc7ba3 #1 SMP PREEMPT Mon Jan 6 16:38:01 UTC 2020 aarch64 GNU/Linux
- SBC device | ROCK Pi 4 (aarch64) (ID=72)
- Power supply used | 5.0V 3.0A
- SDcard used | Seeed Studio 16GB eMMC

#### Details:
- Command | ping -c 1 -W 5 one.one.one.one
- Exit code | 2

#### Additional Information (if applicable)
- Software title | DietPi-Login
- Was the software title installed freshly or updated/migrated? Fresh
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce
```
git clone https://gitlab.collabora.com/rockpi/rockpi4
cd rockpi4
docker run --rm --interactive --tty --device /dev/kvm --workdir /recipes --mount "type=bind,source=$(pwd),destination=/recipes" --security-opt label=disable godebos/debos --scratchsize=8G rockpi4.yml
gzip debian-rockpi4.img
gunzip --stdout debian-rockpi4.img.gz | sudo dd of=/dev/sdh bs=512k conv=fsync status=progress
```
Boot Rock Pi 4 with eMMC containing debian-rockpi4.img.gz
Ctrl+Alt+F1
login to debian account
```
sudo su -
wget https://raw.githubusercontent.com/MichaIng/DietPi/master/PREP_SYSTEM_FOR_DIETPI.sh -O PREP_SYSTEM_FOR_DIETPI.sh
chmod a+x PREP_SYSTEM_FOR_DIETPI.sh
./PREP_SYSTEM_FOR_DIETPI.sh
```
DietPi-Login process fails to connect to one.one.one.one and on Exit from script it asks you to file an issue.

#### Expected behaviour
DNS names resolve properly

#### Actual behaviour
All DNS lookups fail. This appears to be because the /etc/resolv.conf file is a symbolic link to /etc/resolvconf/run/resolv.conf which does not exist.

Running the following commands fixes the issue:
```
rm /etc/resolv.conf
ln -s /run/resolvconf/resolv.conf /etc/resolv.conf
reboot
```

#### Additional logs:
```
ping: one.one.one.one: Temporary failure in name resolution
```

Joulinar, 2020-01-13:
@bgant 
Many thanks for your report. Looks like your issue is similar case as https://github.com/MichaIng/DietPi/issues/2979#issuecomment-572303697

It might be that the resolvconf service was not coming up correctly after you transformed your Debian installation into DietPi.

MichaIng, 2020-01-13:
@Joulinar 
The reason for the issue will be different, but jep `resolvconf` failed to init most likely.

@bgant 
Could you paste:
```
journalctl -u resolvconf
journalctl -u networking
journalctl -u ifup@eth0
```

I also see Debian testing bullseye (ID=6). Note that DietPi does not officially support Bullseye until it has been officially released in ~1.5 years 😉. There is much change, making our scripts breaking from time to time expectedly. I run my own server on Bullseye and run all test on a Bullseye VM as well, but a full test and related fixes will not be done until spring 2021. Also `Linux DietPi 5.5.0-rc2`, don't expect too much stability here, hence tinkering will be more often required for sure. The whole base image is a pure testing image.

The following image will work: https://dietpi.com/downloads/images/testing/DietPi_ROCKPi4-ARMv8-Buster.7z

bgant, 2020-01-14:
Thanks for the info. I debated whether or not to file an issue but decided to go ahead since the fix was fairly simple.

```
journalctl -u resolvconf
-- Logs begin at Mon 2020-01-13 23:05:37 GMT, end at Mon 2020-01-13 23:08:06 GMT. --
-- No entries --

journalctl -u networking
-- Logs begin at Mon 2020-01-13 23:05:37 GMT, end at Mon 2020-01-13 23:08:27 GMT. --
Jan 13 23:05:48 DietPi systemd[1]: Starting Raise network interfaces...
Jan 13 23:05:48 DietPi systemd[1]: Started Raise network interfaces.

journalctl -u ifup@eth0
-- Logs begin at Mon 2020-01-13 23:05:37 GMT, end at Mon 2020-01-13 23:11:54 GMT. --
-- No entries --
```
I will go ahead and close this issue.

closed 2020-01-14 by bgant