MichaIng/DietPi #3339: Auto mount+fstab using udev rules

hifitime, 2020-01-13:
echo 'ACTION=="add", KERNEL=="sd[a-z]", RUN+="/DietPi/dietpi/dietpi-drive_manager 3"' > /etc/udev/rules.d/99-dietpi-auto_mount.rules 
udevadm control --reload-rules  or reboot

hi MichaIng:

That's what I really need, but it doesn't work
Is there any other way?

I added rules to / etc / udev / rules. D /, which also didn't work...

`KERNEL!="sd[a-z][0-9]", GOTO="media_by_label_auto_mount_end"

# Import FS infos
IMPORT{program}="/sbin/blkid -o udev -p %N"

# Get a label if present, otherwise specify one
ENV{ID_FS_LABEL}!="", ENV{dir_name}="%E{ID_FS_LABEL}"
ENV{ID_FS_LABEL}=="", ENV{dir_name}="usbhd-%k"

# Global mount options
ACTION=="add", ENV{mount_options}="relatime"
# Filesystem-specific mount options
ACTION=="add", ENV{ID_FS_TYPE}=="vfat|ntfs", ENV{mount_options}="$env{mount_options},utf8,gid=100,umask=002"

# Mount the device
ACTION=="add", RUN+="/bin/mkdir -p /media/%E{dir_name}", RUN+="/bin/mount -o $env{mount_options} /dev/%k /media/%E{dir_name}"

# Clean up after removal
ACTION=="remove", ENV{dir_name}!="", RUN+="/bin/umount -l /media/%E{dir_name}", RUN+="/bin/rmdir /media/%E{dir_name}"

# Exit
LABEL="media_by_label_auto_mount_end"`

root@DietPi:~# df -h
Filesystem Size Used Avail Use% Mounted on
udev 484M 0 484M 0% /dev
tmpfs 99M 5.7M 94M 6% /run
/dev/mmcblk1p1 2.9G 1.9G 973M 67% /
tmpfs 495M 0 495M 0% /dev/shm
tmpfs 5.0M 4.0K 5.0M 1% /run/lock
tmpfs 495M 0 495M 0% /sys/fs/cgroup
tmpfs 10M 1.4M 8.7M 14% /DietPi
tmpfs 50M 20K 50M 1% /var/log
tmpfs 1023M 4.0K 1023M 1% /tmp

root@DietPi:~# lsblk
NAME MAJ:MIN RM SIZE RO TYPE MOUNTPOINT
sda 8:0 0 931.5G 0 disk
└─sda1 8:1 0 931.5G 0 part
mtdblock0 31:0 0 1.5M 0 disk
mtdblock1 31:1 0 64K 0 disk
mtdblock2 31:2 0 2.4M 0 disk
mmcblk0 179:0 0 3.7G 0 disk
└─mmcblk0p1 179:1 0 3.7G 0 part
mmcblk0boot0 179:8 0 2M 1 disk
mmcblk0boot1 179:16 0 2M 1 disk
mmcblk1 179:24 0 29.7G 0 disk
└─mmcblk1p1 179:25 0 3G 0 part /

MichaIng, 2020-01-13:
@hifitime 
As mentioned in the other issue, this feature has not yet been implemented. Drive manager exits with failure when executing with argument 3. But what could actually work is:
```
G_DEBUG=1 /DietPi/dietpi/dietpi-drive_manager
```
This runs the drive detection loop but then prints information about all detected drives only and exits. But the fstab creation/update is done within the loop, AFAIK.

hifitime, 2020-01-13:
OK, I get it..

in addition，I added rules to /etc/udev/rules.d/， which also didn't work...It looks like the rules are right..

nano /etc/udev/rules.d/media_by_label_auto_mount

`KERNEL!="sd[a-z][0-9]", GOTO="media_by_label_auto_mount_end"
IMPORT{program}="/sbin/blkid -o udev -p %N"
ENV{ID_FS_LABEL}!="", ENV{dir_name}="%E{ID_FS_LABEL}"
ENV{ID_FS_LABEL}=="", ENV{dir_name}="usbhd-%k"
ACTION=="add", ENV{mount_options}="relatime"
ACTION=="add", ENV{ID_FS_TYPE}=="vfat|ntfs", ENV{mount_options}="$env{mount_options},utf8,gid=100,umask=002"
ACTION=="add", RUN+="/bin/mkdir -p /media/%E{dir_name}", RUN+="/bin/mount -o $env{mount_options} /dev/%k /media/%E{dir_name}"
ACTION=="remove", ENV{dir_name}!="", RUN+="/bin/umount -l /media/%E{dir_name}", RUN+="/bin/rmdir /media/%E{dir_name}"
LABEL="media_by_label_auto_mount_end"
`

MichaIng, 2020-01-13:
@hifitime 
Looks a bid complicated but should work. Did you run `udevadm control -R` or rebooted to enable the new rules?

hifitime, 2020-01-13:
udevadm control --reload-rules or reboot
Have tried, but still can't run, can't see any mount..

MichaIng, 2020-01-13:
@hifitime 
Does `dmesg` or `journalctl -r` show some related logs when you plug and/or unplug a drive?
In case increase the udev log level.
```
udevadm control --log-priority=info
```
Or to actively print to console
```
udevadm monitor
```

Ahh there is even a test command to simulate a certain event and print debug log:
```
udevadm test -a add /dev/sda1
```

hifitime, 2020-01-14:
Hi michaing, 

when I use udevadm monitor, I can see that after I insert USB, the console has information output, but when I run udevadm test - a add / dev / sda1, there is an error message:

`root@DietPi:/sys/block/sda/sda1# udevadm test -a add /dev/sda1
This program is for debugging only, it does not run any program
specified by a RUN key. It may show incorrect results, because
some values may be different, or not available at a simulation run.

Load module index
Skipping empty file: /etc/systemd/network/99-default.link
Created link configuration context.
Reading rules file: /etc/udev/rules.d/10-wifi-disable-powermanagement.rules
Reading rules file: /lib/udev/rules.d/50-firmware.rules
Reading rules file: /lib/udev/rules.d/50-udev-default.rules
Reading rules file: /lib/udev/rules.d/55-dm.rules
Reading rules file: /lib/udev/rules.d/60-block.rules
Reading rules file: /lib/udev/rules.d/60-cdrom_id.rules
Reading rules file: /lib/udev/rules.d/60-drm.rules
Reading rules file: /lib/udev/rules.d/60-evdev.rules
Reading rules file: /lib/udev/rules.d/60-input-id.rules
Reading rules file: /lib/udev/rules.d/60-persistent-alsa.rules
Reading rules file: /lib/udev/rules.d/60-persistent-input.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage-dm.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage-tape.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage.rules
Reading rules file: /lib/udev/rules.d/60-persistent-v4l.rules
Reading rules file: /lib/udev/rules.d/60-sensor.rules
Reading rules file: /lib/udev/rules.d/60-serial.rules
Reading rules file: /lib/udev/rules.d/60-sunxi-tools.rules
Reading rules file: /lib/udev/rules.d/64-btrfs.rules
Reading rules file: /lib/udev/rules.d/70-joystick.rules
Reading rules file: /lib/udev/rules.d/70-mouse.rules
Reading rules file: /lib/udev/rules.d/70-power-switch.rules
Reading rules file: /lib/udev/rules.d/70-touchpad.rules
Reading rules file: /lib/udev/rules.d/70-uaccess.rules
Reading rules file: /lib/udev/rules.d/71-axp-power-button.rules
Reading rules file: /lib/udev/rules.d/71-seat.rules
Reading rules file: /lib/udev/rules.d/73-seat-late.rules
Reading rules file: /lib/udev/rules.d/73-special-net-names.rules
Reading rules file: /lib/udev/rules.d/73-usb-net-by-mac.rules
Reading rules file: /lib/udev/rules.d/75-net-description.rules
Reading rules file: /lib/udev/rules.d/75-probe_mtd.rules
Reading rules file: /lib/udev/rules.d/78-sound-card.rules
Reading rules file: /lib/udev/rules.d/80-debian-compat.rules
Reading rules file: /lib/udev/rules.d/80-drivers.rules
Reading rules file: /lib/udev/rules.d/80-ifupdown.rules
Reading rules file: /lib/udev/rules.d/80-net-setup-link.rules
Reading rules file: /lib/udev/rules.d/85-hdparm.rules
Reading rules file: /lib/udev/rules.d/85-hwclock.rules
Reading rules file: /lib/udev/rules.d/89-alsa-ucm.rules
Reading rules file: /lib/udev/rules.d/90-alsa-restore.rules
Reading rules file: /lib/udev/rules.d/90-console-setup.rules
Reading rules file: /lib/udev/rules.d/95-dm-notify.rules
Reading rules file: /etc/udev/rules.d/99-dietpi-auto_mount.rules
Reading rules file: /lib/udev/rules.d/99-systemd.rules
Rules contain 24576 bytes tokens (2048 * 12 bytes), 13161 bytes strings
1842 strings (23191 bytes), 1207 de-duplicated (10666 bytes), 636 trie nodes used
Failed to open device '/sys/dev/sda1': No such device
Unload module index
Unloaded link configuration context.`

Joulinar, 2020-01-14:
@hifitime 
glad that the problem could be solved. Would it be possible to share the steps you have taken to get it solved? It might be other are interested in it as well 😉

hifitime, 2020-01-14:
I'm so happy too early... The problem is still there...

root@DietPi:~# udevadm test -a add /dev/sda1
This program is for debugging only, it does not run any program
specified by a RUN key. It may show incorrect results, because
some values may be different, or not available at a simulation run.

Load module index
Skipping empty file: /etc/systemd/network/99-default.link
Created link configuration context.
Reading rules file: /etc/udev/rules.d/10-wifi-disable-powermanagement.rules
Reading rules file: /etc/udev/rules.d/11-automount.rules
Reading rules file: /lib/udev/rules.d/50-firmware.rules
Reading rules file: /lib/udev/rules.d/50-udev-default.rules
Reading rules file: /lib/udev/rules.d/55-dm.rules
Reading rules file: /lib/udev/rules.d/60-block.rules
Reading rules file: /lib/udev/rules.d/60-cdrom_id.rules
Reading rules file: /lib/udev/rules.d/60-drm.rules
Reading rules file: /lib/udev/rules.d/60-evdev.rules
Reading rules file: /lib/udev/rules.d/60-input-id.rules
Reading rules file: /lib/udev/rules.d/60-persistent-alsa.rules
Reading rules file: /lib/udev/rules.d/60-persistent-input.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage-dm.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage-tape.rules
Reading rules file: /lib/udev/rules.d/60-persistent-storage.rules
Reading rules file: /lib/udev/rules.d/60-persistent-v4l.rules
Reading rules file: /lib/udev/rules.d/60-sensor.rules
Reading rules file: /lib/udev/rules.d/60-serial.rules
Reading rules file: /lib/udev/rules.d/60-sunxi-tools.rules
Reading rules file: /lib/udev/rules.d/64-btrfs.rules
Reading rules file: /lib/udev/rules.d/70-joystick.rules
Reading rules file: /lib/udev/rules.d/70-mouse.rules
Reading rules file: /lib/udev/rules.d/70-power-switch.rules
Reading rules file: /lib/udev/rules.d/70-touchpad.rules
Reading rules file: /lib/udev/rules.d/70-uaccess.rules
Reading rules file: /lib/udev/rules.d/71-axp-power-button.rules
Reading rules file: /lib/udev/rules.d/71-seat.rules
Reading rules file: /lib/udev/rules.d/73-seat-late.rules
Reading rules file: /lib/udev/rules.d/73-special-net-names.rules
Reading rules file: /lib/udev/rules.d/73-usb-net-by-mac.rules
Reading rules file: /lib/udev/rules.d/75-net-description.rules
Reading rules file: /lib/udev/rules.d/75-probe_mtd.rules
Reading rules file: /lib/udev/rules.d/78-sound-card.rules
Reading rules file: /lib/udev/rules.d/80-debian-compat.rules
Reading rules file: /lib/udev/rules.d/80-drivers.rules
Reading rules file: /lib/udev/rules.d/80-ifupdown.rules
Reading rules file: /lib/udev/rules.d/80-net-setup-link.rules
Reading rules file: /lib/udev/rules.d/85-hdparm.rules
Reading rules file: /lib/udev/rules.d/85-hwclock.rules
Reading rules file: /lib/udev/rules.d/89-alsa-ucm.rules
Reading rules file: /lib/udev/rules.d/90-alsa-restore.rules
Reading rules file: /lib/udev/rules.d/90-console-setup.rules
Reading rules file: /lib/udev/rules.d/95-dm-notify.rules
Reading rules file: /lib/udev/rules.d/99-systemd.rules
Rules contain 24576 bytes tokens (2048 * 12 bytes), 13443 bytes strings
1848 strings (23487 bytes), 1210 de-duplicated (10683 bytes), 639 trie nodes used
Failed to open device '/sys/dev/sda1': No such device
Unload module index
Unloaded link configuration context.
root@DietPi:~# lsblk
NAME         MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
sda            8:0    1  7.2G  0 disk 
└─sda1         8:1    1  7.2G  0 part 
mtdblock0     31:0    0  1.5M  0 disk 
mtdblock1     31:1    0   64K  0 disk 
mtdblock2     31:2    0  2.4M  0 disk 
mmcblk0      179:0    0  3.7G  0 disk 
└─mmcblk0p1  179:1    0  3.7G  0 part 
mmcblk0boot0 179:8    0    2M  1 disk 
mmcblk0boot1 179:16   0    2M  1 disk 
mmcblk1      179:24   0 29.7G  0 disk 
└─mmcblk1p1  179:25   0    3G  0 part /

MichaIng, 2020-01-14:
@hifitime 
The output shows that there is at least no syntax issue with the udev rule. But the path was not given correctly, so no rules are applied. Please try:
```
udevadm test -a add $(udevadm info -q path -n /dev/sda1)
```

MichaIng, 2020-02-24:
For reference: https://dietpi.com/phpbb/viewtopic.php?t=5106