MichaIng/DietPi #3340: Dietpi cant sync time with NTP

FrostyMisa, 2020-01-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | #!/bin/bash
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster
- Kernel version | Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- SBC device | RPi 4 Model B (armv7l)
- Power supply used | Original: 5V 3A
- SDcard used | SanDisk ultra - use before with raspberry 3b+ and works (i do format and burn new image from today (1.14.2020)

#### Additional Information (if applicable)
- Software title | clean install

#### Steps to reproduce
Downloaded new image today (DietPi_RPi-ARMv6-Buster), burn with balena etcher like everytime, put in my raspberry and after start get this:
 ─────────────────────────────────────────────────────
 DietPi v6.28.0 : 01:43 - Thu 26/09/19
 ─────────────────────────────────────────────────────
 - LAN IP : 192.168.1.142 (eth0)
[  OK  ] DietPi-Login | Checking network connectivity
[  OK  ] DietPi-Login | Checking DNS resolver
[  OK  ] DietPi-Run_NTPD | systemctl restart systemd-timesyncd
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (1/60)
Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.
[ INFO ] DietPi-Run_NTPD | NTPD: Waiting for completion of systemd-timesyncd (2/60)
etc etc.

#### Expected behaviour
It will work like everytime i do it before (many times without problem, but last time i burn image with number, i thin 6.25? It was here for download maybe 10 days ago and it works)

#### Actual behaviour
I let it running many time till end (60 second) and nothing, do many restarts, nothing. write "systemctl daemon-reload" do nothing. When i write "systemctl restart systemd-timesyncd" i get same warning (Warning: The unit file, source configuration file or drop-ins of systemd-timesyncd.service changed on disk. Run 'systemctl daemon-reload' to reload units.)
Ping work normal, I can ping everything. But without time sync I can do nothing. 
Here how it looks:
root@DietPi:~# apt-get update
Get:1 http://raspbian.raspberrypi.org/raspbian buster InRelease [15.0 kB]
Get:2 https://archive.raspberrypi.org/debian buster InRelease [25.1 kB]                   
Reading package lists... Done                            
E: Release file for http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease is not valid yet (invalid for another 110d 15h 59min 22s). Updates for this repository will not be applied.
E: Release file for https://archive.raspberrypi.org/debian/dists/buster/InRelease is not valid yet (invalid for another 110d 15h 23min 26s). Updates for this repository will not be applied.

#### Extra details
after let it go till 60, i can change NTP server and if I click it i get:
DietPi-Config will now be launched, on the next screen:                                                                                                      
  - Select "NTP Mirror" 
 - Select a different NTP mirror 
  Once completed, exit dietpi-config to resume setup
I click ok and get:
[ INFO ] First run setup has not reached sufficient state.  Please reboot before using DietPi-Config. If the issue persists, please report this as bug.
Funny, cant set anything.

But I can click:
Override   : (NOT RECOMMENDED) Ignore failure and override time sync status
After this i can change my password and continue till dietpi software screen.

And after this all i listen you and try pool.ntp.org and it WORKS! I think i try everythink around, I try geteway, it work yesterday and before with raspberry 3b+, i try others, like europe etc, nothing. This work and now looks its close for me, but somewhere must be some problem, because i never have this problem and everytime i install it I use default and after clean install i start to make some change.
Thanks for advice with pool.ntp.org, but dont know if this exacly help, but I wrote you here exacly what i do ;)

Joulinar, 2020-01-14:
@FrostyMisa 
Many thanks for your report. One small question: What different NTP mirror you used and was not working? I mean the one before you switched to pool.ntp.org

MichaIng, 2020-01-14:
Yes `[0-3].debian.pool.ntp.org` is the default, if no explicit server has been assigned (default, see `/etc/systemd/timesyncd.conf`). This is reached fine here 🤔. However grat that `pool.ntp.org` finally worked.

MichaIng, 2020-01-20:
Okay I mark this as closed. Still strange that system defaults did not work, but seems to be some edge case. Please reopen if required.

closed 2020-01-20 by MichaIng