MichaIng/DietPi #3341: dietpi.com seems to be down

Joulinar, 2020-01-14:
@MichaIng 
I'm getting again the Cloudflare's Always Online message

`Error 522 - Connection timed out` 😠

bbsixzz, 2020-01-15:
Works great from Sweden, must be local.

Atticuss, 2020-01-15:
Not local -- confirmed issue for myself as well. Philadelphia, PA.

MichaIng, 2020-01-15:
Hey guys, I updated phpBB3 which should usually go quick and smooth, but me dump forget to disable extensions and clear cache first and two extensions were incompatible (though very easy to fix), leading to phpBB error. However has been resolved meanwhile.

But hmm, `522` @Joulinar @Atticuss? It should have been `500` internal server error, when accessing the forum only 🤔. However can you confirm its working again?

Joulinar, 2020-01-15:
Nope its already on the main web side dietpi.com, not the forum. Seems similar as last time. There as well it was depending on the location.

MichaIng, 2020-01-15:
@Joulinar 
You're right, same symptoms on server as well, cannot connect/ping certain resources... what a coincidence, I was just happy to have phpBB update finished and now this. Good that we already have a ticket open on VPS provider, I'll reopen it, hopefully we get some investigation this time.

MichaIng, 2020-01-15:
It seems like we were attached, brute-force attempt and DDoS.

Starting with `23:11:42 UTC` I see a bunch of SSH connection attempts. First ~5 times around 10 each within the same second, before blocked by fail2ban, then 20 to 30 each from different IP, hence fail2ban did not trigger, all ending with "`SYN flooding on port 443`".
Now I am wondering, why port 443, since this is not the SSH port. Checking Cloudflare statistics:
- On a usual **day** we have **~300 tausend** requests.
- Today within the one matching **hour** we have **3.6 millions** requests!
- Sadly non of them were detected/blocked by Cloudflare...

So I guess some mechanism on the VPS network blocked the related Cloudflare IPs where those requests have gone through. Through their GUI I see some DDoS protection firewall but the "disable" button does not work 🤔. However have to ask Cloudflare why this was not blocked in the first place, perhaps we have to place/change some firewall/filter rules.

Joulinar, 2020-01-15:
@MichaIng 
yep we are back online. Thx for investigating it that late during night. I hope it did not damage anything on the servers.

Atticuss, 2020-01-15:
Confirmed back up for myself as well. Also worth noting that Cloudflare won't protect you if the scanner was hitting your IP directly -- they'll only be able to see the attack if the scanner is hitting you by domain name, as then all traffic gets routed through CF itself for them to see. Probably just an internet-wide scanner that saw 22 open and threw it's usual giant list of brute-force attempts at it. That's on you to protect against.

https://support.cloudflare.com/hc/en-us/articles/200170166-Best-Practices-DDoS-preventative-measures

MichaIng, 2020-01-15:
@Atticuss 
As said, the SYN flooding was on port 443, also since I found it in the Cloudflare logs, it indeed was proxied through Cloudflare, from all I know. Port 22 requests on the Cloudflare proxied domain should be blocked, but we have bypassing domain as well for SSH of course. I'll ask Cloudflare about this, probably they have some more information and suggestions how to protect against such attacks.

tanderson1992, 2020-01-15:
I hope this is the appropriate place to report. I'm a new user and tried to sign up for the DietPi forum so that I could ask a question. After filling in the registration information, nothing happens when I press the submit button. After several tries, I noticed a box in the lower right corner. Hovering over the box expands the message, "This site key is not enabled for the invisible captcha." I'm guessing the registration doesn't submit because the invisible captcha isn't working properly.

MichaIng, 2020-01-16:
@tanderson1992 
Many thanks for your report. Looks like the reCAPTCHA keys faded out or something, I just created and applied new ones and for me the reCAPTCHA icon appears again. Can you please try again to register?

closed 2020-01-17 by MichaIng

tanderson1992, 2020-01-17:
Registration works now, thank you!

MichaIng, 2020-01-17:
@tanderson1992
Can you please try it again. While investigating the issue, I enabled IPv6, but mail provider blocked that new address by default... I just unlocked it.

tanderson1992, 2020-01-17:
Thank you. I can't register because my name is already in use. Is there a way to ask it to re-send the activation email, or can you delete user tanderson92 and let me try again?

MichaIng, 2020-01-17:
@tanderson1992 
I just triggered a reminder mail for all newly registered accounts, which include the activation link.

tanderson1992, 2020-01-17:
That worked, thank you!

MichaIng, 2020-10-05:
Lol, we are working for the devil:
```
- Uptime : up 6 weeks, 6 days, 6 hours, 46 minutes
```
Just the `4` doesn't fit, dammit 😄.

Joulinar, 2020-10-05:
Software from hell 😆