MichaIng/DietPi #3343: Rockpi4 eMMC mmc_select_hs400es failed

bgant, 2020-01-15:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=6
G_DIETPI_VERSION_SUB=28
G_DIETPI_VERSION_RC=0
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster
- Kernel version | Linux DietPi 5.4.8-rockchip64 #19.11.7 SMP PREEMPT Sun Jan 5 23:30:10 CET 2020 aarch64 GNU/Linux
- SBC device | ROCK Pi 4 (aarch64)
- Power supply used | 5V 3A
- SDcard used | Seeed Studio 16GB eMMC

#### Additional Information (if applicable)
- Software title | boot
- Was the software title installed freshly or updated/migrated? Fresh
- Can this issue be replicated on a fresh installation of DietPi? Yes

#### Steps to reproduce

1. Write non-Radxa third-party image (i.e. DietPI, Armbian, Manjaro) to eMCC
2. Boot Rock Pi 4 with eMMC plugged in
3. No video, pings respond, but no network services (i.e. SSH)

#### Expected behaviour

- Device should boot, display to screen via HDMI, SSH available across network

#### Actual behaviour

- With a USB-to-TTL cable connected, you can see through the serial console that the eMMC has errors during boot:
```
[   48.357047] mmc2: mmc_select_hs400es failed, error -84
[   48.357517] mmc2: error -84 whilst initialising MMC card
[   48.552501] mmc2: mmc_select_hs400es failed, error -110
[   48.552988] mmc2: error -110 whilst initialising MMC card
```

#### Extra details

- This issue is discussed on the Armbian and Radxa forums:
-- https://forum.armbian.com/topic/12573-armbian-buster-current-with-linux-54y-on-the-rock-pi-4/
-- https://forum.radxa.com/t/armbian-buster-5-4-6-emmc-slot-not-being-detected/2422/2
- Changing /boot/dtb/rockchip/rk3399-rock-pi-4.dtb to use mmc-hs200 instead of mmc-hs400 fixes the issue and boots properly

```
dtc -I dtb -O dts -f rk3399-rock-pi-4.dtb -o rk3399-rock-pi-4.dts
-       mmc-hs400-1_8v;
-       mmc-hs400-enhanced-strobe;
+       mmc-hs200-1_8v;
dtc -I dts -O dtb -f rk3399-rock-pi-4.dts -o rk3399-rock-pi-4.dtb
```

- DietPi_ROCKPi4-ARMv8-Buster.7z boots, automatically upgrades, reboots, and you need to apply the dtb change again.
- DietPi_ROCKPi4-ARMv8-Buster_new.7z boots, does not need to be upgraded, and continues to work after a reboot.
- This is a new Rock Pi 4A that was shipped to me last week. I have another Rock Pi 4A that is a few months older that works fine. I could not get either of my 16GB eMMC to work on the new device, so I'm guessing the eMMC interface on the new board has changed?

MichaIng, 2020-01-15:
@bgant 
Many thanks for your report/information. I just moved the `DietPi_ROCKPi4-ARMv8-Buster_new.7z` into stable place and will update the link + remove beta plag for this device.

> DietPi_ROCKPi4-ARMv8-Buster_new.7z boots, does not need to be upgraded, and continues to work after a reboot.

So I understand right, that the new image has the required patch applied already?

> This is a new Rock Pi 4A that was shipped to me last week. I have another Rock Pi 4A that is a few months older that works fine. I could not get either of my 16GB eMMC to work on the new device, so I'm guessing the eMMC interface on the new board has changed?

From what I see model A only difference is missing BT/WiFi and PoE, no word about eMMC: https://wiki.radxa.com/Rockpi4/hardware/revision
Did you try to flash the prior DietPi_ROCKPi4-ARMv8-Buster_new.7z (now DietPi_ROCKPi4-ARMv8-Buster.7z) directly onto the eMMC, e.g. via eMMC to USB or SD adapter and boot without any SDcard attached?

bgant, 2020-01-15:
No, neither DietPi_ROCKPi4-ARMv8-Buster.7z nor DietPi_ROCKPi4-ARMv8-Buster_new.7z work without changing the dtb file. I was just pointing out that if there is an update and the device reboots, you have to apply the change to the dtb file again.

Correct that model A doesn't have BT/WiFI/PoE. Everything else appears to be the same as a model B.

I have only been using eMMC with these devices.

MichaIng, 2020-01-16:
@bgant 
Hmm okay, not sure whether it make sense for us to take action for the eMMC issue. We can patch an `/etc/kernel/postinst.d/` script to apply the patch automatically after kernel upgrades (which usually come along with dtb package upgrades), but this would be quite late with next release, where hopefully it has been fixed package-wise already.

But the information is very helpful of course, as well as the workaround.

obbardc, 2021-02-19:
Was this ever forwarded to linux-rockchip mailing list?

MichaIng, 2021-02-19:
With the latest Armbian Linux 5.10 this is still an issue?

obbardc, 2021-02-19:
I am using 5.9.8, can you point me to the patch in Armbian or kernel which fixes this? :-)

obbardc, 2021-02-19:
Ah, possibly: https://github.com/armbian/build/blob/master/patch/kernel/rockchip64-current/board-rockpi4-FixMMCFreq.patch

MichaIng, 2021-02-19:
This means `apt update; apt full-upgrade` might fix it.

MichaIng, 2021-11-09:
Is this still an issue with the latest Bullseye image?

MichaIng, 2022-12-02:
Should be fixed with current kernel, marking this as closed.

closed 2022-12-02 by MichaIng