MichaIng/DietPi #3345: error after Nextcloud installation

Ewies, 2020-01-17:
Hello I restarted the system after a Nextcloud installation but then the system no longer starts up.

I can only access it via a screen and keyboard.

I get that as a root login error...

-bash: /DietPi/dietpi/func/dietpi-globals: Datei oder Verzeichnis nicht gefunden
[FAILED] DietPi-Login | Failed to load DietPi-Globals. Skipping DietPi login scripts...

What can I do now?

MichaIng, 2020-01-17:
@Ewies 
Many thanks for your report.

Obviously the RAMdisk failed to initialise. Is this on an RPi? Please always fill the bug report template when reporting issues 😉.

However please paste the following command outputs (or photo from screen, what is handy for you), which should give a clue:
```
journalctl -u boot.mount
journalctl -u DietPi.mount
journalctl -u dietpi-ramdisk
tail -4 /var/tmp/dietpi/logs/dietpi-ramdisk.log
```

Ewies, 2020-01-23:
I have now completely reinstalled the system, so unfortunately I can no longer deliver these issues.

Thanks anyway

closed 2020-01-23 by Ewies