MichaIng/DietPi #3346: SABnzbd install fails due to deprecated python 2.7.16

tanderson1992, 2020-01-17:
### Creating a bug report/issue

#### Required Information
- DietPi version 6.28
- Distro version buster
- Kernel version Linux DietPi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
- SBC device RPi 4 Model B (armv7l)
- Power supply used CanaKit 5.1V USB Power Supply
- SDcard used Sandisk Ultra

#### Additional Information (if applicable)
- Software title SABnzbd
- Was the software title installed freshly or updated/migrated? freshly
- Can this issue be replicated on a fresh installation of DietPi? yes, the DietPi install was also fresh

#### Steps to reproduce
1. Use dietpi-software to install SABnzbd

#### Expected behaviour

- SABnzbd should install and configure

#### Actual behaviour

- The install seems to be using deprecated python 2.7.16 instead of 3.7. Both 2.7 and 3.7 directories are in /usr/local/lib. This results in the following error:

 ```
ERROR: Command errored out with exit status 1:
   command: /usr/bin/python /usr/local/lib/python2.7/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-fYGRQm/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; platform_python_implementation != '"'"'PyPy'"'"''
       cwd: None
  Complete output (5 lines):
  DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=40.6.0
    Using cached https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.py3-none-any.whl
  ERROR: Package 'setuptools' requires a different Python: 2.7.16 not in '>=3.5'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python /usr/local/lib/python2.7/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-fYGRQm/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; platform_python_implementation != '"'"'PyPy'"'"'' Check the logs for full command output.
```

#### Extra details

- I've tried dietpi-update 1 and then reinstalling SABnzbd. It results in the same error and then hangs at the following prompt:
```

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Configuring SABnzbd: nzb download manager

[ INFO ] DietPi-Software | Generating initial config, please wait...
```
The SABnzbd service also shows a status of FAILED, and it won't start, most likely due to the initial install failure.

tanderson1992, 2020-01-17:
In case it helps:
```
root@DietPi:/home/dietpi# python --version
Python 2.7.16
root@DietPi:/home/dietpi# python3.7 --version
Python 3.7.3
```

MichaIng, 2020-01-17:
@tanderson1992 
Many thanks for your report. I just rechecked the documentation and found that SABnzbd does not support Python3: https://sabnzbd.org/wiki/installation/install-off-modules
> SABnzbd requires Python 2.7.
Python 3.x is not supported. Check python version with python --version.
Also Python 2.7 is not the reason why install fails, it is just a warning.

The error is that for a strange reason the setuptools package pulled by Python2 `pip` command has a dependency on Python3. For Python3 there is actually a different command `pip3`, while modules installed with `pip` must of course always depend on Python2. So this is a pip repo bug. I'll check if I can replicate this on Debian Buster.

tanderson1992, 2020-01-17:
Good to know, thank you for checking it out so quickly.

MichaIng, 2020-01-17:
@tanderson1992 
Hmm it succeeds on x86_64 Buster:
```
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting pip
  Downloading https://files.pythonhosted.org/packages/00/b6/9cfa56b4081ad13874b0c6f96af8ce16cfbc1cb06bedf8e9164ce5551ec1/pip-19.3.1-py2.py3-none-any.whl (1.4MB)
     |████████████████████████████████| 1.4MB 2.1MB/s
Collecting setuptools
  Downloading https://files.pythonhosted.org/packages/f9/d3/955738b20d3832dfa3cd3d9b07e29a8162edb480bf988332f5e6e48ca444/setuptools-44.0.0-py2.py3-none-any.whl (583kB)
     |████████████████████████████████| 583kB 3.3MB/s
Collecting wheel
  Downloading https://files.pythonhosted.org/packages/00/83/b4a77d044e78ad1a45610eb88f745be2fd2c6d658f9798a15e384b7d57c9/wheel-0.33.6-py2.py3-none-any.whl
Installing collected packages: pip, setuptools, wheel
Successfully installed pip-19.3.1 setuptools-44.0.0 wheel-0.33.6
```
- This is part of the `Installing Python Pip` block.

In the `Installing SABnzbd` block its:
```
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting cheetah
  Downloading https://files.pythonhosted.org/packages/cd/b0/c2d700252fc251e91c08639ff41a8a5203b627f4e0a2ae18a6b662ab32ea/Cheetah-2.4.4.tar.gz (190kB)
     |████████████████████████████████| 194kB 1.3MB/s
Collecting cryptography
  Downloading https://files.pythonhosted.org/packages/c8/52/ad7f2cbe3b6e9340526dc401b38bb67c06160449446e15b0bf5f947fa168/cryptography-2.8-cp27-cp27mu-manylinux2010_x86_64.whl (2.3MB)
     |████████████████████████████████| 2.3MB 3.1MB/s
Collecting sabyenc
  Downloading https://files.pythonhosted.org/packages/17/6c/534b5723dbb14be24754565a7d2e0a53f9232bac712f2cd95fc8a62a0761/sabyenc-3.3.6-cp27-cp27mu-manylinux1_x86_64.whl
Collecting Markdown>=2.0.1
  Downloading https://files.pythonhosted.org/packages/c0/4e/fd492e91abdc2d2fcb70ef453064d980688762079397f779758e055f6575/Markdown-3.1.1-py2.py3-none-any.whl (87kB)
     |████████████████████████████████| 92kB 3.7MB/s
Collecting six>=1.4.1
  Downloading https://files.pythonhosted.org/packages/65/eb/1f97cb97bfc2390a276969c6fae16075da282f5058082d4cb10c6c5c1dba/six-1.14.0-py2.py3-none-any.whl
Collecting cffi!=1.11.3,>=1.8
  Downloading https://files.pythonhosted.org/packages/93/5d/c4f950891251e478929036ca07b22f0b10324460c1d0a4434c584481db51/cffi-1.13.2-cp27-cp27mu-manylinux1_x86_64.whl (384kB)
     |████████████████████████████████| 389kB 1.7MB/s
Collecting ipaddress; python_version < "3"
  Downloading https://files.pythonhosted.org/packages/c2/f8/49697181b1651d8347d24c095ce46c7346c37335ddc7d255833e7cde674d/ipaddress-1.0.23-py2.py3-none-any.whl
Collecting enum34; python_version < "3"
  Downloading https://files.pythonhosted.org/packages/c5/db/e56e6b4bbac7c4a06de1c50de6fe1ef3810018ae11732a50f15f62c7d050/enum34-1.1.6-py2-none-any.whl
Requirement already satisfied: setuptools>=36 in /usr/local/lib/python2.7/dist-packages (from Markdown>=2.0.1->cheetah) (44.0.0)
Collecting pycparser
  Downloading https://files.pythonhosted.org/packages/68/9e/49196946aee219aead1290e00d1e7fdeab8567783e83e1b9ab5585e6206a/pycparser-2.19.tar.gz (158kB)
     |████████████████████████████████| 163kB 2.5MB/s
Building wheels for collected packages: cheetah, pycparser
  Building wheel for cheetah (setup.py) ... done
  Created wheel for cheetah: filename=Cheetah-2.4.4-cp27-none-any.whl size=171053 sha256=1c7855f90d58552b6156412eb358682a376eca92f4a80ff160bdde0552b6d374
  Stored in directory: /root/.cache/pip/wheels/27/8d/d0/7cee58a5a59f3c305cb1117416530229231440293e565030ff
  Building wheel for pycparser (setup.py) ... done
  Created wheel for pycparser: filename=pycparser-2.19-py2.py3-none-any.whl size=111027 sha256=d7c60cf494c1f1d73da9f1f2665d5aed4d0b6bd5a2bc8bc7b03ed031c70b4ad4
  Stored in directory: /root/.cache/pip/wheels/f2/9a/90/de94f8556265ddc9d9c8b271b0f63e57b26fb1d67a45564511
Successfully built cheetah pycparser
Installing collected packages: Markdown, cheetah, six, pycparser, cffi, ipaddress, enum34, cryptography, sabyenc
Successfully installed Markdown-3.1.1 cffi-1.13.2 cheetah-2.4.4 cryptography-2.8 enum34-1.1.6 ipaddress-1.0.23 pycparser-2.19 sabyenc-3.3.6 six-1.14.0
```
> Requirement already satisfied: setuptools>=36

Hence no other setuptools is pulled, as 44.0.0 has been installed before.

Could you paste your two blocks? In your case its somehow pulling setuptools 45.0.0 (not available in my case) as a dependency, which is as well different in my case. On ARM and x86_64 things are can be different, since on ARM most modules need to be compiled.

As a quick fix, actually the following should work:
```
apt install python-setuptools
dietpi-software reinstall 139
```
The APT package ships v40.8.0, hence pip does not need to pull any other version.

tanderson1992, 2020-01-19:
Here's the pip install:
```
 Mode: Installing Python Pip: python pip package installer

[  OK  ] DietPi-Software | Connection test: https://bootstrap.pypa.io/get-pip.py
[ INFO ] DietPi-Software | APT installation for: python python-dev, please wait...
[  OK  ] DietPi-Software | G_AGI python python-dev
--2020-01-18 19:52:01--  https://bootstrap.pypa.io/get-pip.py
Resolving bootstrap.pypa.io (bootstrap.pypa.io)... 151.101.128.175, 151.101.64.175, 151.101.0.175, ...
Connecting to bootstrap.pypa.io (bootstrap.pypa.io)|151.101.128.175|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1775835 (1.7M) [text/x-python]
Saving to: ‘install.py’

install.py                      100%[=====================================================>]   1.69M  --.-KB/s    in 0.1s

2020-01-18 19:52:01 (14.5 MB/s) - ‘install.py’ saved [1775835/1775835]

DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pip
  Using cached https://www.piwheels.org/simple/pip/pip-19.3.1-py2.py3-none-any.whl
Installing collected packages: pip
  Found existing installation: pip 19.3.1
    Uninstalling pip-19.3.1:
      Successfully uninstalled pip-19.3.1
Successfully installed pip-19.3.1
[ INFO ] DietPi-Software | APT installation for: python-pip python3-pip, please wait...
[  OK  ] DietPi-Software | G_AGI python-pip python3-pip
```

And here's sabnzbd (after sucessfully installing unrar):
```
Mode: Installing SABnzbd: nzb download manager

[  OK  ] DietPi-Software | Connection test: https://github.com/sabnzbd/sabnzbd/archive/master.tar.gz
[ INFO ] DietPi-Software | G_THREAD_START_0 | wget https://github.com/sabnzbd/sabnzbd/archive/master.tar.gz -O 139.tar
[ INFO ] DietPi-Software | APT installation for: par2 p7zip-full libffi-dev libssl-dev, please wait...
[  OK  ] DietPi-Software | G_AGI par2 p7zip-full libffi-dev libssl-dev
[  OK  ] DietPi-Software | G_THREAD: All threads finished
[  OK  ] DietPi-Software | tar xf 139.tar
[  OK  ] DietPi-Software | rm 139.tar
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cheetah
  Using cached https://files.pythonhosted.org/packages/cd/b0/c2d700252fc251e91c08639ff41a8a5203b627f4e0a2ae18a6b662ab32ea/Cheetah-2.4.4.tar.gz
Collecting cryptography
  Using cached https://files.pythonhosted.org/packages/be/60/da377e1bed002716fb2d5d1d1cab720f298cb33ecff7bf7adea72788e4e4/cryptography-2.8.tar.gz
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python /usr/local/lib/python2.7/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-uLP9FL/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; platform_python_implementation != '"'"'PyPy'"'"''
       cwd: None
  Complete output (5 lines):
  DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://www.piwheels.org/simple
  Collecting setuptools>=40.6.0
    Using cached https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.py3-none-any.whl
  ERROR: Package 'setuptools' requires a different Python: 2.7.16 not in '>=3.5'
  ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python /usr/local/lib/python2.7/dist-packages/pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-uLP9FL/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple -- 'setuptools>=40.6.0' wheel 'cffi>=1.8,!=1.11.3; platform_python_implementation != '"'"'PyPy'"'"'' Check the logs for full command output.
```
At the end, part of the system shows that sabnzbd is installed (shows up in services, every install starts with "[  OK  ] DietPi-Services | stop : sabnzbd" even though the service isn't running. Part of the system thinks it's not installed (for example, if I run re-install 139 it says 139 wasn't already installed).

tanderson1992, 2020-01-19:
I ran the two commands you specified above. As I mentioned, the system doesn't think sabnzbd was installed, so it informed me it couldn't re-install it. I used the standard install command after successfully running `apt install python-setuptools` but it failed with the same error.

Joulinar, 2020-01-19:
not sure if this is relevant but it seems using cached files on the installation run of @tanderson1992 

```
Collecting cheetah
  Using cached https://files.pythonhosted.org/packages/cd/b0/c2d700252fc251e91c08639ff41a8a5203b627f4e0a2ae18a6b662ab32ea/Cheetah-2.4.4.tar.gz
Collecting cryptography
  Using cached https://files.pythonhosted.org/packages/be/60/da377e1bed002716fb2d5d1d1cab720f298cb33ecff7bf7adea72788e4e4/cryptography-2.8.tar.gz
```

This was not the case on the run @MichaIng did
```
Collecting cheetah
  Downloading https://files.pythonhosted.org/packages/cd/b0/c2d700252fc251e91c08639ff41a8a5203b627f4e0a2ae18a6b662ab32ea/Cheetah-2.4.4.tar.gz (190kB)
     |████████████████████████████████| 194kB 1.3MB/s
Collecting cryptography
  Downloading https://files.pythonhosted.org/packages/c8/52/ad7f2cbe3b6e9340526dc401b38bb67c06160449446e15b0bf5f947fa168/cryptography-2.8-cp27-cp27mu-manylinux2010_x86_64.whl (2.3MB)
     |████████████████████████████████| 2.3MB 3.1MB/s
```

tanderson1992, 2020-01-19:
I think it's using cached files because I attempted the download multiple times, and those files were already available. It seems like sometimes it re-downloads files (install.py, for example) while others it uses the cache. I haven't used an apt system in ages. Would deleting the cache and trying to install again make a difference?

MichaIng, 2020-01-19:
@tanderson1992 
Jep, makes sense that it uses the cache, but probably there was indeed some broken upload temporarily.
Please try the following (as root):
```
rm -R /root/.cache/pip
pip install cheetah cryptography sabyenc
```

MichaIng, 2020-01-19:
Tested again on RPi, result is exactly the same as on VM:
- pip installer pulls setuptools 44.0.0
- SABnzbd installer skips update with: "`Requirement already satisfied: setuptools>=36`"

tanderson1992, 2020-01-20:
> rm -R /root/.cache/pip
> pip install cheetah cryptography sabyenc

That and a fresh install of sabnzbd looks like it worked fine. Thank you again for looking over this. I'm not sure why it failed on my fresh install but works well for you on RPi.

MichaIng, 2020-01-20:
As said, my guess is a temporary wrong or faulty upload/state of the pip repository that has been fixed until I ran the install tests.

MichaIng, 2020-01-20:
Marking this as closed. Feel free to reopen if required.

Joulinar, 2020-03-10:
@MichaIng 
sorry for coming back to this issue. But I found something, not exactly the same but related. Looks like a new version of Markdown was released in Feb 2020 and it seems to require Python >=3.5. Therefore SABnzbd installation seems to fail on my RPi3B+ atm. 

```
Collecting Markdown>=2.0.1
  Using cached https://www.piwheels.org/simple/markdown/Markdown-3.2.1-py2.py3-none-any.whl (88 kB)
ERROR: Package 'Markdown' requires a different Python: 2.7.16 not in '>=3.5'
```

I was able to overcome the situation by installing Markdown 3.1.1 manually once stuck on SABnzbd  software installation.

`pip install Markdown==3.1.1`

afterwards SABnzbd installation could be restarted and finished successfully

MichaIng, 2020-03-10:
@Joulinar 
Okay, first verified that SABnzbd sadly still required Python2: https://github.com/sabnzbd/sabnzbd
We install the right Python (+pip) version and call the correct `pip` command for Python2: `pip install cheetah cryptography sabyenc`
If there is a dependency (tail) of any pip == Python2 module on a Python3 version, that this is a pip repo error which will be resolved soon. I remember facing such during HA development. I'll keep your workaround in mind, to implement before v6.29 release, if this has not yet been solved pip repo-wise.

MichaIng, 2020-03-10:
Ah and similar to above, since pip caches everything, after running in such error one should always purge the cache before retrying: `rm -R /root/.cache/pip`

Joulinar, 2020-03-10:
@MichaIng 
I just grabbed one of the error messages. I was playing with SABnzbd due to an issue reported on our Forum. But I found the original error message as well now 😃 

```
Collecting Markdown>=2.0.1
  Downloading https://www.piwheels.org/simple/markdown/Markdown-3.2.1-py2.py3-none-any.whl (88 kB)
     |████████████████████████████████| 88 kB 802 kB/s
ERROR: Package 'Markdown' requires a different Python: 2.7.16 not in '>=3.5'
```

MichaIng, 2020-03-10:
@Joulinar 
Of course any Python2 module must not depend on Python3, this matches the original bug of this issue. Not sure currently where this can be reported, in case the bug is present already for too long 🤔.

MichaIng, 2020-03-10:
And btw in my case again it pulls correct Markdown 3.1.1 automatically:
```
Collecting Markdown>=2.0.1
  Downloading Markdown-3.1.1-py2.py3-none-any.whl (87 kB)
     |████████████████████████████████| 87 kB 2.0 MB/s
```
So might the bug again be solved already?

Joulinar, 2020-03-10:
@MichaIng 
I was facing different scenarios on my VM as well as on my RPi. Will do some test again by resetting both systems.

Joulinar, 2020-03-10:
@MichaIng 
Yep both installations are behaving complete different. Probably due to platform architecture.

**_Buster VM:_**
```
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Collecting cheetah
  Downloading Cheetah-2.4.4.tar.gz (190 kB)
     |████████████████████████████████| 190 kB 1.7 MB/s
Collecting cryptography
  Downloading cryptography-2.8-cp27-cp27mu-manylinux2010_x86_64.whl (2.3 MB)
     |████████████████████████████████| 2.3 MB 2.8 MB/s
Collecting sabyenc
  Downloading sabyenc-3.3.6-cp27-cp27mu-manylinux1_x86_64.whl (19 kB)
Collecting Markdown>=2.0.1
  Downloading Markdown-3.1.1-py2.py3-none-any.whl (87 kB)
     |████████████████████████████████| 87 kB 423 kB/s
Collecting enum34; python_version < "3"
  Downloading enum34-1.1.9-py2-none-any.whl (11 kB)
Collecting ipaddress; python_version < "3"
  Downloading ipaddress-1.0.23-py2.py3-none-any.whl (18 kB)
Collecting six>=1.4.1
  Downloading six-1.14.0-py2.py3-none-any.whl (10 kB)
Collecting cffi!=1.11.3,>=1.8
  Downloading cffi-1.14.0-cp27-cp27mu-manylinux1_x86_64.whl (387 kB)
     |████████████████████████████████| 387 kB 2.9 MB/s
Requirement already satisfied: setuptools>=36 in /usr/local/lib/python2.7/dist-packages (from Markdown>=2.0.1->cheetah) (44.0.0)
Collecting pycparser
  Downloading pycparser-2.20-py2.py3-none-any.whl (112 kB)
     |████████████████████████████████| 112 kB 3.2 MB/s
Building wheels for collected packages: cheetah
  Building wheel for cheetah (setup.py) ... done
  Created wheel for cheetah: filename=Cheetah-2.4.4-py2-none-any.whl size=171054 sha256=279a07e423e7f8d653ab0e6cfdf7747d68530faa7816cd01f5903aef05f97966
  Stored in directory: /root/.cache/pip/wheels/90/b0/e1/c6661f8fac2bb23914d9289c76df5d3734c94be7233beb46a5
Successfully built cheetah
Installing collected packages: Markdown, cheetah, enum34, ipaddress, six, pycparser, cffi, cryptography, sabyenc
Successfully installed Markdown-3.1.1 cffi-1.14.0 cheetah-2.4.4 cryptography-2.8 enum34-1.1.9 ipaddress-1.0.23 pycparser-2.20 sabyenc-3.3.6 six-1.14.0
[ SUB1 ] DietPi-Services > stop
```

**_Buster RPi3B+_**
```
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting cheetah
  Downloading Cheetah-2.4.4.tar.gz (190 kB)
     |████████████████████████████████| 190 kB 991 kB/s
Collecting cryptography
  Downloading cryptography-2.8.tar.gz (504 kB)
     |████████████████████████████████| 504 kB 2.7 MB/s
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting sabyenc
  Downloading sabyenc-3.3.6.tar.gz (9.8 kB)
Collecting Markdown>=2.0.1
  Downloading https://www.piwheels.org/simple/markdown/Markdown-3.2.1-py2.py3-none-any.whl (88 kB)
     |████████████████████████████████| 88 kB 800 kB/s
ERROR: Package 'Markdown' requires a different Python: 2.7.16 not in '>=3.5'
[ SUB1 ] DietPi-Services > stop
```

MichaIng, 2020-03-10:
@Joulinar 
Ah, https://www.piwheels.org/ is a repo which provides wheels for RPi in particular, but: https://www.piwheels.org/faq.html
> Currently, piwheels supports Python 3 only. While most packages available will work in any version of Python 3, some packages are only compatible with the minor version with which they were built. We build using for the ABI of the Python 3 version distributed with Raspbian releases:
>
> Raspbian Jessie - Python 3.4
Raspbian Stretch - Python 3.5
Raspbian Buster - Python 3.7

So this must not be used by Python2...

Could you try:
```
pip install -U pip
pip install Markdown
# And if it fails the same way
python -m pip install Markdown
```

MichaIng, 2020-03-10:
Bug reported to pypa: https://github.com/pypa/pip/issues/7842

@Joulinar 
Just recognised in your output on RPi:
> Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple

That https://www.piwheels.org/simple is inside the indexes while Python version is 2.X is the core of the issue, as well with `setuptools` at top of this thread. As long as this is the case, such kind of failures can happen randomly whenever piwheels has a certain pre-compiled wheel update. Hopefully it is possible to include/exclude indexes based on underlying Python version, lets see what pypa guys come up with.

MichaIng, 2020-03-10:
Not a new issue, known to pywheels.org maintainers already: https://github.com/piwheels/piwheels/issues/208
Let's hope they get those Python version flags included soon.

MichaIng, 2020-03-11:
The wheels for newest Markdown have been removed from piwheels repo, which should resolve the issue. I'll run some test on RPi later to verify. So basically there is another layer to mark/derive wheels compatibility, i.e. the `py2.py3` in file name, indicating Python2+3 compatibility (which is wrong for newest Markdown version): https://github.com/Python-Markdown/markdown/pull/919

closed 2020-05-03 by MichaIng