MichaIng/DietPi #3347: Two questions.

Dr0bac, 2020-01-18:
Hi

1. I symlinked this folder `/mnt/dietpi_userdata/radarr/MediaCover/` to sd card, so when i open radarr webview i dont wake up my ext hdd
but i noticed few times that this symlink disappeared i quess after dietpi update, but not 100% sure

2. i have zerotier installed and i tried to acess other lan devices through raspberry pi on my local network when i'm outside. in tutorial it is written that i need to add this to `/etc/network/interfaces`

```
auto eth1
iface eth1 inet manual
 
auto br0
iface br0 inet manual
    bridge_ports eth1 zt0
    bridge_fd 0
    bridge_maxage 0
```
after that i restarted, and after few hours i was unable to do what i wanted so i deleted those lines in `/etc/network/interfaces` and now i notice that restart is much slower then before.. what can be issue?

Joulinar, 2020-01-18:
Hi,
many thanks for your report. 

Regarding VPN, I would recommend to use one of the VPN Server available on DietPi Software catalog. There are already pre-configured and much more easier to use.

MichaIng, 2020-01-18:
Zerotier indeed is not a VPN in particular, in the classic server <> client style, but a peer <> peer virtual network which enables different use cases. We have a software request open for it: #2843

About the network setup, from what I know it builds up as tun interface, just the same like OpenVPN. When this is not somehow done dumb, then the service configures it a way that only requests which match the same IP range (I know the embedded interface does not use IP addresses, but everything is embedded in regular TCP/IP connections of course), are sent through the Zerotier interface while all other requests are sent and answered through the regular network adapter. Else regular internet connection would be broken as well. Hence it should be possible regardless of Zerotier to connect to your DietPi from LAN and WAN as usual. How did you install it? Using the official installer? `curl -s https://install.zerotier.com | bash`

About the Radarr symlink, at least DietPi does not touch this in any case. If `/mnt/dietpi_userdata/radarr` exists even on Radarr reinstall, it would be left untouched, only the install dir `/opt/radarr` is upgraded. So not sure why the symlink is removed, probably Radarr itself removes it in some circumstances? I know e.g. that the log file/db/dir symlink we create by default on install (to have it logging to RAMlog) are removed during some background job... here it is: #3133 so only on Jessie as of my tests.

Dr0bac, 2020-01-18:
I installed zerotier with a script from their site months ago and was using it without any problems.. it works perfectly when i want to access rpi (radarr, sonarr, transmission) from android phone, but i wanted to access other lan devices on my local network (playstation 4) and zerotier can do this in bridge mode.. but i didnt understand that well, i guess zerotier and lan must be in the same subnet for this to work, but when i do this i cant connect to local 192.168.1.12 (local adress od dietpi) anymore from my home pc when i ssh.

I followed these tutorials

https://zerotier.atlassian.net/wiki/spaces/SD/pages/7471125/Layer+2+Bridging+of+Ethernet+and+ZeroTier+Networks+on+Linux

MichaIng, 2020-01-18:
@Dr0bac 
Ah I got it, the intention with the bridge.

Did you install [bridge-utils](https://packages.debian.org/buster/bridge-utils)?
```
apt install bridge-utils
```

Also does your DietPi actually have two network adapters, so that `eth1` interface exists? Otherwise it would be the way it is done in the second example: https://gist.github.com/laduke/98406c2eb86644901c24a453736e4014#example
`dns-nameservers 127.0.0.1` would be only correct if a DNS server runs on the machine of course, like Pi-hole, dnsmasq, unbound or such, else use the router IP or some public DNS like 1.1.1.1 (Cloudflare). And yes both IP ranges must match then AFAIK. Not sure if/how this is controlled by Zerotier, e.g. if there is a config file or such.
And of course the other eth0 definition in `/etc/network/interface` must be removed. At best add everything to a drop-in config `/etc/network/interface.d/zerotier-bridge`, so dietpi-config cannot accidentally overwrite it 😉.

However IMO best solution is to install Zerotier directly on all devices which shall be available and/or have connection to the Zerotier network. This bridge method looks kinda un{intended,secure,stable} to me, but partly because I do not know much about both, Zerotier and network bridges 😅.

Dr0bac, 2020-01-18:
Yes i installed bridge-utils and added eth1 to network/interface and after that i did `ifup eth1` `ifup br0`
Zerotier is super easy to configure if two devices have zerotier installed, this bridge mode is nothing like that :D will try wireguard, i heard only good things about it.. (super fast, secure, etc)

Joulinar, 2020-01-18:
@Dr0bac 
yes WireGuard is pretty easy to configure and it's working for me on all my mobile devices (Android, Win10 & iPhone). I have the server installed on my DietPi system (RPi4B) and can connect from all over the world to my Home Network. The client can be configured that way, that only the traffic for you network will be routed through the VPN tunnel. Rest of the traffic will go to the normal internet.

MichaIng, 2020-01-18:
Agree, if it is about connecting with the home network, a classic VPN, should be best, preferably WireGuard.

Zerotier is great if you do not want to connect two "mobile" devices and do not want or cannot rely on a running server somewhere else + want to avoid the overhead of relaying data from peer1 to server and from server to peer2, but peer to peer directly instead.
___________
I am running Radarr currently, created the MediaCover symlink and went through all tasks I can do (incl. update) from the Radarr web UI. So far everything fine. Now setting up Transmission to do some test downloads:
```
2020-01-19 00:42:59 root@VM-Buster:/mnt/dietpi_userdata/radarr$ l
total 432K
drwxr-xr-x 3 radarr dietpi 4.0K Jan 19 00:27 .config
-rw-r--r-- 1 radarr dietpi  420 Jan 19 00:38 config.xml
lrwxrwxrwx 1 radarr dietpi   15 Jan 18 23:56 logs -> /var/log/radarr
lrwxrwxrwx 1 radarr dietpi   23 Jan 18 23:56 logs.db -> /var/log/radarr/logs.db
lrwxrwxrwx 1 radarr dietpi   27 Jan 18 23:56 logs.db-shm -> /var/log/radarr/logs.db-shm
lrwxrwxrwx 1 radarr dietpi   27 Jan 18 23:56 logs.db-wal -> /var/log/radarr/logs.db-wal
lrwxrwxrwx 1 root   root     27 Jan 19 00:23 MediaCover -> /home/dietpi/radarr_posters
-rw-r--r-- 1 radarr dietpi 252K Jan 19 00:38 nzbdrone.db
-rw-r--r-- 1 radarr dietpi  32K Jan 19 00:42 nzbdrone.db-shm
-rw-r--r-- 1 radarr dietpi 133K Jan 19 00:42 nzbdrone.db-wal
-rw-r--r-- 1 radarr dietpi    4 Jan 19 00:38 nzbdrone.pid
```

Dr0bac, 2020-01-19:
One more thing that i have configured is that when movie is downloaded i set it with radarr custom script to delete movie from radarr database, when movie is deleted from database, poster images are also deleted for that movie, i don't believe that this can delete root folder (mediacover) folder.
Something similar to this

https://www.reddit.com/r/radarr/comments/aicnuw/automate_removing_movies_from_radarr_once_they/

I don't know, this happened only few times in last 6 months i guess..

So we can quess that problem was with /home/dietpi/MediaCover folder, because this folder was deleted symlink was then deleted also in /mnt/dietpi_userdata/radarr so then radarr made new mediacover folder there..
In home/dietpi/ i have two script files for radarr custom script menu, and those files are still there

MichaIng, 2020-01-19:
@Dr0bac 
Down see any reason why the script would delete dir dir/symlink. It uses the Radarr API to remove the movie from database:
```
curl "http://$RADARRIP:$RADARRPORT/api/movie/$radarr_movie_id" -X DELETE -H "X-Api-Key: $RADARRAPIKEY" -k
```
Adding URL/IP, port (7878), the movie ID and personal API key, it could simply be tested manually from console.

Dr0bac, 2020-01-19:
Maybe the problem can be with DietPi-Arr_to_RAM
We update .bak database files when radarr download movie and when system goes to power off state. When i add some movie to radarr that will not be downloaded for few days, months (there is no release yet for download) and in that time dietpi shutdown via power loss or similar, i will lose that movie in radarr because .bak files were not updated. But folder with posters in mediacover directory will stay because they are not in ram.. maybe when i add next movie, that movie will have the same id name like previous movie because radarr don't know that i added last movie, because that new database don't exist anymore, and radarr will name that new folder the same like previos time ( radarr name poster folders "id1" "id2") and radarr get confused i guess.. just a crazy theory, can i test this theory without the need to unplug rpi power cable?

MichaIng, 2020-01-20:
@Dr0bac 
Probably this is related, although I am still not sure why Radarr would remove the symlink then which is still valid.

To test:
```
dietpi-arr_to_RAM 1 # Link database to RAM
# Add a new movie
systemctl stop radarr
rm -R /tmp/radarr_db_link # Remove database from RAM
dietpi-arr_to_RAM 0 # Store back to disk, this will fail
dietpi-arr_to_RAM 1 # Link to RAM again, which will now restore database from backup
systemctl start radarr
```

Dr0bac, 2020-01-23:
I will test today few times..
Can i ask, if is not to much of a hassle, can you consider changing dietpi-arr_to_RAM a little?
Can you make saving .bak files to a user predefined location? So i can store them on the sd card and update them every hour through crontab?

If you think it is not easy and quick to rewrite the script, you can leave it as it is now.

MichaIng, 2020-01-23:
@Dr0bac 
You should be able to do this by simply placing symlinks?

Would be not difficult to implement, but having too much options might get confusing and usually why would you want to move such data to the SDcard when you have dietpi_userdata already moved to an external drive to NOT have and constantly write such data to the vulnerable SDcard?

Dr0bac, 2020-01-23:
Ah, ok i will create them..
We moved database files to ram because they update itself every minute originaly because of this process in the picture

https://github.com/MichaIng/DietPi/issues/2223#issuecomment-435928563

If i update these files every hour or two on sd card that will not harm sd card to much?

MichaIng, 2020-01-23:
@Dr0bac 
The files are not that large, hence influence on SDcard should be marginal, when written only once an hour. You don't want those backup files where they are now to avoid external HDD spin-up hourly?

Dr0bac, 2020-01-23:
Yes, i like to wake up hdd only when downloading something or watching.. hdd sometimes stays for days in sleep mode..

Dr0bac, 2020-02-19:
@MichaIng I moved .bak files to sd card and created symlinks to radarr database folder, and while testing (updating with `dietpi_arr_to_RAM 2`) everything was ok, but when i restarted raspberry pi, i againg had .bak files in radarr database folder, symlinks are deleted.. Maybe something that dietpi_arr_to_RAM script does on system startup?

MichaIng, 2020-02-19:
@Dr0bac 
Ah yes that does not work currently: https://github.com/MichaIng/DietPi/blob/dev/dietpi/misc/dietpi-arr_to_RAM#L81
`mv file symlink` does not move the file to the symlink target but overwrites the symlink with the actual file.
We could use `mv file $(readlink -f symlink)`, but there are other places where having a symlinked backup does not work currently: https://github.com/MichaIng/DietPi/blob/dev/dietpi/misc/dietpi-arr_to_RAM#L183
Okay would be possible to use readlink in all cases, actually. A better long-term solution would be to  allow setting the backup dir via env var, option or config file.

Dr0bac, 2020-02-19:
Thanks @MichaIng for response, i'm not quite sure what to do from your response? did you hinted me to do something with "readlink" command, or do i need to wait for you to implement custom backup dir variable?

MichaIng, 2020-02-19:
@Dr0bac 
Basically it means that those symlinks are indeed not yet supported but I found a solution how to add it, although requires testing. If you want to test it:
- Change line: https://github.com/MichaIng/DietPi/blob/dev/dietpi/misc/dietpi-arr_to_RAM#L81
`mv $FP_DISK ${FP_DISK}.bak` => `mv $FP_DISK $(readlink -f ${FP_DISK}.bak)`
- Change line: https://github.com/MichaIng/DietPi/blob/dev/dietpi/misc/dietpi-arr_to_RAM#L85
`cp -a --no-preserve=timestamps ${FP_DISK}.bak $FP_RAM` => `cp -a --no-preserve=timestamps $(readlink -f ${FP_DISK}.bak) $FP_RAM`
- And change line: https://github.com/MichaIng/DietPi/blob/dev/dietpi/misc/dietpi-arr_to_RAM#L183
`mv ${FP_DISK}.bak $FP_DISK` => `mv $(readlink -f ${FP_DISK}.bak) $FP_DISK`

Basically all cases where `${FP_DISK}.bak` is handled, it needs to be wrapped into `$(readlink -f ... )`, so it it is a symlink, thinks will be done with the symlink target. If the backup file does not exist, readlink btw just outputs the input path/name, hence a new backup on default location would be created.

Dr0bac, 2020-02-27:
@MichaIng i edited arr_to_RAM script, and now everything work like charm. I tried:

1. To update backups with dietpi-arr_to_RAM 2
2. To reboot system
3. To stop radarr and delete nzb files from ram and let it restore nzb files from backup

Backup symlinks survive everytime.

Also i Mediafolder symlink also survive little test you told me few posts above. So something else must be the problem.

I will close the issue, and will come back if i notice something new about that issue..

Can you update the arr_to_RAM script with new changes you mentioned few comments back?

closed 2020-02-27 by Dr0bac

MichaIng, 2020-02-28:
@Dr0bac 
Okay I'll implement that or indeed a custom backup location. I'm just not keen to add another config file for that, would like to keep the script simple and failsafe.