MichaIng/DietPi #3348: GMrender service fails to start normally

kikigarel, 2020-01-19:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /DietPi/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC device | `echo $G_HW_MODEL_DESCRIPTION` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `sed -n 5p /DietPi/dietpi/.hw_model`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2020-01-20:
@kikigarel 
Many thanks for your report.

> ACTIVE_IP=$(mawk NR==4 /DietPi/dietpi/.network) (code=exited, status=1/FAILURE)
> ...
> dash[885]: Failed to set environment: Invalid environment assignments

This can only mean that the IP line is still empty when the service is started. Strange, since this is started long after dietpi-boot.service where network interfaces should be fully initialised.
- Did you set `CONFIG_BOOT_WAIT_FOR_NETWORK=0` in `dietpi.txt`?
- Do you use WiFi or Ethernet?

> I can see Dietpi as an available endpoint in Audirvana but no music will play!

Does the RPi otherwise play music fine, outside of GMediaRender?

kikigarel, 2020-01-20:
@MichaIng 

Thanks for your consideration. Well, that's really strange because the RPI has a static IP address on my network.
 
In dietpi.txt, _CONFIG_BOOT_WAIT_FOR_NETWORK=1_
In Dietpi-config, the value of the "Boot Net Wait" parameters is set to: _[10 Seconds MAX (default)]_. 

I didn't modify anything. I use ethernet. An IP address issue would make sense since the music won't play on the target playback device. Yes, I have used the RPi with RoPieeeXL and it played music without any problem.

MichaIng, 2020-01-20:
@kikigarel 
If you find time, could you run `systemd-analyze plot > systemd-analyze.svg` and paste a screen of this systemd-analyze.svg file opened in any browser or capable graphics viewer here?
Additionally: `journalctl -u ifup@eth0; journalctl -u dietpi-boot`

The following should resolve the autostart issue:
- Open `dietpi-services` and select `gmrender`
- Select `Edit`
- nano opens with commented service file content
- Below the commented `ExecStartPre=` line, add the following (uncommented):
```
ExecStartPre=
ExecStartPre=/DietPi/dietpi/func/obtain_network_details
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=$(mawk 'NR==4' /DietPi/dietpi/.network)'
```
- This updates the network info file prior to scraping the active IP.

kikigarel, 2020-01-20:
Here is the content of the SVG file: 

<?xml version="1.0" standalone="no"?>
<!DOCTYPE svg PUBLIC "-//W3C//DTD SVG 1.1//EN" "http://www.w3.org/Graphics/SVG/1.1/DTD/svg11.dtd">
<svg width="3841px" height="3250px" version="1.1" xmlns="http://www.w3.org/2000/svg">

<defs>
  <style type="text/css">
    <![CDATA[
      rect       { stroke-width: 1; stroke-opacity: 0; }
      rect.background   { fill: rgb(255,255,255); }
      rect.activating   { fill: rgb(255,0,0); fill-opacity: 0.7; }
      rect.active       { fill: rgb(200,150,150); fill-opacity: 0.7; }
      rect.deactivating { fill: rgb(150,100,100); fill-opacity: 0.7; }
      rect.kernel       { fill: rgb(150,150,150); fill-opacity: 0.7; }
      rect.initrd       { fill: rgb(150,150,150); fill-opacity: 0.7; }
      rect.firmware     { fill: rgb(150,150,150); fill-opacity: 0.7; }
      rect.loader       { fill: rgb(150,150,150); fill-opacity: 0.7; }
      rect.userspace    { fill: rgb(150,150,150); fill-opacity: 0.7; }
      rect.security     { fill: rgb(144,238,144); fill-opacity: 0.7; }
      rect.generators   { fill: rgb(102,204,255); fill-opacity: 0.7; }
      rect.unitsload    { fill: rgb( 82,184,255); fill-opacity: 0.7; }
      rect.box   { fill: rgb(240,240,240); stroke: rgb(192,192,192); }
      line       { stroke: rgb(64,64,64); stroke-width: 1; }
//    line.sec1  { }
      line.sec5  { stroke-width: 2; }
      line.sec01 { stroke: rgb(224,224,224); stroke-width: 1; }
      text       { font-family: Verdana, Helvetica; font-size: 14px; }
      text.left  { font-family: Verdana, Helvetica; font-size: 14px; text-anchor: start; }
      text.right { font-family: Verdana, Helvetica; font-size: 14px; text-anchor: end; }
      text.sec   { font-size: 10px; }
    ]]>
   </style>
</defs>

<rect class="background" width="100%" height="100%" />
<text x="20" y="50">Startup finished in 1.181s (kernel) + 36.429s (userspace) = 37.610s
graphical.target reached after 36.390s in userspace</text><text x="20" y="30">Raspbian GNU/Linux 10 (buster) DietPi (Linux 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019) arm </text><g transform="translate(20.000,100)">
<rect class="box" x="0" y="0" width="3761.055" height="3000.000" />
  <line class="sec5" x1="0.000" y1="0" x2="0.000" y2="3000.000" />
  <text class="sec" x="0.000" y="-5.000" >0.0s</text>
  <line class="sec01" x1="10.000" y1="0" x2="10.000" y2="3000.000" />
  <line class="sec01" x1="20.000" y1="0" x2="20.000" y2="3000.000" />
  <line class="sec01" x1="30.000" y1="0" x2="30.000" y2="3000.000" />
  <line class="sec01" x1="40.000" y1="0" x2="40.000" y2="3000.000" />
  <line class="sec01" x1="50.000" y1="0" x2="50.000" y2="3000.000" />
  <line class="sec01" x1="60.000" y1="0" x2="60.000" y2="3000.000" />
  <line class="sec01" x1="70.000" y1="0" x2="70.000" y2="3000.000" />
  <line class="sec01" x1="80.000" y1="0" x2="80.000" y2="3000.000" />
  <line class="sec01" x1="90.000" y1="0" x2="90.000" y2="3000.000" />
  <line class="sec1" x1="100.000" y1="0" x2="100.000" y2="3000.000" />
  <text class="sec" x="100.000" y="-5.000" >1.0s</text>
  <line class="sec01" x1="110.000" y1="0" x2="110.000" y2="3000.000" />
  <line class="sec01" x1="120.000" y1="0" x2="120.000" y2="3000.000" />
  <line class="sec01" x1="130.000" y1="0" x2="130.000" y2="3000.000" />
  <line class="sec01" x1="140.000" y1="0" x2="140.000" y2="3000.000" />
  <line class="sec01" x1="150.000" y1="0" x2="150.000" y2="3000.000" />
  <line class="sec01" x1="160.000" y1="0" x2="160.000" y2="3000.000" />
  <line class="sec01" x1="170.000" y1="0" x2="170.000" y2="3000.000" />
  <line class="sec01" x1="180.000" y1="0" x2="180.000" y2="3000.000" />
  <line class="sec01" x1="190.000" y1="0" x2="190.000" y2="3000.000" />
  <line class="sec1" x1="200.000" y1="0" x2="200.000" y2="3000.000" />
  <text class="sec" x="200.000" y="-5.000" >2.0s</text>
  <line class="sec01" x1="210.000" y1="0" x2="210.000" y2="3000.000" />
  <line class="sec01" x1="220.000" y1="0" x2="220.000" y2="3000.000" />
  <line class="sec01" x1="230.000" y1="0" x2="230.000" y2="3000.000" />
  <line class="sec01" x1="240.000" y1="0" x2="240.000" y2="3000.000" />
  <line class="sec01" x1="250.000" y1="0" x2="250.000" y2="3000.000" />
  <line class="sec01" x1="260.000" y1="0" x2="260.000" y2="3000.000" />
  <line class="sec01" x1="270.000" y1="0" x2="270.000" y2="3000.000" />
[... 975 lines cut ...]
  <rect class="activating" x="3757.156" y="2960.000" width="0.000" height="19.000" />
  <rect class="active" x="3757.156" y="2960.000" width="3.900" height="19.000" />
  <rect class="deactivating" x="3761.055" y="2960.000" width="0.000" height="19.000" />
  <text class="right" x="3752.156" y="2974.000">graphical.target</text>
  <rect class="activating" x="3757.499" y="2980.000" width="3.394" height="19.000" />
  <rect class="active" x="3760.893" y="2980.000" width="0.000" height="19.000" />
  <rect class="deactivating" x="3760.893" y="2980.000" width="0.000" height="19.000" />
  <text class="right" x="3752.499" y="2994.000">systemd-update-utmp-runlevel.service (33ms)</text>
</g>
<g transform="translate(20,100)">
  <rect class="activating" x="0.000" y="3020.000" width="30.000" height="19.000" />
  <text class="left" x="45.000" y="3034.000">Activating</text>
  <rect class="active" x="0.000" y="3040.000" width="30.000" height="19.000" />
  <text class="left" x="45.000" y="3054.000">Active</text>
  <rect class="deactivating" x="0.000" y="3060.000" width="30.000" height="19.000" />
[... cut ...] .000">Deactivating</text>
  <rect class="security" x="0.000" y="3080.000" width="30.000" height="19.000" />
  <text class="left" x="45.000" y="3094.000">Setting up security module</text>
  <rect class="generators" x="0.000" y="3100.000" width="30.000" height="19.000" />
  <text class="left" x="45.000" y="3114.000">Generators</text>
  <rect class="unitsload" x="0.000" y="3120.000" width="30.000" height="19.000" />
  <text class="left" x="45.000" y="3134.000">Loading unit files</text>
</g>

</svg>

MichaIng, 2020-01-20:
Okay here the relevant part, visualised in browser:
[image]
So `ifup@eth0` starts before `dietpi-boot` does. dietpi-boot looks like it takes VERY long, but this is only due to network time sync, raising system time about the seconds lost during reboot phase. As can be seen shortly after `dietpi-boot`, `time-sync.target` is reached, immediately before the large gap.

The journal entries show that `dhclient` assigns an IP ~7 seconds after dietpi-boot has started. Thinkable that this is too late. It is the type of those `ifup@` services (Type=simple) that allows any follow-up services to start even that `ifup@eth0` did not do any step yet, rather than finished its tasks.

@kikigarel 
Please try to apply a static IP via `dietpi-config` > `Network Options: Adapters`, copying the values assigned from DHCP before.

`gmrender.service` fails now manually? Must be a syntax issue. Please paste:
```
systemctl cat gmrender
systemctl status gmrender
```

kikigarel, 2020-01-20:
Here is the requested info:

**systemctl cat gmrender**

# /etc/systemd/system/gmrender.service
[Unit]
Description=GMediaRender (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
User=gmrender
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=$(mawk 'NR==4' /
DietPi/dietpi/.network)'
ExecStart=/usr/local/bin/gmediarender -u 'b56c9bf7-dc42-42f1-811f-d1def6a26e5a'
-f 'DietPi' --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout
-initial-volume-db=-1 --logfile=stdout -I "$ACTIVE_IP"

[Install]
WantedBy=multi-user.target

# /etc/systemd/system/gmrender.service.d/dietpi-process_tool.conf
# WARNING: Do not manually edit this file, use "dietpi-services" to adjust value
s!
[Service]
CPUSchedulingPolicy=fifo
IOSchedulingClass=realtime
CPUSchedulingPriority=50

# /etc/systemd/system/gmrender.service.d/dietpi-services_edit.conf

[Unit]
#Description=GMediaRender (DietPi)
#Wants=network-online.target
#After=network-online.target dietpi-boot.service

[Service]
#User=gmrender
#ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=$(mawk 'NR==4' /DietPi/dietpi/.network)'
ExecStartPre=
ExecStartPre=DietPi/dietpi/func/obtain_network_details
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=$(mawk 'NR==4' /DietPi/dietpi/.network)'
#ExecStart=/usr/local/bin/gmediarender -u 'b56c9bf7-dc42-42f1-811f-d1def6a26e5a' -f 'DietPi' --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-ini
tial-volume-db=-1 --logfile=stdout -I "$ACTIVE_IP"

[Install]
#WantedBy=multi-user.target

**systemctl status gmrender**

root@DietPi:~# systemctl status gmrender
● gmrender.service - GMediaRender (DietPi)
   Loaded: loaded (/etc/systemd/system/gmrender.service; disabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/gmrender.service.d
           └─dietpi-process_tool.conf, dietpi-services_edit.conf
   Active: failed (Result: exit-code) since Mon 2020-01-20 23:41:48 CET; 4min 5s ago
  Process: 1369 ExecStart=/usr/local/bin/gmediarender -u b56c9bf7-dc42-42f1-811f-d1def6a26e5a -f DietPi --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault -
-gstout-initial-volume-db=-1 --logfile=stdout -I $ACTIVE_IP (code=exited, status=1/FAILURE)
 Main PID: 1369 (code=exited, status=1/FAILURE)

janv. 20 23:41:48 DietPi systemd[1]: Started GMediaRender (DietPi).
janv. 20 23:41:48 DietPi gmediarender[1369]: Failed to initialize: Missing argument for -I
janv. 20 23:41:48 DietPi systemd[1]: gmrender.service: Main process exited, code=exited, status=1/FAILURE
janv. 20 23:41:48 DietPi systemd[1]: gmrender.service: Failed with result 'exit-code'.
root@DietPi:~#

MichaIng, 2020-01-20:
@kikigarel 
There is a missing slash in your service file edit:
```
ExecStartPre=DietPi/dietpi/func/obtain_network_details
```
must be
```
ExecStartPre=/DietPi/dietpi/func/obtain_network_details
```

kikigarel, 2020-01-21:
@MichaIng

I have corrected the missing slash and now the service starts normally. However, I still don't get music to play :-( Audirvana detects DietPi as endpoint/renderer but when I click "Play" nothing happens and the icon is grayed. Really strange :-/

MichaIng, 2020-01-21:
@kikigarel 
Hmm, does the log show anything (note, the missing png files are expected and not related): `journalctl -u gmrender`

Could you try to start the binary manually as root (or via sudo)?
```
systemctl stop gmrender
/usr/local/bin/gmediarender -u b56c9bf7-dc42-42f1-811f-d1def6a26e5a -f DietPi --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=stdout -I 192.168.1.26
```
If still no music, try without any gstout settings:
```
/usr/local/bin/gmediarender -u b56c9bf7-dc42-42f1-811f-d1def6a26e5a -f DietPi --logfile=stdout -I 192.168.1.26
```

kikigarel, 2020-01-21:
@MichaIng
Here is the journal. I'm not a developer but there seems to be a few issues with devices not found.

root@DietPi:~# dietpi-services
root@DietPi:~# journalctl -u gmrender
-- Logs begin at Wed 2020-01-22 00:14:27 CET, end at Wed 2020-01-22 00:36:58 CET
. --
janv. 22 00:14:47 DietPi systemd[1]: Starting GMediaRender (DietPi)...
janv. 22 00:14:47 DietPi systemd[1]: Started GMediaRender (DietPi).
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.715136 |
main] gmediarender 0.0.8 log started [ gmediarender 0.0.8 (libupnp-1.8.4; glib-2
.58.3; gstreamer-1.14.4) ]
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.715459 |
output] Using output module: gst (GStreamer multimedia framework)
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.720323 |
connmgr] Registering support for 'application/x-apetag'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.720366 |
connmgr] Registering support for 'audio/*'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.721620 |
connmgr] Registering support for 'video/x-raw'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.723169 |
connmgr] Registering support for 'application/x-yuv4mpeg'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.726189 |
connmgr] Registering support for 'audio/x-raw'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.773494 |
connmgr] Registering support for 'application/x-srtp'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.773532 |
connmgr] Registering support for 'application/x-srtcp'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.773703 | connmgr] Registering support for 'application/x-rtp'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.773718 | connmgr] Registering support for 'application/x-rtcp'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779501 | connmgr] Registering support for 'video/x-vp8'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779537 | connmgr] Registering support for 'video/x-vp9'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779549 | connmgr] Registering support for 'video/x-av1'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779561 | connmgr] Registering support for 'audio/x-vorbis'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779572 | connmgr] Registering support for 'audio/x-opus'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779583 | connmgr] Registering support for 'subtitle/x-kate'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779594 | connmgr] Registering support for 'text/x-raw'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779605 | connmgr] Registering support for 'application/x-ssa'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779615 | connmgr] Registering support for 'application/x-ass'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779624 | connmgr] Registering support for 'application/x-usf'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779634 | connmgr] Registering support for 'subpicture/x-dvd'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779644 | connmgr] Registering support for 'application/x-subtitle-unknown'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779707 | connmgr] Registering support for 'video/mpeg'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779720 | connmgr] Registering support for 'video/x-h264'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779730 | connmgr] Registering support for 'video/x-h265'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779741 | connmgr] Registering support for 'video/x-divx'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779751 | connmgr] Registering support for 'video/x-huffyuv'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779761 | connmgr] Registering support for 'video/x-dv'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779771 | connmgr] Registering support for 'video/x-h263'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779781 | connmgr] Registering support for 'video/x-msmpeg'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779790 | connmgr] Registering support for 'image/jpeg'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779800 | connmgr] Registering support for 'video/x-theora'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779811 | connmgr] Registering support for 'video/x-dirac'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779821 | connmgr] Registering support for 'video/x-pn-realvideo'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779833 | connmgr] Registering support for 'video/x-prores'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779843 | connmgr] Registering support for 'video/x-wmv'
janv. 22 00:14:50 DietPi gmediarender[745]: INFO  [2020-01-22 00:14:50.779862 | connmgr] Registering support for 'audio/mpeg'
[... cut ...]
janv. 22 00:36:58 DietPi gmediarender[745]: INFO  [2020-01-22 00:36:58.234596 | webserver] 404 Not found. (attempt to access non-existent '/upnp/grender-64x64.png')
janv. 22 00:36:58 DietPi gmediarender[745]: INFO  [2020-01-22 00:36:58.240736 | webserver] 404 Not found. (attempt to access non-existent '/upnp/grender-128x128.png'
)
janv. 22 00:36:58 DietPi gmediarender[745]: INFO  [2020-01-22 00:36:58.267749 | webserver] Access /upnp/rendercontrolSCPD.xml (text/xml) len=13317
janv. 22 00:36:58 DietPi gmediarender[745]: INFO  [2020-01-22 00:36:58.282392 | webserver] Access /upnp/rendertransportSCPD.xml (text/xml) len=15697

kikigarel, 2020-01-21:
@MichaIng

There is one strange thing: each time I select dietpi as renderer, the volume is set to -1.2db. Not sure if it is related to the issue (it should be 0db)?

kikigarel, 2020-01-22:
Here is the result of the first command:

root@DietPi:~# systemctl stop gmrender
root@DietPi:~# /usr/local/bin/gmediarender -u b56c9bf7-dc42-42f1-811f-dldef6a26e5a -f DietPi --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstout-initial-volume-db=-1 --logfile=stdout -I 192.168.1.26
INFO  [2020-01-22 01:00:24.362580 | main] gmediarender 0.0.8 log started [ gmediarender 0.0.8 (libupnp-1.8.4; glib-2.58.3; gstreamer-1.14.4) ]
INFO  [2020-01-22 01:00:24.362875 | output] Using output module: gst (GStreamer multimedia framework)
INFO  [2020-01-22 01:00:24.367593 | connmgr] Registering support for 'application/x-apetag'
INFO  [2020-01-22 01:00:24.367638 | connmgr] Registering support for 'audio/*'
INFO  [2020-01-22 01:00:24.368908 | connmgr] Registering support for 'video/x-raw'
INFO  [2020-01-22 01:00:24.370471 | connmgr] Registering support for 'application/x-yuv4mpeg'
INFO  [2020-01-22 01:00:24.373510 | connmgr] Registering support for 'audio/x-raw'
INFO  [2020-01-22 01:00:24.416479 | connmgr] Registering support for 'application/x-srtp'
INFO  [2020-01-22 01:00:24.416522 | connmgr] Registering support for 'application/x-srtcp'
INFO  [2020-01-22 01:00:24.416737 | connmgr] Registering support for 'application/x-rtp'
INFO  [2020-01-22 01:00:24.416791 | connmgr] Registering support for 'application/x-rtcp'
INFO  [2020-01-22 01:00:24.422500 | connmgr] Registering support for 'video/x-vp8'
INFO  [2020-01-22 01:00:24.422587 | connmgr] Registering support for 'video/x-vp9'
INFO  [2020-01-22 01:00:24.422631 | connmgr] Registering support for 'video/x-av1'
INFO  [2020-01-22 01:00:24.422674 | connmgr] Registering support for 'audio/x-vorbis'
INFO  [2020-01-22 01:00:24.422715 | connmgr] Registering support for 'audio/x-opus'
INFO  [2020-01-22 01:00:24.422759 | connmgr] Registering support for 'subtitle/x-kate'
INFO  [2020-01-22 01:00:24.422800 | connmgr] Registering support for 'text/x-raw'
INFO  [2020-01-22 01:00:24.422841 | connmgr] Registering support for 'application/x-ssa'
INFO  [2020-01-22 01:00:24.422882 | connmgr] Registering support for 'application/x-ass'
INFO  [2020-01-22 01:00:24.422924 | connmgr] Registering support for 'application/x-usf'
INFO  [2020-01-22 01:00:24.422966 | connmgr] Registering support for 'subpicture/x-dvd'
INFO  [2020-01-22 01:00:24.423007 | connmgr] Registering support for 'application/x-subtitle-unknown'
INFO  [2020-01-22 01:00:24.423109 | connmgr] Registering support for 'video/mpeg'
INFO  [2020-01-22 01:00:24.423158 | connmgr] Registering support for 'video/x-h264'
INFO  [2020-01-22 01:00:24.423200 | connmgr] Registering support for 'video/x-h265'
INFO  [2020-01-22 01:00:24.423240 | connmgr] Registering support for 'video/x-divx'
INFO  [2020-01-22 01:00:24.423281 | connmgr] Registering support for 'video/x-huffyuv'
INFO  [2020-01-22 01:00:24.423322 | connmgr] Registering support for 'video/x-dv'
INFO  [2020-01-22 01:00:24.423363 | connmgr] Registering support for 'video/x-h263'
INFO  [2020-01-22 01:00:24.423404 | connmgr] Registering support for 'video/x-msmpeg'
INFO  [2020-01-22 01:00:24.423445 | connmgr] Registering support for 'image/jpeg'
INFO  [2020-01-22 01:00:24.423487 | connmgr] Registering support for 'video/x-theora'
INFO  [2020-01-22 01:00:24.423528 | connmgr] Registering support for 'video/x-dirac'
INFO  [2020-01-22 01:00:24.423569 | connmgr] Registering support for 'video/x-pn-realvideo'
INFO  [2020-01-22 01:00:24.423614 | connmgr] Registering support for 'video/x-prores'
INFO  [2020-01-22 01:00:24.423655 | connmgr] Registering support for 'video/x-wmv'
INFO  [2020-01-22 01:00:24.423698 | connmgr] Registering support for 'audio/mpeg'
INFO  [2020-01-22 01:00:24.423740 | connmgr] Registering support for 'audio/x-mpeg'
INFO  [2020-01-22 01:00:24.423781 | connmgr] Registering support for 'audio/x-scpls'
INFO  [2020-01-22 01:00:24.423821 | connmgr] Registering support for 'audio/L16;rate=44100;channels=2'
INFO  [2020-01-22 01:00:24.423865 | connmgr] Registering support for 'audio/x-ac3'
INFO  [2020-01-22 01:00:24.423906 | connmgr] Registering support for 'audio/x-eac3'
INFO  [2020-01-22 01:00:24.423948 | connmgr] Registering support for 'audio/x-dts'
INFO  [2020-01-22 01:00:24.423991 | connmgr] Registering support for 'audio/x-flac'
INFO  [2020-01-22 01:00:24.424034 | connmgr] Registering support for 'audio/x-speex'
INFO  [2020-01-22 01:00:24.424077 | connmgr] Registering support for 'audio/x-tta'
INFO  [2020-01-22 01:00:24.424118 | connmgr] Registering support for 'audio/x-pn-realaudio'
INFO  [2020-01-22 01:00:24.424162 | connmgr] Registering support for 'audio/x-wma'
INFO  [2020-01-22 01:00:24.424204 | connmgr] Registering support for 'audio/x-alaw'
INFO  [2020-01-22 01:00:24.424245 | connmgr] Registering support for 'audio/x-mulaw'
INFO  [2020-01-22 01:00:24.424287 | connmgr] Registering support for 'audio/x-adpcm'
INFO  [2020-01-22 01:00:24.424329 | connmgr] Registering support for 'audio/G722'
INFO  [2020-01-22 01:00:24.424584 | connmgr] Registering support for 'audio/x-matroska'
INFO  [2020-01-22 01:00:24.424638 | connmgr] Registering support for 'video/x-matroska'
INFO  [2020-01-22 01:00:24.424680 | connmgr] Registering support for 'video/x-matroska-3d'
INFO  [2020-01-22 01:00:24.424722 | connmgr] Registering support for 'audio/webm'
INFO  [2020-01-22 01:00:24.424764 | connmgr] Registering support for 'video/webm'
INFO  [2020-01-22 01:00:24.426499 | connmgr] Registering support for 'audio/x-aiff'
[... cut ...]
<Loudness val="0" channel="Master"></Loudness>
<RedVideoGain val="0"></RedVideoGain>
<ColorTemperature val="0"></ColorTemperature>
<Sharpness val="0"></Sharpness>
<RedVideoBlackLevel val="0"></RedVideoBlackLevel>
<BlueVideoGain val="0"></BlueVideoGain>
<Mute val="0" channel="Master"></Mute>
<HorizontalKeystone val="0"></HorizontalKeystone>
<VolumeDB val="-307" channel="Master"></VolumeDB>
<PresetNameList val=""></PresetNameList>
<Contrast val="0"></Contrast>
<Brightness val="0"></Brightness>
</InstanceID>
</Event>

kikigarel, 2020-01-22:
Second command: 

root@DietPi:~# dietpi-services
[  OK  ] DietPi-Services | restart : gmrender
root@DietPi:~# /usr/local/bin/gmediarender -u b56c9bf7-dc42-42f1-811f-d1def6a26e5a -f Dietpi --logfile=stdout -I 192.168.1.26
INFO  [2020-01-22 01:08:26.361183 | main] gmediarender 0.0.8 log started [ gmediarender 0.0.8 (libupnp-1.8.4; glib-2.58.3; gstreamer-1.14.4) ]
INFO  [2020-01-22 01:08:26.361343 | output] Using output module: gst (GStreamer multimedia framework)
INFO  [2020-01-22 01:08:26.365929 | connmgr] Registering support for 'application/x-apetag'
INFO  [2020-01-22 01:08:26.366024 | connmgr] Registering support for 'audio/*'
INFO  [2020-01-22 01:08:26.367349 | connmgr] Registering support for 'video/x-raw'
INFO  [2020-01-22 01:08:26.368930 | connmgr] Registering support for 'application/x-yuv4mpeg'
INFO  [2020-01-22 01:08:26.371965 | connmgr] Registering support for 'audio/x-raw'
INFO  [2020-01-22 01:08:26.415227 | connmgr] Registering support for 'application/x-rtp'
INFO  [2020-01-22 01:08:26.415318 | connmgr] Registering support for 'application/x-rtcp'
INFO  [2020-01-22 01:08:26.415635 | connmgr] Registering support for 'application/x-srtp'
INFO  [2020-01-22 01:08:26.415688 | connmgr] Registering support for 'application/x-srtcp'
INFO  [2020-01-22 01:08:26.420960 | connmgr] Registering support for 'audio/x-matroska'
INFO  [2020-01-22 01:08:26.421048 | connmgr] Registering support for 'video/x-matroska'
INFO  [2020-01-22 01:08:26.421092 | connmgr] Registering support for 'video/x-matroska-3d'
INFO  [2020-01-22 01:08:26.421134 | connmgr] Registering support for 'audio/webm'
INFO  [2020-01-22 01:08:26.421176 | connmgr] Registering support for 'video/webm'
INFO  [2020-01-22 01:08:26.421960 | connmgr] Registering support for 'video/mpeg'
INFO  [2020-01-22 01:08:26.422025 | connmgr] Registering support for 'video/x-h264'
INFO  [2020-01-22 01:08:26.422069 | connmgr] Registering support for 'video/x-h265'
INFO  [2020-01-22 01:08:26.422111 | connmgr] Registering support for 'video/x-divx'
INFO  [2020-01-22 01:08:26.422153 | connmgr] Registering support for 'video/x-huffyuv'
INFO  [2020-01-22 01:08:26.422194 | connmgr] Registering support for 'video/x-dv'
INFO  [2020-01-22 01:08:26.422235 | connmgr] Registering support for 'video/x-h263'
INFO  [2020-01-22 01:08:26.422276 | connmgr] Registering support for 'video/x-msmpeg'
INFO  [2020-01-22 01:08:26.422316 | connmgr] Registering support for 'image/jpeg'
INFO  [2020-01-22 01:08:26.422357 | connmgr] Registering support for 'video/x-theora'
INFO  [2020-01-22 01:08:26.422398 | connmgr] Registering support for 'video/x-dirac'
INFO  [2020-01-22 01:08:26.422438 | connmgr] Registering support for 'video/x-pn-realvideo'
INFO  [2020-01-22 01:08:26.422480 | connmgr] Registering support for 'video/x-vp8'
INFO  [2020-01-22 01:08:26.422520 | connmgr] Registering support for 'video/x-vp9'
INFO  [2020-01-22 01:08:26.422562 | connmgr] Registering support for 'video/x-prores'
INFO  [2020-01-22 01:08:26.422603 | connmgr] Registering support for 'video/x-wmv'
INFO  [2020-01-22 01:08:26.422645 | connmgr] Registering support for 'audio/mpeg'
INFO  [2020-01-22 01:08:26.422685 | connmgr] Registering support for 'audio/x-mpeg'
INFO  [2020-01-22 01:08:26.422725 | connmgr] Registering support for 'audio/x-scpls'
INFO  [2020-01-22 01:08:26.422765 | connmgr] Registering support for 'audio/L16;rate=44100;channels=2'
INFO  [2020-01-22 01:08:26.422807 | connmgr] Registering support for 'audio/x-ac3'
INFO  [2020-01-22 01:08:26.422848 | connmgr] Registering support for 'audio/x-eac3'
INFO  [2020-01-22 01:08:26.422889 | connmgr] Registering support for 'audio/x-dts'
INFO  [2020-01-22 01:08:26.422930 | connmgr] Registering support for 'audio/x-vorbis'
INFO  [2020-01-22 01:08:26.422971 | connmgr] Registering support for 'audio/x-flac'
INFO  [2020-01-22 01:08:26.423012 | connmgr] Registering support for 'audio/x-opus'
INFO  [2020-01-22 01:08:26.423052 | connmgr] Registering support for 'audio/x-speex'
INFO  [2020-01-22 01:08:26.423095 | connmgr] Registering support for 'audio/x-tta'
INFO  [2020-01-22 01:08:26.423136 | connmgr] Registering support for 'audio/x-pn-realaudio'
INFO  [2020-01-22 01:08:26.423176 | connmgr] Registering support for 'audio/x-wma'
INFO  [2020-01-22 01:08:26.423217 | connmgr] Registering support for 'audio/x-alaw'
INFO  [2020-01-22 01:08:26.423259 | connmgr] Registering support for 'audio/x-mulaw'
INFO  [2020-01-22 01:08:26.423300 | connmgr] Registering support for 'audio/x-adpcm'
INFO  [2020-01-22 01:08:26.423342 | connmgr] Registering support for 'audio/G722'
INFO  [2020-01-22 01:08:26.423384 | connmgr] Registering support for 'subtitle/x-kate'
INFO  [2020-01-22 01:08:26.423426 | connmgr] Registering support for 'text/x-raw'
INFO  [2020-01-22 01:08:26.423467 | connmgr] Registering support for 'application/x-ssa'
INFO  [2020-01-22 01:08:26.423509 | connmgr] Registering support for 'application/x-ass'
INFO  [2020-01-22 01:08:26.423550 | connmgr] Registering support for 'application/x-usf'
INFO  [2020-01-22 01:08:26.423591 | connmgr] Registering support for 'subpicture/x-dvd'
INFO  [2020-01-22 01:08:26.423632 | connmgr] Registering support for 'application/x-subtitle-unknown'
INFO  [2020-01-22 01:08:26.423882 | connmgr] Registering support for 'video/x-av1'
INFO  [2020-01-22 01:08:26.425249 | connmgr] Registering support for 'audio/x-aiff'
INFO  [2020-01-22 01:08:26.425331 | connmgr] Registering support for 'audio/aiff'
[... cut ...]
ERROR [2020-01-22 01:09:29.716849 | upnp] UpnpInit(ip=192.168.1.26, port=49494) Error: UPNP_E_SOCKET_BIND (-203). Retrying... (2s)
ERROR [2020-01-22 01:09:30.719388 | upnp] UpnpInit(ip=192.168.1.26, port=49494) Error: UPNP_E_SOCKET_BIND (-203). Retrying... (1s)
ERROR [2020-01-22 01:09:31.721622 | upnp] UpnpInit(ip=192.168.1.26, port=49494) Error: UPNP_E_SOCKET_BIND (-203). Retrying... (0s)
ERROR [2020-01-22 01:09:31.723737 | upnp] UpnpInit(ip=192.168.1.26, port=49494) Error: UPNP_E_SOCKET_BIND (-203). Giving up.
ERROR [2020-01-22 01:09:31.723822 | main] ERROR: Failed to initialize UPnP device
root@DietPi:~#

MichaIng, 2020-01-22:
@kikigarel 
Note that you must stop the service before you can manually execute the binary (running gmrender in foreground then), else the UPnP device is blocked by the service. Executing it manually is to find out if there is a permission issue.
After stopping the service and running the first command, are you able to stream music then?

kikigarel, 2020-01-22:
@MichaIng

No, after running the first command Dietpi is no longer detected (I get a "Peripheral not connected" message in Audirvana)

kikigarel, 2020-01-22:
Same thing with the second command (without restarting the service)

MichaIng, 2020-01-22:
@kikigarel 
Note that you need to keep the command running (no ctrl+c or such) while testing. Since the command is just a copy&paste from the service file is must work just the same as when running the service. And please assure that the service is stopped:
```
systemctl stop gmrender
killall -w gmediarender
```

Btw is it onboard sound? Did you configure it via dietpi-config?
Please also paste:
```
cat /etc/asound.conf
aplay -l
```

kikigarel, 2020-01-22:
> @kikigarel
> Note that you need to keep the command running (no ctrl+c or such) while testing. Since the command is just a copy&paste from the service file is must work just the same as when running [...]

Yes it was stopped

> 
> Btw is it onboard sound? Did you configure it via dietpi-config?

I'm not sure I understand what you mean. I'm trying the play a file from Audirvana to the DAC via the RPi/Dietpi. Yes, the DAC is detected and configured as sound card in Dietpi-Config.

> Please also paste:
> 
> ```
> cat /etc/asound.conf
> aplay -l
> ```

There you go:

root@DietPi:~# cat /etc/asound.conf
pcm.!default {
        type hw
        card 1
        device 0
}

ctl.!default {
        type hw
        card 1
}
root@DietPi:~#

root@DietPi:~# aplay -l
**** Liste des Périphériques Matériels PLAYBACK ****
carte 1: UD503 [UD-503], périphérique 0: USB Audio [USB Audio]
  Sous-périphériques: 1/1
  Sous-périphérique #0: subdevice #0
root@DietPi:~#

kikigarel, 2020-01-22:
Hi @MichaIng, 

I confirm the service was stopped while running the command. And I couldn't play music while the command was running, or after. Please find below the journal:

root@DietPi:~# systemctl stop gmrender
root@DietPi:~# killall -w gmediarender
gmediarender: aucun processus trouvé
root@DietPi:~# /usr/local/bin/gmediarender -u b56c9bf7-dc42-42f1-811f-d1def6a26e                             5a -f DietPi --gstout-audiosink=alsasink --gstout-audiodevice=sysdefault --gstou                             t-initial-volume-db=-1 --logfile=stdout -I 192.168.1.26
INFO  [2020-01-22 21:52:45.097216 | main] gmediarender 0.0.8 log started [ gmedi                             arender 0.0.8 (libupnp-1.8.4; glib-2.58.3; gstreamer-1.14.4) ]
INFO  [2020-01-22 21:52:45.097330 | output] Using output module: gst (GStreamer                              multimedia framework)
INFO  [2020-01-22 21:52:45.101862 | connmgr] Registering support for 'applicatio                             n/x-apetag'
INFO  [2020-01-22 21:52:45.101955 | connmgr] Registering support for 'audio/*'
INFO  [2020-01-22 21:52:45.103280 | connmgr] Registering support for 'video/x-ra                             w'
INFO  [2020-01-22 21:52:45.104873 | connmgr] Registering support for 'applicatio                             n/x-yuv4mpeg'
INFO  [2020-01-22 21:52:45.107950 | connmgr] Registering support for 'audio/x-ra                             w'
INFO  [2020-01-22 21:52:45.150591 | connmgr] Registering support for 'applicatio                             n/x-rtp'
INFO  [2020-01-22 21:52:45.150680 | connmgr] Registering support for 'applicatio                             n/x-rtcp'
INFO  [2020-01-22 21:52:45.151016 | connmgr] Registering support for 'applicatio                             n/x-srtp'
INFO  [2020-01-22 21:52:45.151106 | connmgr] Registering support for 'applicatio                             n/x-srtcp'
INFO  [2020-01-22 21:52:45.156340 | connmgr] Registering support for 'audio/x-ma                             troska'
INFO  [2020-01-22 21:52:45.156427 | connmgr] Registering support for 'video/x-ma                             troska'
INFO  [2020-01-22 21:52:45.156470 | connmgr] Registering support for 'video/x-ma                             troska-3d'
INFO  [2020-01-22 21:52:45.156535 | connmgr] Registering support for 'audio/webm                             '
INFO  [2020-01-22 21:52:45.156587 | connmgr] Registering support for 'video/webm                             '
INFO  [2020-01-22 21:52:45.157449 | connmgr] Registering support for 'video/mpeg                             '
INFO  [2020-01-22 21:52:45.157513 | connmgr] Registering support for 'video/x-h2                             64'
INFO  [2020-01-22 21:52:45.157571 | connmgr] Registering support for 'video/x-h2                             65'
INFO  [2020-01-22 21:52:45.157626 | connmgr] Registering support for 'video/x-di                             vx'
INFO  [2020-01-22 21:52:45.157688 | connmgr] Registering support for 'video/x-hu                             ffyuv'
INFO  [2020-01-22 21:52:45.157743 | connmgr] Registering support for 'video/x-dv                             '
INFO  [2020-01-22 21:52:45.157802 | connmgr] Registering support for 'video/x-h2                             63'
INFO  [2020-01-22 21:52:45.157851 | connmgr] Registering support for 'video/x-ms                             mpeg'
INFO  [2020-01-22 21:52:45.157903 | connmgr] Registering support for 'image/jpeg                             '
INFO  [2020-01-22 21:52:45.157962 | connmgr] Registering support for 'video/x-th                             eora'
INFO  [2020-01-22 21:52:45.158005 | connmgr] Registering support for 'video/x-di                             rac'
INFO  [2020-01-22 21:52:45.158148 | connmgr] Registering support for 'video/x-pn                             -realvideo'
INFO  [2020-01-22 21:52:45.158199 | connmgr] Registering support for 'video/x-vp                             8'
INFO  [2020-01-22 21:52:45.158288 | connmgr] Registering support for 'video/x-vp                             9'
INFO  [2020-01-22 21:52:45.158340 | connmgr] Registering support for 'video/x-pr                             ores'
INFO  [2020-01-22 21:52:45.158416 | connmgr] Registering support for 'video/x-wm                             v'
INFO  [2020-01-22 21:52:45.158489 | connmgr] Registering support for 'audio/mpeg                             '
INFO  [2020-01-22 21:52:45.158538 | connmgr] Registering support for 'audio/x-mp                             eg'
INFO  [2020-01-22 21:52:45.158593 | connmgr] Registering support for 'audio/x-sc                             pls'
INFO  [2020-01-22 21:52:45.158666 | connmgr] Registering support for 'audio/L16;                             rate=44100;channels=2'
INFO  [2020-01-22 21:52:45.158751 | connmgr] Registering support for 'audio/x-ac                             3'
INFO  [2020-01-22 21:52:45.158814 | connmgr] Registering support for 'audio/x-ea                             c3'
INFO  [2020-01-22 21:52:45.158863 | connmgr] Registering support for 'audio/x-dt                             s'
INFO  [2020-01-22 21:52:45.158936 | connmgr] Registering support for 'audio/x-vo                             rbis'
INFO  [2020-01-22 21:52:45.159015 | connmgr] Registering support for 'audio/x-fl                             ac'
INFO  [2020-01-22 21:52:45.159076 | connmgr] Registering support for 'audio/x-op                             us'
[... cut ...]
&lt;upnp:class&gt;object.item.audioItem.musicTrack&lt;/upnp:class&gt;
&lt;res duration=&quot;0:03:33.000&quot; size=&quot;37676732&quot; protocolInfo=                             &quot;http-get:*:audio/x-wav:*&quot; sampleFrequency=&quot;44100&quot; bitsPerSa                             mple=&quot;16&quot; nrAudioChannels=&quot;2&quot;&gt;http://192.168.1.5:49152/au                             dirvana/audio_1_1.wav&lt;/res&gt;
&lt;/item&gt;
&lt;/DIDL-Lite&gt;
"></NextAVTransportURIMetaData>
</InstanceID>
</Event>

MichaIng, 2020-01-22:
@kikigarel 
Okay many thanks for your testing. The error message is just the same, so it is no permissions issue.

To rule it out, could you install the package from APT repo:
```
mv /usr/local/bin/gmediarender /usr/local/bin/gmediarender_dietpi
systemctl stop gmrender
apt install gmediarender
systemctl start gmediarender
```
Lets see if this works instead. It did not work for other users, which was the reason we compiled own binaries which solved the issues reported before.

If the same errors happen/music does not play as well, to revert:
```
apt purge gmediarender
mv /usr/local/bin/gmediarender_dietpi /usr/local/bin/gmediarender
```

We then need to do some more research. What you could as well test is simply another client instead of Audirvana.

kikigarel, 2020-01-23:
Hey @MichaIng,

So... Dietpi is being a bad boy :-)  Installing the package from APT repo didn't help. Actually, it made the endpoint name change from "dietpi" to "gmediarender on dietpi", and I couldn't select it anymore ("peripheral not connected" error message). I have reverted everything. 

I have used another client as well: DS Audio, the app that is bundle with the Synology NAS. It can stream to Dietpi but again nothing happens: playback just doesn't start, exactly like with Audirvana. That's is really puzzling. 

Could the issue be related to the autologin? 

Also, in Dietpi-Config / Audio Options, I realized that I have two auto detection options: _usb-dac_ (USB Audio DAC, any) or _hw:1,0_ (UD503 USB Audio). I had selected the second one from the start since UD 503 is the model of my DAC. I tried selecting the first one but it made no difference.

MichaIng, 2020-01-24:
@kikigarel 
Okay, so its not about the `gmediarender` binary, that is good to know so far.

About audio card selection:
- `usb-dac` basically searches and selects the first found USB DAC
- `hw:1,0` is the specific one that was found
- The two above hence do exactly the same, actually this needs some cleanup. I think the idea was to apply some USB DAC specific tweaks when selecting the first, but that can and should be done as well when a USB DAC is selected via its `hw:<card>,<device>` identifier...

Autologin cannot be related. GMediaRender runs as systemd service, independently from any login.

I am as well puzzled. `aplay -l` shows the one and only attached sound card, with card and device number that correctly match `asound.conf` entry. And you say playing audio locally from the Pi works as well. Permissions and specific gmediarender binary is ruled out as well.

Last idea I have so far is to check and play with `alsamixer` switches and sliders.

We maybe need to open an issue on https://github.com/hzeller/gmrender-resurrect/issues. Probably hzeller has some idea based on those error messages: https://github.com/MichaIng/DietPi/issues/3348#issuecomment-576949158

kikigarel, 2020-01-28:
Hi @MichaIng,

I have just opened a ticket here, let's see if someone finds out what's wrong :-)

https://github.com/hzeller/gmrender-resurrect/issues/205

MichaIng, 2020-02-19:
Workaround: https://github.com/hzeller/gmrender-resurrect/issues/205#issuecomment-581184651
ToDo:
- Test if sysdefault e.g. always points to hw:0,0 regardless of ALSA settings.
- In case we need scrape card + device from asound.conf or we need to assign an alias in asound.conf when sound card is chosen, that can be used by programs.

MichaIng, 2020-03-07:
For reference, PM from sudeep from Allo:

> Hi MichaIng,
> 
> Sparky Dietpi v6.28 update gm rendrer having issues , audio device always routing to device 0, not going to the default device.
> 
> In the new update noticed that --gstout-audiosink=alsasink added in service compare to previous versions.
> manually editing /etc/systemd/system/gmrender.service works fine but gmrendrer not starts on next reboot.
> 
> can you please do the needful in next update.

Actually we want ALSA to be used, since this is what DietPi-Config and DietPi-Software configures for everything and what is expected. Still strange that either due to `--gstout-audiosink=alsasink` for for some other reason sysdefault does not work in every case.

MichaIng, 2020-03-08:
So I ran some tests and indeed `sysdefault` is not a reliable PCM (audio device) name for the "default" audio device that was selected/configured via `dietpi-config` => ALSA. It appears that those PCM names are by default created multiple times for every available sound card by ALSA:
```
2020-03-08 18:01:29 root@VM-Buster:~# aplay -L | grep default
default:CARD=I82801AAICH
sysdefault:CARD=I82801AAICH
default:CARD=Device
sysdefault:CARD=Device
```
So using `sysdefault` leads to simply the first match being used, regardless if this is the wanted/chosen default or not, the same for `default`:
```
2020-03-08 18:09:10 root@VM-Buster:/tmp# aplay -f cd -vD 'default' test.wav # "sysdefault" results in exactly the same
Playing raw data 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
...
Hardware PCM card 0 'Intel 82801AA-ICH' device 0 subdevice 0
```

Using `dietpi-config`, we create an asound.conf like this:
```
2020-03-08 18:03:24 root@VM-Buster:~# cat /etc/asound.conf
pcm.!default {
        type hw
        card 1
        device 0
}

ctl.!default {
        type hw
        card 1
}
```
We overwrite the `default` PCM by this, and this leads to `aplay -L` doesn't list this PCM name anymore for any sound card:
```
2020-03-08 18:03:29 root@VM-Buster:~# aplay -L | grep default
sysdefault:CARD=I82801AAICH
sysdefault:CARD=Device
```
Playing a test file using PCM name `default` now leads to the correct sound device being used:
```
2020-03-08 18:05:40 root@VM-Buster:/tmp# aplay -f cd -vD 'default' test.wav
Playing raw data 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
Hardware PCM card 1 'USB Sound Device' device 0 subdevice 0
```
Using `sysdefault` still uses the old "wrong" one:
```
2020-03-08 18:05:47 root@VM-Buster:/tmp# aplay -f cd -vD 'sysdefault' test.wav
Playing raw data 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
...
Hardware PCM card 0 'Intel 82801AA-ICH' device 0 subdevice 0
```

Of course we can also overwrite the `sysdefault` PCM name:
```
2020-03-08 18:11:57 root@VM-Buster:/tmp# cat /etc/asound.conf
pcm.!default {
        type hw
        card 1
        device 0
}

ctl.!default {
        type hw
        card 1
}

pcm.!sysdefault {
        type hw
        card 1
        device 0
}

ctl.!sysdefault {
        type hw
        card 1
}
2020-03-08 18:12:02 root@VM-Buster:/tmp# aplay -L | grep default
2020-03-08 18:12:04 root@VM-Buster:/tmp# aplay -f cd -vD 'sysdefault' test.wav
Playing raw data 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
Hardware PCM card 1 'USB Sound Device' device 0 subdevice 0
```

But before changing too much around, the easiest thing is indeed to use `default` PCM name instead of `sysdefault`. Both are equivalent when not adding a custom `asound.conf`:
```
2020-03-08 18:31:47 root@VM-Buster:/tmp# cat /usr/share/alsa/alsa.conf | grep -E '(pcm|ctl).default'
pcm.default cards.pcm.default
pcm.sysdefault cards.pcm.default
ctl.default ctl.sysdefault
```
So to me it looks like a good convention to take `sysdefault` as really static system default, regardless of user settings, while `default` is the one based on user choice/settings.
________________
There is btw an alternative, using this:
```
2020-03-08 18:39:28 root@VM-Buster:/tmp# cat /etc/asound.conf
defaults.pcm.card 1
defaults.pcm.device 0
defaults.ctl.card 1
```
This leads to correct default sound devices being used, keeping the default ALSA `/usr/share/alsa/alsa.conf` config for those in place. This means the plughw module is used, which includes software conversion, and such, making things compatible but degrade sound quality and increase system load, from what I know and remember when testing such with Mycroft AI audio + microphone setup some time ago:
```
2020-03-08 18:39:31 root@VM-Buster:/tmp# aplay -f cd -v -D default test.wav # leaving out -D option has same result
Playing raw data 'test.wav' : Signed 16 bit Little Endian, Rate 44100 Hz, Stereo
Plug PCM: Rate conversion PCM (48000, sformat=S16_LE)
...
```
So for now we'll keep things as they are but use the PCM name `default` wherever it is required for audio software.
_______________
@kikigarel 
If you find time, might you verify this for your case, editing `/etc/systemd/system/gmrender.service` to contain `--gstout-audiodevice=default`, then
```
systemctl daemon-reload
systemctl restart gmrender
```
and check whether things work just the same as with the workaround before, where you used `hw:1,0`?

MichaIng, 2020-03-09:
I'll reopen to keep this up for testing, at least until sudeep verified this working reliable on Sparky SBC as well.

dsnyder0pc, 2020-04-16:
FWIW, changing `sysdefault` to `default` in `/etc/systemd/system/gmrender.service` worked for me.

Only problem I still have is that the gmrender service tries to start before dhcpcd has acquired an IP address:

```
root@roon-rpi3a:~# systemctl status gmrender
● gmrender.service - GMediaRender (DietPi)
   Loaded: loaded (/etc/systemd/system/gmrender.service; disabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2020-04-15 17:31:18 PDT; 21s ago
  Process: 708 ExecStartPre=/bin/dash -c systemctl set-environment ACTIVE_IP=$(mawk NR==4 /DietPi/dietpi/.network) (code=exited, status=1/FAILURE)

Apr 15 17:31:18 roon-rpi3a systemd[1]: Starting GMediaRender (DietPi)...
Apr 15 17:31:18 roon-rpi3a dash[708]: Failed to set environment: Invalid environment assignments
Apr 15 17:31:18 roon-rpi3a systemd[1]: gmrender.service: Control process exited, code=exited, status=1/FAILURE
Apr 15 17:31:18 roon-rpi3a systemd[1]: gmrender.service: Failed with result 'exit-code'.
Apr 15 17:31:18 roon-rpi3a systemd[1]: Failed to start GMediaRender (DietPi).
```
I do have `CONFIG_BOOT_WAIT_FOR_NETWORK=2`

Could try adding a sleep statement, but that seem dodgy. Do we really need the IP address?

MichaIng, 2020-04-16:
@dsnyder0pc 
Is there a certain reason why you need to run `dhcpcd`? DietPi by default comes with `isc-dhcp-client` (`dhclient`) which integrates fine with ifupdown (`/etc/network/interfaces`) and should have finished before GMediaRender is started.

Generally network, hence an IP address needs to be up before GMediaRender can bind to it. The workaround above should work for you as well: https://github.com/MichaIng/DietPi/issues/3348#issuecomment-576277715
Switching to a static IP of course is another possibility, as long as you router allows this, e.g. assigning the IP that was assigned via DHCP first as static IP via `dietpi-config`.

I see that generally the problem with missing DHCP assignment is not yet solved, regardless with which DHCP server. Either we need to switch to a ping-based network check during boot and/or we need to change `ifup@.service` (for hotplug network devices) to a different service type so that `network-online.target` is assured to be reached after interface setup has finished, rather than after setup has started, as it is currently by default on Debian 🤔.

ejokic, 2020-04-18:
i am owner of allo sparky usbridge and after update to 6.28 my gmrender does not work.
it starts and i can choose it as dlna renderer but no sound. following this conversation i edit service file but when i look at systemd status log it still says --gstout-audiodevice=sysdefault.

also i would really like to change starting volume to 100 (in bubble upnp it always starts with 97 !)

I am newbie to all this so please can you help me to get sound back on. 
my default usb device is audioquest usb and it is the choice in alsa as usb 1.1 (any)

thank you
regards

ejokic, 2020-04-19:
I tried changing asound.conf like this that you suggested in my Allo usbridge:

pcm.!default {
        type hw
        card 1
        device 0
}

ctl.!default {
        type hw
        card 1
}

pcm.!sysdefault {
        type hw
        card 1
        device 0
}

ctl.!sysdefault {
        type hw
        card 1
}

and now music plays when --gstout-audiodevice=default

Alsa info for flac 16bit 44100hz stereo file is: 

access: RW_INTERLEAVED | format: S24_3LE | subformat: STD | channels: 2 | rate: 44100 (44100/1) | period_size: 441 | buffer_size: 8820 

Still when i look at gmrender status log it says --gstout-audiodevice=sysdefault
I also tried changed default startup sound from -1 to 0 in service file but in status log it says -1 and indeed starts at 97 in bubble upnp. My dlna server is minimserver

So it seeems to me that gmrender status log and gmrender service file are not same thing.

Still, as i said my knowledge is only basic but i would like to be able to have working startup volume at 0, and possibility to search song position with slider of dlna controler (now if i try allo usbridge freezes and have to be unplugged from electricity socket.

Thank you

MichaIng, 2020-04-21:
@ejokic 
How did you try changing the --gstout-audiodevice and startup sound? Basically this needs to be done via systemd unit, followed by `systemctl daemon-reload` and `systemctl restart gmrender`. You can check whether your changes are applied via `systemctl cat gmrender`.

ejokic, 2020-04-22:
In the beggining i changed simply in dietpi-explorer the file.

Now after your message i searched internet and found command 
systemctl edit --full gmrender.service
so i changed service with this command by deleting sys part and changing -1 to 0.0 for volume part
and then i wrote back asound.conf to
pcm.!default {
        type hw
        card 1
        device 0
}

ctl.!default {
        type hw
        card 1
}

and used 

systemctl daemon-reload
systemctl restart gmrender

Now it is also working and i have startup volume at 100 (thank you very much)
when i use systemctl cat gmrender i get this:

# /etc/systemd/system/gmrender.service
[Unit]
Description=GMediaRender (DietPi)
Wants=network-online.target
After=network-online.target dietpi-boot.service

[Service]
User=gmrender
ExecStartPre=+/bin/dash -c 'systemctl set-environment ACTIVE_IP=$(mawk 'NR==4' /DietPi/dietpi/.network)'
ExecStart=/usr/local/bin/gmediarender -u '6208a890-d725-4377-b09f-3ce1a988c5d6' -f 'DietPi' --gstout-audiosink=alsasink --gstout-audiodevice=default --gstout-initial-volume-db=0.0 --logfile=stdout -I "$ACTIVE_IP"

[Install]
WantedBy=multi-user.target

i hope it is ok like this.

Can you maybe also point me where can i find more information about gmrender slide and search position in song via dlna controllers possibility?
Thank you very much

MichaIng, 2020-04-22:
@ejokic 
Many thanks for testing. That is great and matches the solution/change we did for v6.29. I'll also adjust the default volume to 0.0 as you did, "-1" does not seem to be a good default and also does not match any default/guide I can find in GMediaRender docs.

> Can you maybe also point me where can i find more information about gmrender slide and search position in song via dlna controllers possibility?

That is out of my knowledge as well, would require me to read into the topic. I guess best is to start reading through the specific DLNA controller/client documentation 🤔.

MichaIng, 2020-04-22:
Ah about the volume: https://github.com/hzeller/gmrender-resurrect/blob/master/INSTALL.md#--gstout-initial-volume-db
"-1" means -1 decibel, hence ~97% volume should be shown on controller. Actually I don't know any reason to not set 100% (0.0) which is default anyway. So we can simply remove this option from the systemd unit.

ejokic, 2020-04-22:
Excellent.
Thank you once again for your quick response, i will try to find something in documentation as suggested. If i find anything i will put it here.

ejokic, 2020-04-22:
Btw, is this that we are talking about old version of gmrender or new, updated version called gmrender-resurrect?
i found some info about this so maybe give it a try if it is still here old version we are talking about.
Thanks

MichaIng, 2020-04-22:
@ejokic 
Yes, we use `gmrender-resurrect`. The package from Debian repo, called `gmediarender` is build from `gmrender-resurrect` as well, as ours. The distros, including us, kept the old name in place to have a smooth transition, but actually the project name is `gmrender-resurrect`.

closed 2020-05-03 by MichaIng